esp8266/Arduino #7150: MQTT fails after 2.4.2

full-stack-ex, 2020-03-12:
----------------------------- Delete below -----------------------------
### Basic Infos

- [x ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x ] I have searched the issue tracker for a similar issue.
- [x ] If there is a stack dump, I have decoded it.
- [ x] I have filled out all fields below.

#### Platform

- Hardware:				ESP8266
- Core Version:			2.6.3 (seems to be any one after 2.4.2)
- Development Env:		Arduino IDE
- Operating System:		Ubuntu

### Settings in IDE

- Module:  			NodeMCU 1.0 (ESP12E Module)
- Flash Mode:			default (not changed)
- Flash Size: 		4MB
- lwip Variant:   v2 Lower Memory
- Reset Method:		default (not changed)
- Flash Frequency:	40Mhz
- CPU Frequency:		80Mhz
- Upload Using:		SERIAL
- Upload Speed:   115200

### Problem Description

MQTT connects fine in 2.4.2 but it fails to connect in 2.6.3 and, apparently, any other version after 2.4.2

Connected to IBM cloud via MQTT/SSL (port 8883).

The same behavior observed with a proprietary program and with the standard example.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch
https://github.com/knolleary/pubsubclient/blob/master/examples/mqtt_esp8266/mqtt_esp8266.ino

### Debug Messages

Attempting MQTT connection...failed, rc=-2 try again in 5 seconds

d-a-v, 2020-03-12:
Can you retry with [git master version](https://arduino-esp8266.readthedocs.io/en/latest/installing.html#using-git-version) or [beta version 0.0.1](https://d-a-v.github.io) ?

TD-er, 2020-03-12:
Can you also try with my version of the PubSubClient?

https://github.com/letscontrolit/ESPEasy/tree/mega/lib/pubsubclient/src

That one is also used in Tasmota.

full-stack-ex, 2020-03-12:
@d-a-v , I only had the time to try the git master version, and it failed. But I need to try one more time to make sure there was no mistake on my side.

@TD-er , I'll do my best to try it shortly as well.

(For sure, when reverted to 2.4.2, it works again.)

d-a-v, 2020-03-13:
@full-stack-ex FYI this sketch is running fine here with git master (and we are not alone using this library).
Have you tried the option "erase all flash" in the arduino menu ?

full-stack-ex, 2020-03-13:
@d-a-v , actually, I realized my sketch may be essentially different, for what it's worth.
The original by the link above was modified to connect to a secured server (port 8883, TLS) with additional parameters in connect().
That seems line "another layer" but you never know.
Anyway, I'm going to get back to it shortly.

devyte, 2020-03-13:
The WiFiClientSecure underlying lib was replaced since back then due to fragility, a long list of issues, and stagnation. It was axtls and was replaced with bearssl. Please look at the examples for usage.

Closing due to user error.

closed 2020-03-13 by devyte

full-stack-ex, 2020-03-17:
Sorry, really got stuck with this. Please help find out what that user error was.

My code used to build, and devices are being manufactured with it. It does not build any longer. I understand it about the underlying library replacement, but the current versions you mentioned fail anyway.

@d-a-v , yes,I have tried both the proprietary code and several open examples with what you recommended
> Can you retry with git master version or beta version 0.0.1 ?
Cloned master, installed the beta from https://d-a-v.github.io/esp8266/Arduino/package_esp8266com_index.json

Got state() = -2 in all cases

I'll follow up with the exact test code in a separate clean post.

Thanks

full-stack-ex, 2020-03-17:
This is the code with credentials masked, MQTT in IBM IoT cloud.
[pubsub-auth.ino.txt](https://github.com/esp8266/Arduino/files/4341767/pubsub-auth.ino.txt)

Getting this with the master, beta, and the recent release (2.6.2):
> WiFi connected
> IP address: 
> xxx.xx.xx.xxx
> Attempting MQTT connection...failed, rc=-2 try again in 5 seconds
> Attempting MQTT connection...failed, rc=-2 try again in 5 seconds

Same code with 2.4.2 shows this:
> WiFi connected
> IP address: 
> 172.16.42.111
> Attempting MQTT connection...connected
> Published: 1
> Publish message: hello world #1
> Publised: 1
> State: 0
> 

What am I doing wrong?

Does anybody have a working example for any of the recent builds with MQTT over TLS?

@TD-er , I still need to try your version. Tomorrow.

d-a-v, 2020-03-17:
@full-stack-ex As said above, SSL has changed since 2.4.2.
We are now using BearSSL (previously AxTLS not anymore maintained) and you may be interested in [this SSL validation example](https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/examples/BearSSL_Validation/BearSSL_Validation.ino).
In short, you must tell *whether* or *how* you accept to validate remote peer certificate, otherwise you will not be connected.
Your MQTT library should have been updated in that way since then.