esp8266/Arduino #7158: HTTP and HTTPS request at same time

array81, 2020-03-18:
Is there problem to use HTTP and HTTPS request on same device.
On my device I have a webserver to config it web html interface. Besides I use telegram to remote control (out of home). I use a https://github.com/witnessmenow/Universal-Arduino-Telegram-Bot library to read telegram messages every 5/10 seconds
The device worked until a few weeks ago when I have upgrade library from 2.5.2 to 2.6.3 and added telegram fingerprint on code (without fingerprint no longer works).
Now telegram works without problem but if I open the webserver page the device restart the next telegram control.
I also have try to replace telegram library with the same result.

Note. I use a nodemcu as device.

d-a-v, 2020-03-18:
> Now telegram works without problem but if I open the webserver page the device restart the next telegram control.

This is unclear. In the issue template that you deleted, it is stated that we need an MCVE to be able to reproduce the error. You provided more data in the [original issue](https://github.com/witnessmenow/Universal-Arduino-Telegram-Bot/issues/144) but this is still not enough.

We are of course interested in fixing regressions, but we need insights.

devyte, 2020-03-19:
It depends a lot on the app, but my experience is that a webserver leaves about 22KB while handling a request, assuming nothing else is in mem, and a secure connection requires quite a bit of mem. 
The old core used axtls, which I understand had a slightly lower mem requirement due to older ciphers. Our current core uses bearssl, which is (mostly) current tech.
My suspicion is you're oom.

devyte, 2020-03-22:
No MCVE, closing due to not compliant.

closed 2020-03-22 by devyte

fritsjan, 2020-07-20:
I would also like to know if it is possible to have a webserver running AND telegram bot at the same time.
An example how to integrate the two would be fantastic...

In my very brief search, I found that it is not easy to have a client and server running at the same time. You have to start and stop one before stopping and starting the other.

I would like to listen to a get request from an url (esp32 running as a webserver) and accordingly send a message with the telegram bot...