esp8266/Arduino #7159: String with more than about 10 characters doesn't stay in EEPROM after reset.

marian97-esp, 2020-03-19:
I recently discovered a problem with storing long strings or chars in EEPROM memory.
Here is what I do:
-I write a String to the EEPROM
-I read the String from the EEPROM and it's the good string
-I reboot the ESP
-If I read the String again from the EEPROM it's not the original string, it's garbage, sometimes even pieces of serial output or code.

I posted the test code below. 
To reproduce this step you need to write the data to eeprom, modify the code so it doesn't upload it again, upload the code to read the data from eeprom and check to see if the data is correct.

Here is my test code:
#include <EEPROM.h>

String data;

void setup() {
  Serial.begin(115200);
  Serial.println();
  Serial.println();
  Serial.println();

//comment from here to below after the data has been wrote in the eeprom
  data = "very_long_string_to_write"; 

  EEPROM.begin(100);
  EEPROM.put(10, data);
  EEPROM.commit();
  EEPROM.end();
//comment until here  

  delay(1000);
  
  EEPROM.begin(100);                              
  EEPROM.get(10, data);
  EEPROM.commit();
  EEPROM.end();

  Serial.println(data);
}

void loop() {
}

pipi61, 2020-03-19:
The eeprom address range somewhere in flash. If you upload another size program, the eeprom address space can be moved to a different address range. If you need data storage, use the spiffs files to store it.
I think....

pipi61, 2020-03-19:
check this [https://www.john-lassen.de/en/projects/esp-8266-arduino-ide-webconfig](url)

jandrassy, 2020-03-19:
put stores the String object, but not the data it manages. String object has only a pointer to data in heap. don't put String in EEPROM. it is like putting a dog leash in the cage instead of a dog.

marian97-esp, 2020-03-19:
@jandrassy Hmm, it makes sense what you are saying but if the String it's smaller than 9 or 10 characters it has no problem storing it and reading it after a reboot. I tried it with char too and the same behavior applies.

I will look into spiffs, but what other method can I use to store around ~20 characters in to EEPROM?

devyte, 2020-03-19:
@jandrassy is correct, String is a managed object and is therefore not self contained, so you can't just store it like that. You either have to serialize it, or extract the relevant data from it to store that. 
I suggest storing the length in the first 2 bytes, then the chars after that (data.c_str()). Then read the first 2 bytes into a length var, then read length chars into a char[] buffer, then assign the buffer to your data String. 
Please seek such generic help at a community forum like esp8266.com or stackoverflow. This is an issue tracker meant for tracking issues in the core. 
Closing due to user error.

closed 2020-03-19 by devyte