esp8266/Arduino #7160: New router from ISP and now no ESP8266 modules will connect

majenkotech, 2020-03-20:
Code that has been working for ages now suddenly will not connect to the new router that I have been sent by my ISP.

I have tried every combination of encryption and key exchange on the router, but nothing makes a difference.

Here's a full debug log:

```
[WIFI] delete old wifi config...
scandone
[WIFI] start scan
:ur 1
scandone
[WIFI] scan done
[WIFI] 5 networks found
 --->  0: [2][xx:CF:xx:C2:xx:2C] MySSID (-56) *
       1: [6][xx:FE:xx:4D:xx:B9] Neighbour-1 (-86) *
       2: [6][xx:3B:xx:66:xx] Neighbour-2 (-86)  
       3: [6][xx:FE:xx:4D:xx:BA] Neighbour-3 (-86)  
       4: [11][xx:10:xx:0C:xx:BD] Neighbour-4 (-88)  

[WIFI] Connecting BSSID: xx:CF:xx:C2:xx:2C SSID: MySSID Channel: 2 (-56)
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 2
cnt 

connected with MySSID, channel 2
dhcp client start...
wifi evt: 0
ap_loss
state: 5 -> 0 (0)
rm 0
reconnect
wifi evt: 1
STA disconnect: 8
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 2
cnt 
ap_loss
state: 5 -> 0 (0)
rm 0
reconnect
wifi evt: 1
STA disconnect: 8
scandone
no MySSID found, reconnect after 1s
wifi evt: 1
STA disconnect: 201
reconnect
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 1
cnt 

connected with MySSID, channel 2
dhcp client start...
wifi evt: 0
ap_loss
state: 5 -> 0 (0)
rm 0
reconnect
wifi evt: 1
STA disconnect: 8
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 2
cnt 

connected with MySSID, channel 2
dhcp client start...
wifi evt: 0
ap_loss
state: 5 -> 0 (0)
rm 0
reconnect
wifi evt: 1
STA disconnect: 8
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 1
cnt 
ap_loss
state: 5 -> 0 (0)
rm 0
reconnect
wifi evt: 1
STA disconnect: 8
[WIFI] Connecting Failed (6).
:ur 1
:ur 1
:ur 1
:ur 1
:ur 1
:ur 1
```

Now, I have no idea what any of that means, but from my reading it looks like the AP vanishes part way through connecting. But if someone could help decode that lot it would certainly help me work out what is going wrong here...

d-a-v, 2020-03-20:
What version of the core and firmware are you using ?
(First line at boot in debug mode)

majenkotech, 2020-03-20:
SDK:3.0.0-dev(c0f7b44)/Core:2.6.3=20603000/lwIP:STABLE-2_1_2_RELEASE/glue:1.2-16-ge23a07e/BearSSL:89454af

Although I have tried the different versions of the SDK, and I was using an earlier version of the core, but upgraded to the latest and greatest to try and fix the problem.

I think it may be related to #2795 but nothing I have tried from that thread has worked. Well - at one point I did manage to get an IP address briefly, but it disconnected almost immediately afterwards.

It's obviously something related to this router specifically (EchoLife DG8041W Home Gateway, provided by TalkTalk, arguably the worst ISP in the UK - almost as bad as some American ones...), and for now I have configured a spare Pi as an access point dedicated to my IoT devices (of which I have a plethora and now have to go round them all and reprogram them manually because I haven't yet got round to implementing WiFiManager on them...)

MarkyAD, 2020-03-20:
If router has MAC address filtering enabled, disable it and test.
If router has DHCP server disabled, enable it and test.

majenkotech, 2020-03-20:
MAC filtering is disabled already. DHCP is enabled already.  BTW - some history: I'm a former network administrator of a large network of servers, so I have tried the obvious things you can be sure.

majenkotech, 2020-03-20:
Ah, I think I have something.

I haven't managed to get one of my pre-existing devices to connect, but I *have* managed to get one of my newly programmed ones to connect.

A combination of a newer version of the core / SDK than I had before plus turning off ICMP Flood DoS protection on the router.  It seems to be thinking it's undergoing an ICMP flood attack and tells the ESP8266 to sod off.

jandrassy, 2020-03-21:
changing SDK always causes problems (2.5 has SDK 3 all other versions of core have 2.2).

flood DoS detection on LAN side?

majenkotech, 2020-03-21:
That's certainly the way it appears. With ICMP Flood protection turned on it kicks off all the ESP8266 modules. Turn it off again and they are allowed to connect.

TD-er, 2020-03-22:
Hmm that's an interesting option to check.
I have noticed a specific MikroTik AP I'm using often has issues with newly connected nodes.
It may take minutes before the first successful network reply is transferred via that AP.

Maybe the Gratuitous ARP is triggering some protection mechanism?