esp8266/Arduino #7161: Support ESP8285 (2MB flash)

mcmega, 2020-03-21:
Hello!
I have ESP8285 installed with 2MB flash (new version of chips).
I understand you need to edit the file boards.txt
How to make support for these chips?

d-a-v, 2020-03-21:
`tools/boards.txt.py` needs to be updated (it generates `boards.txt`)
I wasn't aware of this new version, that's nice

TD-er, 2020-03-21:
Hmm that's a nice development.
What products do use this new version and how can we distinguish them?

Tech-TX, 2020-03-21:
If it's as simple as this in boards.txt.py, I can do a quick PR for it:
```
    ( 'esp8285', {
        'name': 'Generic ESP8285 Module',
        'opts': {
            '.build.board': 'ESP8266_ESP01',
            '.build.variant': 'esp8285'
            },
        'macro': [
            'resetmethod_menu',
            'resetmethod_menu_extra',
            'crystalfreq_menu',
            'flashmode_dout',
            'flashfreq_40',
            '1M', '2M',  <<<
            'led',
            ],
```
I only have the N08 chips, so I can't verity it works correctly.

mcmega, 2020-03-21:
ESP8285N08 - 1 MB (–40°C ~ 85°C)
ESP8285H08 - 1 MB (–40°C ~ 105°C)
ESP8285H16 - 2 MB (–40°C ~ 105°C)

d-a-v, 2020-03-22:
@Tech-TX I guess you're right
Also #6690 would help to have flash-size agnostic sketches

PR #7163 (2020-03-22): Add 2MB obtion to 8285 generic board
The ESP8285H16 is reported to have 2MB of flash, so add an option the
the Generic 8285 board for 2MB.

Fixes #7161
commit 1127a090 (2020-03-22): Add 2MB obtion to 8285 generic board (#7163)

The ESP8285H16 is reported to have 2MB of flash, so add an option the
the Generic 8285 board for 2MB.

Fixes #7161
closed 2020-03-22 by d-a-v