esp8266/Arduino #7164: i2s Input

artua, 2020-03-24:
#### Platform

- Hardware:					[other]
- Core Version:			[db75d2c448bfccc6dc308bdeb9fbd3efca7927ff]
- Development Env:		[Arduino IDE]
- Operating System:		[MacOS]

### Settings in IDE

- Module:  			[Generic ESP8266 Module]
- Flash Mode:			[qio]
- Flash Size: 		[2MB]
- lwip Variant:   [v2 Lower Memory]
- Reset Method:		[ck]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[160MHz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200]

### Problem Description

Trying to get i2s input to work with i2s microphone without luck.
The example provided just returns 0 0 as output.
The microphone attached is Knowles SPH0645LM4H-B.
https://eu.mouser.com/datasheet/2/218/ph0645lm4h-b-datasheet-rev-c-1525723.pdf

The main concern, i2s microphones are 24-bit i2s. How can we get 16-bit stream from them?

Also tried this code: https://github.com/joextodd/listener/blob/master/listener.ino
It gives me back some sound, but distorted. 
Tried ICS-43434 mic also, got the same distorted sound with this code.

Any ideas?

Thanks!
### Debug Messages

```
SDK:2.2.2-dev(a58da79)/Core:2.6.3-96-gdb75d2c4=20603096/lwIP:STABLE-2_1_2_RELEASE/glue:1.2-30-g92add50/BearSSL:5c771be
force slp enable,type: 2
fpm open,type:2 0
i2S real rate: 32051.281250]
0 0
0 0
0 0
0 0
0 0
0 0
0 0
0 0
```

artua, 2020-03-24:
@earlephilhower help please!

earlephilhower, 2020-03-24:
There are a few gotchas with that chip, AFAICT.

First, it's not speaking real I2S (at least not per the Philips spec https://www.sparkfun.com/datasheets/BreakoutBoards/I2SBUS.pdf ).  In I2S there's a 1-clock delay from WS changing to the MSB of the channel, but that chip has 0-bit delay.  So, you will need to edit the listener.ino or Arduino core to adjust the I2S bit delay or everything will be shifted.

Next, per your example output, it looks like it absolutely requires the exactly 64 clocks between successive readouts or it doesn't report anything.  With other 24b I2S microphones (incl. the pair on the GOOG AIY voice hat) they can handle having only 32 clocks between readouts so the existing Arduino core just gets the 16 MSB bits (which works just fine for record/playback, equivalent of dividing every sample by 256).  So you're going to need to make sure that the listener.ino sketch is running at 32 clocks/sample (64clocks total WS period).

The sampling frequency looks also needs to be between 16K and 32K (looks OK from your sketch).

For Arduino core, we don't have an API yet to support 24 of 32bps, so there's nothing we can do now.  That's one of the 3.0 things and not there just yet.  It's simple enough, but going >16b means you have space for only 1/2 the samples meaning RAM use either gets doubled or you have 1/2 as long between IRQs to do whatever work you need to do...

artua, 2020-03-24:
For your first question (quoted from datasheet):
> I2S Timing And Port Configuration Guidelines
> • BCLK period must be in the range shown in Figure 2 and the table below it
> • WS must be synchronized (changing) on the falling edge of BCLK
> • WS must be BCLK/64
> • The Hold Time (see Figure 2) must be greater than the Hold Time of the Receiver
**> • The mode must be I2S with MSB delayed 1 BCLK cycle after WS changes**

So seems it's ok with real i2s.

Google AIY seems to be ics-43432 which is the same as 43434 and I have the same problems with it.

Quoted from ICS-43432 datasheet:
> I2S DATA INTERFACE
> The slave serial data port’s format is I2S, 24-bit, twos complement. There must be 64 SCK cycles in each WS stereo frame. The L/R control pin determines whether the ICS-43432 outputs data in the left or right channel. When set to the left channel, the data will be output following WS’s falling edge and when set to output on the right channel, data will be output following WS’s rising edge.

artua, 2020-03-24:
I'm ok with 16 bit resolution. Any chance to get it work simple?

earlephilhower, 2020-03-24:
If you have an GOOG AIY and it's not working, there is something else going on.  That's the MEMS microphone pair I used myself to develop the .INO and changes to I2S.c.  Are you sure you have a good connection (and proper pin connections) for the I2S bus?  It is a 2MHz signal that's coming in, so it needs to be low impedence and low noise.

earlephilhower, 2020-05-20:
@artua, I just re-hooked up the Google AIY to see if the I2S input stuff got broken since I last tried it.  With master, it's running fine:

[image]

So, I think for this you really need to check your wiring, AFAICT the core itself is fine...

devyte, 2020-05-20:
Closing for now due to can't reproduce.

closed 2020-05-20 by devyte

artua, 2020-05-20:
Where are many strange things happen, I’ll update this as I came back to
project.
Thanks for checking!

Fireflaker, 2020-10-09:
Great information! I am trying to build a wifi mic with the Google AIY mic hat and ESP8266. Are there example codes for I2S mic to webstream audio?

remyhx, 2024-11-19:
> In I2S there's a 1-clock delay from WS changing to the MSB of the channel, but that chip has 0-bit delay. So, you will need to edit the listener.ino or Arduino core to adjust the I2S bit delay or everything will be shifted.

[image]

Maybe it changed (Ends with -1 now, not -B), but the bit shifting seems not necessary anymore, if we can trust the datasheet.