esp8266/Arduino #7171: ESP8266WiFiMesh.h private typedefs prevent new'ing object

RotateAt60MPH, 2020-03-28:
----------------------------- Delete below -----------------------------

If your issue is a general question, starts similar to "How do I..", is related to 3rd party libs, or is related to hardware, please discuss at a community forum like esp8266.com.

INSTRUCTIONS
============
If you do not follow these instructions, your issue may be dismissed.

1. Follow the checklist under Basic Infos and fill in the [ ] spaces with an X.
2. Fill in all the fields under Platform and Settings in IDE marked with [ ] (pick the correct option for you in each case, delete the others).
3. If you haven't already done so, test your issue against current master branch (aka latest git), because it may have been already fixed.
4. Describe your problem.
5. If you have a STACK DUMP decode it:

https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/stack_dump.html

6. Include a Minimal Complete Reproducible Example sketch that shows your issue. Do not include your entire project, or a huge piece of code.
7. Include debug messages:

https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/debugging.html 

8. Use markup (buttons above) and the Preview tab to check what the issue will look like.
9. Delete these instructions from the above to the below marker lines before submitting this issue.

----------------------------- Delete above -----------------------------

### Basic Infos

- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [X] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [X] If there is a stack dump, I have decoded it.
- [X] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12|ESP-01|ESP-07|ESP8285 device|other]
- Core Version:			[latest git hash or date]
- Development Env:		[Arduino IDE|Platformio|Make|other]
- Operating System:		[Windows|Ubuntu|MacOS]

### Settings in IDE

- Module:  			[Generic ESP8266 Module|Nodemcu]
- Flash Mode:			[dio]
- Flash Size: 		[4MB]
- lwip Variant:   [v1.4]
- Reset Method:		[nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200] (serial upload only)

### Problem Description

Cannot instantiate ESP8266WiFiMesh via 'new' in a user-defined class because needed typedefs are private.  The typedefs should be marked public.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```h

#include <Arduino.h>
#include <ESP8266WiFiMesh.h>

class MyMeshWrapper
{
public:
  MyMeshWrapper::MyMeshWrapper();
private:
  ESP8266WiFiMesh * _pMesh;
}

```

```cpp
#include <Arduino.h>
#include <ESP8266WiFiMesh.h>

MyMeshWrapper::MyMeshWrapper()
{
  ESP8266WiFiMesh::requestHandlerType rxType /* = <something> */;
  ESP8266WiFiMesh::responseHandlerType rspType /* = <something> */;
  ESP8266WiFiMesh::networkFilterType filterType /* = <something> */;
  
  _pMesh = new ESP8266WiFiMesh(rxType, rspType, filterType);
}
```

### Debug Messages

```
In file included from include/includes.h:2:0,
                 from src\MeshHelper.cpp:1:
C:\Users\<user>\.platformio\packages\framework-arduinoespressif8266\libraries\ESP8266WiFiMesh\src/ESP8266WiFiMesh.h: In constructor 'MeshHelper::MeshHelper()':
C:\Users\<user>\.platformio\packages\framework-arduinoespressif8266\libraries\ESP8266WiFiMesh\src/ESP8266WiFiMesh.h:64:68: error: 'typedef class std::function<String(const String&, ESP8266WiFiMesh&)> ESP8266WiFiMesh::requestHandlerType' is private
   typedef std::function<String(const String &, ESP8266WiFiMesh &)> requestHandlerType;
```

devyte, 2020-04-02:
CC @aerlon

aerlon, 2020-04-13:
This should work with the updated classes in the next version of the mesh framework, see PR #6280. Note that the `ESP8266WiFiMesh` class has been deprecated, though. It is advised that you use either of the `EspnowMeshBackend` or the `TcpIpMeshBackend` classes in the new framework (or `MeshBackendBase` if you really want to build your own class). Also check out the new `FloodingMesh` class and `HelloMesh` example, it may already do what you want to do.