esp8266/Arduino #7172: BareSSL setTimeout does not work

nimaltd, 2020-03-28:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ ] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12|
- Core Version:			[2.3.3]
- Development Env:		[Platformio]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Generic ESP8266 Module]
- Flash Mode:			[qio]
- Flash Size: 		[4MB]
- lwip Variant:   [v2 Lower Memory|Higher Bandwidth]
- Reset Method:		[nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200)

### Problem Description
BearSSL::WiFiClientSecure   clientssl;
.
.
.

I try to change timeout but , it does not effect. it is always 15 seconds.
clientssl.setTimeout(1000);

devyte, 2020-03-29:
> Core Version: [2.3.3]

That can't be correct.

d-a-v, 2020-03-29:
This is probably [the commented cause](https://github.com/esp8266/Arduino/blob/a04c3244d5bca7434a8825ad855c1af62489a762/libraries/ESP8266WiFi/src/WiFiClientSecureBearSSL.cpp#L72).

nimaltd, 2020-03-30:
if this timeout work with connecting timeout, it does not work for me.
I have another problem yet. but It is working before.
esp reset after connecting SSL.

[hostByName] request IP for: google.com
[hostByName] Host: google.com IP: 216.58.208.78
:ref 1
BSSL:_connectSSL: start connection
:wr 215 0
:wrc 215 215 0

Soft WDT reset

>>>stack>>>

ctx: cont
sp: 3ffff2d0 end: a57f23a8 offset: 01b0
3ffff480:  01aaa3d3 00000000 3fff23a4 40207f73  
3ffff490:  00000000 00000000 3ffef028 00000008

d-a-v, 2020-03-30:
@nimaltd I understand you are trying for a longer timeout.
Can you comment the line I cite above and report ?

nimaltd, 2020-03-30:
I try to shorter timeout. but it is not a big problem, now i have another problem.
system reset after 
clientssl.setInSecutre();
clientssl.connect(google.com",443");

debug message 

[hostByName] request IP for: google.com
[hostByName] Host: google.com IP: 216.58.208.78
:ref 1
BSSL:_connectSSL: start connection
:wr 215 0
:wrc 215 215 0

Soft WDT reset

stack>>>

ctx: cont
sp: 3ffff2d0 end: a57f23a8 offset: 01b0
3ffff480: 01aaa3d3 00000000 3fff23a4 40207f73
3ffff490: 00000000 00000000 3ffef028 00000008
.
.
.

but if i call that command in main loop, it works.
I want to call that function in my AT-COMMANDS. I am so Confused

#include <Arduino.h>
#include "Functions.h"

void setup()
{  
  wifi_init();
  serial_init();
}

void loop()
{
  serial_loop();
  wifi_loop();
}

//////////////////////////////////

and in wifi_loop() is 

       //  SSL connecting
        if(connectionType == 2)
        {
            if(WiFi.status() != WL_CONNECTED)
            {
                connectionTry = 0;
                Serial.printf_P("\r\n+SSL:ERROR\r\n");
                Serial.printf_P("\r\nERROR\r\n");
                goto END;
            }
            Serial.printf("\r\n+SSL:TRY LEFT %d\r\n",connectionTry);
            clientSsl.setTimeout(connectionTimeout);
            clientSsl.setInsecure();
            if(clientSsl.connect(connectionAddress, connectionPort))
            {
                Serial.printf_P("\r\n+SSL:OPEN\r\n");
                connectionTry = 0;
                goto END;
            }
            connectionTry--;
            if(connectionTry == 0)
            {
                Serial.printf_P("\r\n+SSL:ERROR\r\n");
                Serial.printf_P("\r\nERROR\r\n");
            }
        }

d-a-v, 2020-03-30:
We don't see your messages `TRY LEFT...` in the debug log.
It works in the main loop ?

Pablo2048, 2020-03-30:
@nimaltd  if you want to use `printf_P`, then you have to use strings from PROGMEM, or use the `PSTR()` macro.

nimaltd, 2020-03-30:
@d-a-v i dont copy it
Yes it is in main loop.

nimaltd, 2020-03-30:
@Pablo2048 printf_p works fine.

Pablo2048, 2020-03-30:
It's not about "works fine", it's about you did it wrong so it needs to be corrected...

d-a-v, 2020-03-30:
If it works in main loop and not in another function, it may be because the function is a callback.
Callbacks must not do complex things. You need to defer complex things from callback by using scheduled functions.

Anyway we can't understand your problem if you give partial informations.
If SSL works in main loop but not in your functions, we can't help you much as the bug is not in the core. If you give full informations that show the there is an issue with the core libraries then we can investigate.

Best is to give an MCVE.

nimaltd, 2020-03-30:
I did not use any callback function in my project.

nimaltd, 2020-03-30:
@Pablo2048 thank you . I try with printf .

nimaltd, 2020-03-30:
@Pablo2048 i change it . but still reset after ssl.connect

nimaltd, 2020-03-30:
@d-a-v I solved my problem. I defined a local variable(2048 bytes). after I defined global  , its work.
but SSL timeout still can not be 5000

devyte, 2020-03-30:
SSL timeout is 15000.
Closing due to user error.

closed 2020-03-30 by devyte

d-a-v, 2020-03-30:
@nimaltd timeout is set to such duration because these 15secs are needed to complete ssl connections.
It wouldn't be fair to abort due to a timeout when more time is needed.