esp8266/Arduino #7173: string: cannot explicitly set length

Spongman, 2020-03-29:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have searched the issue tracker for a similar issue.

#### Platform

- Hardware:				esp8266
- Core Version:			not sure. latest platformio 3/28/2020. "2.4.0" ??
- Development Env:		Platformio
- Operating System:		Windows

### Problem Description

`String::begin()` return writable buffer. but there's no way to explicitly set the length of the string after writing to it. `setLen()` is not available.

the code below outputs `0`, expected `5`.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp
#include <Arduino.h>

void setup() {
  Serial.begin(115200);

  String foo;
  foo.reserve(20);
  vsnprintf(foo.begin(), 20, "%d", 12345);
  Serial.println(foo.length());
}

void loop() {
}
```

d-a-v, 2020-03-29:
What's the purpose of this line ?

mathertel, 2020-03-29:
It is a general programming question and not related to esp8266. 

You can't do this because the snprintf is evaluated before setLen.
So it will first fill up the (unallocated) buffer and the allocate the buffer in the length 19 or more.
The sketch is right by doing it in the right order.

devyte, 2020-03-29:
The purpose of begin() and end() is for iterator-like behavior as so that String can be used like a container. They are not meant for direct manipulation of the internals as you intend to do. Strictly speaking, they should return an iterator type instead of a char *, but that's a different discussion.
Allowing to set the length externally defeats the purpose of a managed object, and opens up a big can of worms for users to trip and get themselves into trouble due to user mistakes. 
There cleaner ways to achieve what you want. 
Closing as won't fix.

closed 2020-03-29 by devyte

Spongman, 2020-03-30:
| They are not meant for direct manipulation of the internals

that's weird, because there are both const and non-const versions of begin:

```c++
char* begin() { return wbuffer(); }
const char* begin() const { return c_str(); }
```

shall i add a bug to remove the non-const version, then?

| There cleaner ways to achieve what you want.

i'm trying to initialize a string value without first having to allocate another temporary buffer. i couldn't find a cleaner way to achieve this. any chance you could point me to one?

earlephilhower, 2020-03-30:
@Spongman The root cause of this is that we're just implementing the existing Arduino String class API which doesn't have a setLen()-type method that I can see.  https://github.com/arduino/ArduinoCore-samd/blob/master/cores/arduino/WString.h

I believe @devyte's description is correct and that the Arduino folks didn't intend for an end user to change anything in the `String` buffer directly.  Which leaves @Spongman's question as to why there are 2 begin() calls. 

@devyte maybe the right thing to do is break with the Arduino API compat. by dropping the writable buffer return call (`char *begin()`), leaving only the `const char *begin()` call.

Spongman, 2020-03-30:
just going to put this here: https://en.cppreference.com/w/cpp/string/basic_string/resize

Spongman, 2020-03-31:
> ...  the Arduino folks didn't intend for an end user to change anything in the

the title of this commit would seem to indicate otherwise: https://github.com/arduino/Arduino/commit/f68a722bde0ea12b8dcdca545fb689a08a1ee4e2.

devyte, 2020-03-31:
> internals 

as in private class members, not as in the buffer content. You can manipulate the buffer content externally, but not manage the buffer yourself. 
Please read about iterator and const_iterator in STL. The use cases here are similar. The const version is sort of like a read-only access to one char, while the non const is like a rw access to one char. As I said, strictly speaking begin() should return an iterator type, and not a char ptr. A char ptr behaves like an iterator (or rather, one case of an iterator is an element ptr), but you're confusing the intended semantics with a ptr to a char[], where you can do C-style manipulation of the char buffer, which leads to resizing/realloc etc, and that's a No.

devyte, 2020-03-31:
> there's no way to explicitly set the length of the string after writing to it. setLen() is not available

That was your request. That is not the same as a resize method like std::string::resize(). Is your need really for a resize() method? Its use is not the same as your intended use in your original post.

Spongman, 2020-03-31:
yes, a `resize()` method would allow me to do exactly what i want:

```cpp
  String foo;
  foo.resize(20);
  foo.resize(min(foo.length()-1, snprintf(foo.begin(), foo.length(), "%d", 12345)));
  Serial.println(foo.length());
```

devyte, 2020-03-31:
And how is that different from e. g.:
```cpp
String foo;
foo.concat(12345);
```
or
```cpp
String foo;
foo.reserve(20);
foo += 12345;
```
?

Spongman, 2020-03-31:
that works for the **MCVE**, but not for the general case.

devyte, 2020-03-31:
What temporary buffer? I don't see a difference in internal allocation from a potential resize(). 
There is also StreamString, which inherits from Print, which has print/printf methods. Why is that not enough for your use case?

Spongman, 2020-03-31:
sorry, yes, i forgot about std::move in the trivial case, but change the format string to anything more complicated than the minimal example that was requested and you get the unnecessary reallocation(s). no StreamString also reallocates.