esp8266/Arduino #7175: i2c slave issue

tharun4477, 2020-03-29:
# # Basic Infos

- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:			       [ESP-12E LOLIN]
- Core Version:			[2.6.3]
- Development Env:		[Arduino IDE 1.8.8]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Nodemcu]
- Flash Mode:			[DOUT]
- Flash Size: 		[4MB]
- lwip Variant:   [v2 Lower Memory]
- Reset Method:		[ck]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200]

### Problem Description
ESP as a slave unable to receive and send data to master

Here I am using Nodemcu as a slave and Arduino as a master [problem comes here]
Nodemcu (ESP 12E)  as a slave not receiving any kind of data means onReceive (receiveevent)  and onRequest(requestevent ) are not working.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

Arduino code:

#include<Wire.h>                             //Library for I2C Communication functions
void setup() 

{
  
    Serial.begin(9600);                       //Begins Serial Communication at 9600 baud rate
    Wire.begin(0x42);                        //Begins I2C communication
   
} 

void loop() 
{
  
    Wire.beginTransmission(8);                // start transmit to slave (8)
    Wire.write("Hi");                               // sends 6 bytes value to slave
    Wire.endTransmission();                   // stop transmitting

    
    Wire.requestFrom(8,13);                   // request 13 bytes from slave (8)
    while (Wire.available())                   // slave may send less than requested
     { 
      char c = Wire.read();                   // receive a byte as character
      Serial.print(c);                              // print the character
     }
     
    Serial.println(); 
    
    delay(500);
   
}

ESP code:

#include<Wire.h>                            //Library for I2C Communication functions
void setup() 

{ 
 
    Serial.begin(9600);                     //Begins Serial Communication at 9600 baud rate
    Wire.begin(8);                          //Begins I2C communication with Slave Address as 8 
    Wire.onReceive(receiveEvent);           //Function call when Slave receives value from master
    Wire.onRequest(requestEvent);           //Function call when Master request value from Slave
}

void loop() 
{
  delay(500);
}

void receiveEvent (int howMany)       //This function is called when Slave receives value from the master
{
  
  while (Wire.available())                 // slave may send less than requested
   { 
     char c = Wire.read();                 // receive a byte as character
     Serial.print(c);                          // print the character
   }
  Serial.println();
   
}     

void requestEvent()                           //This Function is called when Master wants value from slave
{  
    Wire.write("Hello NodeMcu");           // sends 13 bytes value to master
}

### Debug Messages

```
Debug messages go here
```
please give high priority to this issue because this issue not yet resolved from years.

Tech-TX, 2020-04-02:
The slave code only works at 50KHz bus speed.  I'm slowly working on a complete re-write,  but it will likely have the same 50KHz limit.

tharun4477, 2020-04-02:
Then how to make it slave and use it with an arduino and an esp

closed 2020-04-02 by devyte

tharun4477, 2020-04-05:
hi @Tech-TX and @devyte  I even tried Wire.setClock(50000)  statement at the master side
still its not working
so if have any code that is currently working or any guidelines to make it work
please share it over here

thanks
tharun

rajhlinux, 2020-12-01:
I have the same problem, how did you made it work in slave mode? I have a WeMos D1 Mini (ESP8266).