esp8266/Arduino #7181: Can't make ANY HTTP GET request

ireun, 2020-04-01:
# Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:				ESP-12E
- Core Version:			2.6.3
- Development Env:		Visual Micro ( Ardunio IDE )
- Operating System:		Windows

### Settings in IDE
[image]

### Problem Description

I've tried the minimal sketch from the example, linked below, unfortunately i can't connect at all.. 

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266HTTPClient/examples/BasicHttpClient/BasicHttpClient.ino 

### Debug Messages

```
[HTTP] begin...
[HTTP-Client][begin] url: http://jigsaw.w3.org/HTTP/connection.html
[HTTP-Client][begin] host: jigsaw.w3.org port: 80 url: /HTTP/connection.html
[HTTP] GET...
[HTTP-Client][sendRequest] type: 'GET' redirCount: 0
[HTTP-Client] failed connect to jigsaw.w3.org:80
[HTTP-Client][returnError] error(-1): connection refused
[HTTP] GET... failed, error: connection refused
[HTTP-Client][end] tcp is closed
```

Tech-TX, 2020-04-02:
From the error,  you haven't edited the example.  You asked for nothing,  and got nothing in return.  
https://www.tutorialspoint.com/http/http_requests.htm
Google Is Your Friend

MarkyAD, 2020-04-02:
The error is "connection refused". Both the example and the url work here without issue, can't reproduce.

ireun, 2020-04-02:
Finally got it to work. Though I have no idea what was wrong there. Thanks and sorry. :)

closed 2020-04-02 by ireun