esp8266/Arduino #7184: WebUpdater Example mDNS Not Working

brendanmatkin, 2020-04-02:
### Basic Infos

- [ x ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ x ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ x ] I have tested that the issue is present in current master branch (aka latest git).
- [ x ] I have searched the issue tracker for a similar issue.
- [ x ] If there is a stack dump, I have decoded it.
- [ x ] I have filled out all fields below.

#### Platform

- Hardware:			ESP-12
- Core Version:			2.6.3
- Development Env:		Arduino IDE
- Operating System:		Windows

### Settings in IDE

- Module:  			Wemos D1 mini r2 & Feather Huzzah
- Flash Mode:			qio
- Flash Size: 		        4MB
- lwip Variant:                   v2 Lower Memory
- Reset Method:		
- Flash Frequency:	        40Mhz
- CPU Frequency:		80Mhz
- Upload Using:		OTA & SERIAL
- Upload Speed:               115200

### Problem Description

`ESP8266HTTPUpdateServer` example `WebUpdater` mDNS does not work. Everything else seems to work as expected. Tried on multiple boards both OTA and Serial uploads. Tried reinstalling board definitions. If I change `Wifi.mode(WIFI_AP_STA)` to `Wifi.mode(WIFI_STA)` then it works! Likely mDNS library does not expect basic service to work in AP mode (which makes sense).

I'm thinking it's just a legacy error in the example code, but it could be a problem with the mDNS library?

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

This is the slightly modified example code:
```cpp

#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <ESP8266WebServer.h>
#include <ESP8266mDNS.h>
#include <ESP8266HTTPUpdateServer.h>

#ifndef STASSID
#define STASSID "your-ssid"
#define STAPSK  "your-password"
#endif

const char* host = "esp8266-update";
const char* ssid = STASSID;
const char* password = STAPSK;

ESP8266WebServer httpServer(80);
ESP8266HTTPUpdateServer httpUpdater;

void setup(void) {

  Serial.begin(115200);
  Serial.println();
  Serial.println("Booting Sketch...");
  // WiFi.mode(WIFI_AP_STA);         // THIS DOES NOT WORK
  WiFi.mode(WIFI_STA);               // THIS WORKS FINE
  WiFi.begin(ssid, password); 

  while (WiFi.waitForConnectResult() != WL_CONNECTED) {
    WiFi.begin(ssid, password);
    Serial.println("WiFi failed, retrying.");
  }

  MDNS.begin(host);

  httpUpdater.setup(&httpServer);
  httpServer.begin();

  MDNS.addService("http", "tcp", 80);
  Serial.printf("HTTPUpdateServer ready! Open http://%s.local/update in your browser\n", host);
}

void loop(void) {
  httpServer.handleClient();
  MDNS.update();
}

```

devyte, 2020-04-02:
The MDNS lib global object works only over the station interface. If you want MDNS on the ap interface you have to use a 2nd MDNS object bound to the ap interface.
There is some work ongoing in the MDNS code to handle all interfaces internally, but that's a different story.
Closing due to working as intended.

brendanmatkin, 2020-04-02:
@devyte but the webupdater example code is still wrong then? If mDNS is working as intended then the example code is incorrect.

brendanmatkin, 2020-04-06:
@devyte to clarify - this issue is about the example code not working. I just wasn't sure if it was the example code that was written wrong, or if mDNS was behaving badly. So if MDNS is behaving correctly then the example code IS written wrong and should be changed. Can you please re-open the issue?

devyte, 2020-04-06:
For a moment you convinced me, but then I remembered in those examples in ap mode you don't need mdns, because you can access the http server with the ap side IP address.
Therefore, whether the examples are correct or not depends on your application. 
Case 1: hybrid current case. You access via domain name with MDNS from sta side, or fixed IP address from ap side. 
Case 2: simpler case, but requires static dhcp server on router ap. You don't need MDNS at all and can access from both sides with fixed IP address.
Case 3: what you want: access with the same domain name from either sta or ap sides. Requires a 2nd MDNS instance tied to the ap side. Requires increased example complexity, so we're not doing that, at least not in these examples. 
Closing again.

closed 2020-04-06 by devyte

brendanmatkin, 2020-04-06:
Ok! But just in case it's my bad explaining I'll try to clarify a bit more - if you run the example and try to access via domain from the STA side it does not resolve (when, [as in the example](https://github.com/esp8266/Arduino/blob/c548958f6ebbf372e3771e8a9012a5faab34d73b/libraries/ESP8266HTTPUpdateServer/examples/WebUpdater/WebUpdater.ino#L28), mode is set to `WiFi.mode(WIFI_AP_STA);`). From what I understand, this is case 1 BUT you _can't_ access via domain name with MDNS from sta side. It never resolves. Am I understanding correctly?

There is no documentation or comment that I could find anywhere that explains that you can only use MDNS if `WIFI.mode` is set to `WIFI_STA` for it to work, and `WIFI_AP_STA` will not work even if accessed through the STA side. Again, I might be missing something. 

If the example is working as intended then I propose to just remove the MDNS portion of the example entirely because that definitely doesn't work in the code as written. Maybe just print the STA IP to the console instead?

brendanmatkin, 2020-04-06:
p.s. I can do a PR if you agree

devyte, 2020-04-06:
That confusion is precisely why the MDNS handling was changed: the singleton will now always handle and respond on the sta side. If you want ap side handling, you need a 2nd MDNS object.
Your report here is for 2.6.3. You said:
> [ x ] I have tested that the issue is present in current master branch

But your comments make me think that is not true. I think the MDNS change was post 2.6.3 release, so you would have to test with current master, aka latest git, to see the effect. 

You can install latest git with the readthedocs instructions, or try @d-a-v 's alpha channel snapshot generator, which is much easier to install but I believe is a bit harder to update.

brendanmatkin, 2020-04-15:
Shoot you are totally right. I hadn't installed the git version properly and it was still running on 2.6.3. Reinstalled everything and tried again and it is now behaving as expected (2.7.0-dev as of April 15, 2020). I found the earlier issue discussions as well (e.g. #6975, #7042) - not sure why I couldn't find them before when searching past issues.

Sorry to waste your time and appreciate your work!