esp8266/Arduino #7187: client.connect() timeout when use IP-Number

Hans-Zwiesel, 2020-04-02:
Arduino IDE 1.8.12, Board ESP8266 2.6.3

#define RobonectIP  192,111,250,33 
#define RobonectPort  80 

if (WiFiClient.connect(IPAddress (RobonectIP), RobonectPort))

result in a timeout
--------------------------------------------------------------
const char* RobonectIP = "Robo-Max.fritz.box"; 
#define RobonectPort  80  

while (!client.connect(RobonectIP, RobonectPort)) 

connect time is about 25msec:
------------------------------------------------------------------

Is there a way to use the IP-number without a timeout?

Best regards,
Hans

devyte, 2020-04-02:
Issue template ignored. Please open a new issue and follow the issue template instructions, especially the part about MCVE.
Closing due to not compliant.

Hans-Zwiesel, 2020-04-03:
> 
> 
> Issue template ignored. Please open a new issue and follow the issue template instructions, especially the part about MCVE.
> Closing due to not compliant.

Hi,
i am sorry, but i am a German and my english is not so good to understand all the instructions.
I think the most important thing is, the issue is written and the author knows the bug.

closed 2020-04-03 by devyte