esp8266/Arduino #7191: git 2.26.0 issues

uzi18, 2020-04-04:
command will fail on git 2.26.0 (last working one is 2.25.2)

```
$ git clone --recursive --depth 1 --branch 2.6.3 https://github.com/esp8266/Arduino.git /home/users/linaja/.platformio/packages/_tmp_installing-bgoetgw6-package
git version 2.26.0
Cloning into '/home/users/linaja/.platformio/packages/_tmp_installing-bgoetgw6-package'...
remote: Enumerating objects: 1647, done.
remote: Counting objects: 100% (1647/1647), done.
remote: Compressing objects: 100% (1401/1401), done.
remote: Total 1647 (delta 201), reused 841 (delta 78), pack-reused 0
Receiving objects: 100% (1647/1647), 23.62 MiB | 291.00 KiB/s, done.
Resolving deltas: 100% (201/201), done.
Note: switching to '3d128e5c785cbe2096a0def394554d1d8091601d'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by switching back to a branch.                                                                                                                                                              
                                                                                                                                                                                                                                 
If you want to create a new branch to retain commits you create, you may                                                                                                                                                         
do so (now or later) by using -c with the switch command. Example:                                                                                                                                                               
                                                                                                                                                                                                                                 
  git switch -c <new-branch-name>                                                                                                                                                                                                
                                                                                                                                                                                                                                 
Or undo this operation with:                                                                                                                                                                                                     
                                                                                                                                                                                                                                 
  git switch -

Turn off this advice by setting config variable advice.detachedHead to false

Submodule 'libraries/ESP8266SdFat' (https://github.com/earlephilhower/ESP8266SdFat.git) registered for path 'libraries/ESP8266SdFat'
Submodule 'libraries/LittleFS/lib/littlefs' (https://github.com/ARMmbed/littlefs.git) registered for path 'libraries/LittleFS/lib/littlefs'
Submodule 'libraries/SoftwareSerial' (https://github.com/plerup/espsoftwareserial.git) registered for path 'libraries/SoftwareSerial'
Submodule 'tools/esptool' (https://github.com/espressif/esptool.git) registered for path 'tools/esptool'
Submodule 'tools/pyserial' (https://github.com/pyserial/pyserial.git) registered for path 'tools/pyserial'
Submodule 'lwip2' (https://github.com/d-a-v/esp82xx-nonos-linklayer.git) registered for path 'tools/sdk/lwip2/builder'
Submodule 'tools/sdk/ssl/bearssl' (https://github.com/earlephilhower/bearssl-esp8266.git) registered for path 'tools/sdk/ssl/bearssl'
Cloning into '/home/users/linaja/.platformio/packages/_tmp_installing-bgoetgw6-package/libraries/ESP8266SdFat'...
remote: Enumerating objects: 3534, done.        
remote: Total 3534 (delta 0), reused 0 (delta 0), pack-reused 3534        
Receiving objects: 100% (3534/3534), 10.16 MiB | 1.40 MiB/s, done.
Resolving deltas: 100% (1916/1916), done.
Cloning into '/home/users/linaja/.platformio/packages/_tmp_installing-bgoetgw6-package/libraries/LittleFS/lib/littlefs'...
remote: Enumerating objects: 2163, done.        
remote: Total 2163 (delta 0), reused 0 (delta 0), pack-reused 2163        
Receiving objects: 100% (2163/2163), 1.26 MiB | 1.40 MiB/s, done.
Resolving deltas: 100% (1508/1508), done.
Cloning into '/home/users/linaja/.platformio/packages/_tmp_installing-bgoetgw6-package/libraries/SoftwareSerial'...
remote: Enumerating objects: 69, done.        
remote: Counting objects: 100% (69/69), done.        
remote: Compressing objects: 100% (55/55), done.        
remote: Total 1855 (delta 29), reused 46 (delta 13), pack-reused 1786        
Receiving objects: 100% (1855/1855), 370.85 KiB | 1.10 MiB/s, done.
Resolving deltas: 100% (1076/1076), done.
Cloning into '/home/users/linaja/.platformio/packages/_tmp_installing-bgoetgw6-package/tools/esptool'...
remote: Enumerating objects: 55, done.        
remote: Counting objects: 100% (55/55), done.        
remote: Compressing objects: 100% (41/41), done.        
remote: Total 2189 (delta 20), reused 46 (delta 13), pack-reused 2134        
Receiving objects: 100% (2189/2189), 7.80 MiB | 1.37 MiB/s, done.
Resolving deltas: 100% (1351/1351), done.
Cloning into '/home/users/linaja/.platformio/packages/_tmp_installing-bgoetgw6-package/tools/pyserial'...
remote: Enumerating objects: 47, done.        
remote: Counting objects: 100% (47/47), done.        
remote: Compressing objects: 100% (47/47), done.        
remote: Total 5878 (delta 20), reused 13 (delta 0), pack-reused 5831        
Receiving objects: 100% (5878/5878), 1.65 MiB | 1.10 MiB/s, done.
Resolving deltas: 100% (4338/4338), done.
Cloning into '/home/users/linaja/.platformio/packages/_tmp_installing-bgoetgw6-package/tools/sdk/lwip2/builder'...
remote: Enumerating objects: 22, done.        
remote: Counting objects: 100% (22/22), done.        
remote: Compressing objects: 100% (15/15), done.        
remote: Total 924 (delta 8), reused 14 (delta 6), pack-reused 902        
Receiving objects: 100% (924/924), 481.52 KiB | 399.00 KiB/s, done.
Resolving deltas: 100% (543/543), done.
Cloning into '/home/users/linaja/.platformio/packages/_tmp_installing-bgoetgw6-package/tools/sdk/ssl/bearssl'...
remote: Enumerating objects: 86, done.        
remote: Counting objects: 100% (86/86), done.        
remote: Compressing objects: 100% (58/58), done.        
remote: Total 2820 (delta 46), reused 50 (delta 28), pack-reused 2734        
Receiving objects: 100% (2820/2820), 1.79 MiB | 1.52 MiB/s, done.
Resolving deltas: 100% (2225/2225), done.
Submodule path 'libraries/ESP8266SdFat': checked out 'b240d2231a117bbd89b79902eb54cae948ee2f42'
Submodule path 'libraries/LittleFS/lib/littlefs': checked out '6b65737715039ef92d348014316b575b52547019'
Submodule path 'libraries/SoftwareSerial': checked out '9da4d3729a57a181307bf9acf73473d052a38874'
Submodule path 'tools/esptool': checked out 'de30f21a222ec62f5a023dd955439b4f57702768'
Submodule path 'tools/pyserial': checked out 'c54c81d933b847458d465cd77e96cd702ff2e7be'
Submodule path 'tools/sdk/lwip2/builder': checked out '354887a25f83064dc0c795e11704190845812713'
Submodule 'lwip2-src' (https://git.savannah.nongnu.org/git/lwip.git) registered for path 'tools/sdk/lwip2/builder/lwip2-src'
Cloning into '/home/users/linaja/.platformio/packages/_tmp_installing-bgoetgw6-package/tools/sdk/lwip2/builder/lwip2-src'...
remote: Counting objects: 51280, done.        
remote: Compressing objects: 100% (12186/12186), done.        
remote: Total 51280 (delta 38697), reused 50936 (delta 38448)        
Receiving objects: 100% (51280/51280), 10.06 MiB | 1.05 MiB/s, done.
Resolving deltas: 100% (38697/38697), done.
error: Server does not allow request for unadvertised object 159e31b689577dbf69cf0683bbaffbd71fa5ee10
Fetched in submodule path 'tools/sdk/lwip2/builder/lwip2-src', but it did not contain 159e31b689577dbf69cf0683bbaffbd71fa5ee10. Direct fetching of that commit failed.
Submodule path 'tools/sdk/ssl/bearssl': checked out '89454af34e3e61ddfc9837f3da5a0bc8ed44c3aa'
Failed to recurse into submodule path 'tools/sdk/lwip2/builder'
```

earlephilhower, 2020-04-04:
I can repro it on a local GIT compile:
````
earle@server:/var/samba/space01/22/git$ PATH="/home/earle/usr/bin:$PATH" git --version
git version 2.26.0.106.g9fadedd637
earle@server:/var/samba/space01/22/git$ PATH="/home/earle/usr/bin:$PATH" git clone --recursive --depth 1 --branch 2.6.3 https://github.com/esp8266/Arduino.git
Cloning into 'Arduino'...
remote: Enumerating objects: 1647, done.
remote: Counting objects: 100% (1647/1647), done.
....
Submodule path 'tools/esptool': checked out 'de30f21a222ec62f5a023dd955439b4f57702768'
Submodule path 'tools/pyserial': checked out 'c54c81d933b847458d465cd77e96cd702ff2e7be'
Submodule path 'tools/sdk/lwip2/builder': checked out '354887a25f83064dc0c795e11704190845812713'
Submodule 'lwip2-src' (https://git.savannah.nongnu.org/git/lwip.git) registered for path 'tools/sdk/lwip2/builder/lwip2-src'
Cloning into '/var/samba/space01/22/git/Arduino/tools/sdk/lwip2/builder/lwip2-src'...
remote: Counting objects: 51280, done.        
remote: Compressing objects: 100% (12120/12120), done.        
remote: Total 51280 (delta 38696), reused 51041 (delta 38514)        
Receiving objects: 100% (51280/51280), 10.07 MiB | 6.11 MiB/s, done.
Resolving deltas: 100% (38696/38696), done.
error: Server does not allow request for unadvertised object 159e31b689577dbf69cf0683bbaffbd71fa5ee10
Fetched in submodule path 'tools/sdk/lwip2/builder/lwip2-src', but it did not contain 159e31b689577dbf69cf0683bbaffbd71fa5ee10. Direct fetching of that commit failed.
Submodule path 'tools/sdk/ssl/bearssl': checked out '89454af34e3e61ddfc9837f3da5a0bc8ed44c3aa'
Failed to recurse into submodule path 'tools/sdk/lwip2/builder'
````

That said, the repo being clones is hosted in GitHub and has no issues elsewhere, which leads me to believe you've found a (backwards compat?) bug in git itself.  There's nothing we can do here, but I suggest checking out https://git-scm.com/community and filing something on the GIT bug mailing list.

d-a-v, 2020-04-04:
Side note, this issue should not happen
- with master (lwIP is not sub-submoduled anymore)
- or without `--recursive` (then this is needed: `git submodule update --init` also without `--recursive`)

other that that it seems @earlephilhower is right.

uzi18, 2020-04-04:
maybe it is more like platformio specific bug and @ivankravets can point someone here ?

d-a-v, 2020-04-04:
@uzi18 like you said, older version of git is working (I tried with 2.20.1 too).
The only thing to do like @earlephilhower proposed is to tell git staff about it:
```shell
$ git clone --recursive --depth 1 --branch 2.6.3 https://github.com/esp8266/Arduino.git 
```
is not working with git 2.26.0.

uzi18, 2020-04-04:
ouch, have wrongly understand you @earlephilhower sorry, my issue is just a hint 
something goes wrong and we found more/less where it was introduced

devyte, 2020-04-04:
I don't see an issue in our core here. I agree with previous comments: this should be reported at the git tracker. @uzi18 you did good work figuring out that it's a git version problem, thank you for that. I suggest pursuing as explained above. 
Closing due to not a core issue.

closed 2020-04-04 by devyte