esp8266/Arduino #7198: EEPROM Library on esp8285 only commits once

joaolrc, 2020-04-07:
Hi all!
I have been using the esp family a lot but more the esp32. Nevertheless, recently i am reprogramming a SONOFF 4CH PRO which has an ESP8285 (esp8266 with 1Mb flash) and from my experiences, doing EEPROM.commit() only works at first time it is issued from boot.
Code 

```arduino
#include <Arduino.h>
#include <SonoffProPins.h>
#include <EEPROM.h>

#define MYTESTINT   "-69"
#define MYTESTINT2   "79"

void write_string(EEPROMClass eep, uint16_t add, String data)
{
  int _size = data.length();
  for (int i = 0; i < _size; i++)
  {
    eep.write(add + i, data[i]);
  }
  eep.write(add + _size, '\0'); //Add termination null character for String Data
}

String read_String(EEPROMClass eep, uint16_t add)
{
  char data[100]; //Max 100 Bytes
  int len = 0;
  unsigned char k;
  k = eep.read(add);
  while (k != '\0' && len < 500) //Read until null character
  {
    k = eep.read(add + len);
    data[len] = k;
    len++;
  }
  data[len] = '\0';
  return String(data);
}

void clearEEPROM(EEPROMClass eep)
{
  Serial.printf("Will clean %u bytes", eep.length());
  for (uint16_t i = 0; i < eep.length(); i++)
  {
    eep.put(i, '\0');
  }
}

EEPROMClass myEEPROM;

//Global Variables
const char *swVersion = "0.0.1";
unsigned long prevMillis = 0;
unsigned long intervalToPrint = 3000;

void setup(){
  Serial.begin(115200);
  Serial.printf("Starting Sonoff4CHPro testing V%s\n", swVersion);
  myEEPROM.begin(512);
  delay(100);
  Serial.printf("READ0 %s\n", read_String(myEEPROM, 0).c_str());
 
  write_string(myEEPROM, 0, "123456");
  myEEPROM.commit();
  myEEPROM.begin(512);
  Serial.printf("READ1 %s\n", read_String(myEEPROM, 0).c_str());

  write_string(myEEPROM, 0, "54321");
  myEEPROM.commit();
  Serial.printf("READ2 %s\n", read_String(myEEPROM, 0).c_str());

  write_string(myEEPROM, 0, "69696969");
  myEEPROM.commit();
  Serial.printf("READ3 %s\n", read_String(myEEPROM, 0).c_str());
 
}

void loop()
{
  unsigned long aux_time_now = millis();
  if (aux_time_now - prevMillis >= intervalToPrint)
  {
    Serial.println("I AM HERE\n");

    prevMillis = aux_time_now;
  }
}

```
So i let it run and reset in the middle. The READ0 should be 69696969. And its clearly not.
the output is below. The garbage is because of reset:
```
Starting Sonoff4CHPro testing V0.0.1
READ0 123456
READ1 123456
READ2 54321
READ3 69696969
I AM HERE

rl␀l�r␂$␒�n␌␌␌�␌l�␌b|��␂␇␒�r␒b�␌b␄�nn␂lnn␒b␌b␜p�$b␎lrlp�n0␂␌␌�␌l␌�␂␌␌␌b␌n�nl�l�␌b��nn'l�␄l`␂�␒␒nn␌l`␂␎␂nr���n␌␌␐␂l`␂␎r��n␌␌␐␂l`␂�r␌␒�␌l�␇�`␂��nb�
SDK:2.2.2-dev(38a443e)/Core:2.6.3=20603000/lwIP:STABLE-2_1_2_RELEASE/glue:1.2-16-ge23a07e/BearSSL:89454af
Starting Sonoff4CHPro testing V0.0.1
READ0 123456
READ1 123456
READ2 54321
READ3 69696969
I AM HERE
```

Am i doing something wrong. Help is much appreciated. Thanks!

joaolrc, 2020-04-07:
I put another myEEPROM.begin(512) just to see if that would made some diference but the result is exactly the same if it isnt there

devyte, 2020-04-07:
You're passing the class to your functions by value, and the class isn't implemented for that, so no wonder. 
You're not checking the return value of ::commit(), so you have no idea whether it succeeded or not. 
You're declaring your own eeprom object(s) instead of using the global singleton, which should only be done when using the no globals macro. 
And issue template ignored.
Closing due to user error and due to not compliant.

closed 2020-04-07 by devyte

joaolrc, 2020-04-07:
Thank you very much @devyte . Real good response time. Wasnt expecting. Sorry for not be using the template but just wanted to go straight to the point. Yeah i should be passing by reference, you are totally right **but that does not explain how it works well with the first value stored each boot.** I wanted to have 2 memory spaces so i would like to use my own instances but i did not knew i had to use a macro. I come from the Preferences library for esp32 which handles this logic natively