esp8266/Arduino #7200: Wrong date in time.h

nimaltd, 2020-04-08:
----------------------------- Delete above -----------------------------

### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12]
- Core Version:			[last]
- Development Env:		[Platformio]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Generic ESP8266 Module]
- Flash Mode:			[qio]
- Flash Size: 		[4MB]
- lwip Variant:   [v2 Lower Memory]
- Reset Method:		[nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz|]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200|other] (serial upload only)

### Problem Description

I got wrong month . 

bool wifi_getTime(tm* t)
{    
    time_t now = time(nullptr);
    if (now < 8 * 3600 * 2)
    {
        return false;
    }
    gmtime_r(&now, t);
    return true;
}

if(wifi_getTime(&currentTime))
{
            Serial.printf("\r\n+DATE:%02d,%02d,%02d,%02d\r\n",currentTime.tm_year - 100, 
            currentTime.tm_mon, currentTime.tm_mday, currentTime.tm_wday);

}

this is answer
+DATE:20,03,08,03
but true is 
+DATE:20,04,08,03

nimaltd, 2020-04-08:
my server
configTime(0, 0, "pool.ntp.org", "time.nist.gov");

nimaltd, 2020-04-08:
I found it . month start from 0. I must be add 1.

closed 2020-04-08 by nimaltd

jandrassy, 2020-04-08:
https://www.tutorialspoint.com/c_standard_library/time_h.htm