esp8266/Arduino #7201: error: array used as initializer

redPrint7, 2020-04-08:
----------------------------- Delete below -----------------------------
### Basic Infos

- [ x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:				[ESP-12] [NodeMCU v09/v1]
- Core Version:			[2.6.3]
- Development Env:		[Arduino IDE 8.1.12]
- Operating System:		[Windows|MacOS]

### Settings in IDE
Below does not matter its a toolchain problem
- Module:  			[Generic ESP8266 Module|Nodemcu|]
- Flash Mode:			[dout]
- Flash Size: 		[4MB/1MB]
- lwip Variant:   [v1.4|v2 Lower Memory|Higher Bandwidth]
- Reset Method:		[ck|nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz|160MHz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200|other] (serial upload only)

### Problem Description
The foliowing (condensed for reproduction) code compiles fine on all Arduinos (UNO/pro/Nano...) avr 1.8.2 and the ESP32 core 1.04. It fails to compile on the esp8266 core. It seems to be a compiler related tool chain problem.

Setup: Basic *.ino file  with include to the header file and the following header with a simple class definition arraytest.h

#ifndef ARRAYTEST_H_
#define ARRAYTEST_H_
/******************************************************************************************************/
#define MAX_A_FUNCTIONS 29
/******************************************************************************************************/
/** @brief Data associated to each entry in the A roster.*/
class A_Manager {
  private:
    uint8_t _aKey ;         /**< The numeric place (0-10 = slot number) of the part on the roster (=unique ID) */
    const char* _typeField;                            /** part - Added for more flexibility */
    const  char* _subTypeField;                 /** Added for more flexibility for part -  active for the selected part */
    char _functionMap[MAX_A_FUNCTIONS + 1] = ""; /** < Current status of the functions: 0 for Off, 1 for On - Length of array is MAX_A_FUNCTIONS */

  public:
    char functionMap[MAX_A_FUNCTIONS + 1] = "";
    A_Manager(uint8_t , const char* ,  const char*,  char []);
    ~A_Manager();
    uint8_t getAKey() const ;
};
/***************************************** CPP part *************************************************/

A_Manager::A_Manager(uint8_t aKey, const char* typeField = "APart", const char* subTypeField = "ASubPart",  char functionMap[MAX_A_FUNCTIONS] = {'\0'})
{
  aKey = _aKey;
  typeField = _typeField;
  subTypeField = _subTypeField;
  strcpy (functionMap, _functionMap);
  for (uint8_t functionID = 0; functionID <= MAX_A_FUNCTIONS; functionID++) {
    functionMap[functionID] = '0';
  }
}

A_Manager::~A_Manager() {}

uint8_t A_Manager::getAKey() const {
  return _aKey;
}

#endif /* ARRAYTEST_H_ */
### Debug Messages
In file included from C:\Dev\Arduino\ESP8266_Dev\_Examples\error_array_class\error_array.ino:1:0:

c:\XXXX\arduino_build_XXXX\sketch\arraytest.h: In constructor 'A_Manager::A_Manager(uint8_t, const char*, const char*, char*)':

arraytest.h:22:151: error: array used as initializer

 A_Manager::A_Manager(uint8_t aKey, const char* typeField = "APart", const char* subTypeField = "ASubPart",  char functionMap[MAX_A_FUNCTIONS] = {'\0'})

arraytest.h:22:151: error: array used as initializer

exit status 1
array used as initializer

Seems like a similar error the gnu compilers had until 5.03
c++11 is activated in platform.txt

There is no option to change the program as it works in production on all other named platforms for a year.
If you need more Info -glad to help

d-a-v, 2020-04-08:
This will be fixed by #6294 (we are currently using g++-4.8)

closed 2020-07-07 by earlephilhower

jamesmyatt, 2020-08-26:
In case anyone else has this problem, the simplest workaround is here: https://stackoverflow.com/a/35414564/21322