esp8266/Arduino #7202: mklittlefs support for 32 bit Linux is missing in version 2.6.0 and later

andrews999, 2020-04-09:
### Basic Infos

- [ y] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [y ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [y ] I have tested that the issue is present in current master branch (aka latest git).
- [y ] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [y ] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12|ESP-01|ESP-07|ESP8285 device|other]
- Core Version:			[2.6.3]
- Development Env:		[Arduino IDE]
- Operating System:		[Linux 32 bit]

### Settings in IDE

- Module:  			[Generic ESP8266 Module|Wemos D1 mini r2|Nodemcu|other]
- Flash Mode:			[qio|dio|other]
- Flash Size: 		[4MB/1MB]
- lwip Variant:   [v1.4|v2 Lower Memory|Higher Bandwidth]
- Reset Method:		[ck|nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz|160MHz]
- Upload Using:		[OTA|SERIAL]
- Upload Speed:   [115200|other] (serial upload only)

### Problem Description

When mklittlefs was introduced in version 2.6.0 the lines to support 32 bit Linux were missing from package_esp8266com_index.json. They can be manually added in a local file as the files exist in https://github.com/earlephilhower/esp-quick-toolchain/releases/download/2.5.0-4/i686-linux-gnu.mklittlefs-7f77f2b.1563313032.tar.gz.
Can the build be changed to include support for 32 bit Linux?

d-a-v, 2020-04-09:
This has been fixed since 2.6.3, you need to wait for release 2.7.0 or use the [beta (0.0.1)](https://d-a-v.github.io) version.

andrews999, 2020-04-10:
Thanks for your help.

closed 2020-04-10 by andrews999