esp8266/Arduino #7210: mDNS problem

BbIKTOP, 2020-04-12:
I've been told by David to open an issue about it, so here it is:

    #include <Arduino.h>
    
    #include <ESP8266WiFi.h>
    #include <ESP8266mDNS.h>
    
    
    char *wifiNet = "mywifi";
    char *wifiPassword = "mywifipasswd";
    
    char *apName = "box";
    char *apPassword = "box";
    char *apIp = "10.1.1.1";
    char *apMask = "255.255.255.0";
    
    
    void setup()
    {
      int res;
    
      Serial.begin(115200);
      Serial.println("\nStarting");
    
      WiFi.persistent(false);
      WiFi.disconnect();
    
      WiFi.mode(WIFI_AP_STA);
    
      IPAddress localIp;
      IPAddress localMask;
      IPAddress localGw;
    
      localIp.fromString(apIp);
      localMask.fromString(apMask);
      localGw.fromString(apIp);
    
      res = WiFi.softAPConfig(localIp, localGw, localMask);
      Serial.printf("After softAPConfig: %d\n", res);
    
      res = WiFi.softAP(apName, apPassword, 2, false, 20);
      Serial.printf("After softAP: %d\n", res);
    
      Serial.printf("After softAP IP: %s\n", WiFi.softAPIP().toString().c_str());
    
      WiFi.hostname(apName);
      WiFi.setAutoReconnect(true);
      WiFi.begin();
    
      if (!MDNS.begin(apName))
      {
        Serial.printf("Cannot start mDNS responder\n");
      }
      else
      {
        Serial.printf("mDNS responder started\n");
      }
      MDNS.addService("http", "tcp", 80);
    
      Serial.println("Started");
    }
    
    void loop()
    {
      static int counter = 0;
      static int counter2 = 0;
    
      MDNS.update();
    
      if (counter++ > 10)
      {
        static int lastStatus = WiFi.status();
    
        if (WiFi.status() == WL_IDLE_STATUS)
        {
          Serial.println("Reconnecting...");
          WiFi.begin(wifiNet, wifiPassword);
          WiFi.reconnect();
        }
    
        if (WiFi.status() == WL_CONNECTED && lastStatus != WL_CONNECTED)
        {
          Serial.printf("Connected, IP is: %s\n", WiFi.localIP().toString().c_str());
          counter = 0;
    
          MDNS.end();
          if (!MDNS.begin(apName))
          {
            Serial.printf("Cannot restart mDNS responder\n");
          }
          else
          {
            Serial.printf("mDNS responder restarted\n");
          }
          MDNS.addService("http", "tcp", 80);
        }
        lastStatus = WiFi.status();
      }
    
      if (counter2++ > 10)
      {
        int sc = MDNS.queryService("http", "tcp");
        Serial.printf("MDNS: %d services found\n", sc);
        counter2 = 0;
      }
    
      delay(1000);
    }

Produces

    Starting
    After softAPConfig: 1
    After softAP: 0
    After softAP IP: 10.1.1.1
    mDNS responder started
    Started
    Reconnecting...
    MDNS: 0 services found
    Connected, IP is: 172.16.5.79
    mDNS responder restarted
    MDNS: 0 services found
    MDNS: 0 services found
    MDNS: 0 services found
    MDNS: 0 services found
    MDNS: 0 services found
    MDNS: 0 services found
    MDNS: 0 services found
    MDNS: 0 services found
    MDNS: 0 services found
    MDNS: 0 services found

And no services are visible in service discovery like `dns-sd -B`

    viktor@Work:~ $ telnet box.local 80
    box.local: nodename nor servname provided, or not known
    viktor@Work:~ $

Versions:

    $ platformio update
    Updating tool-unity                      @ 1.20500.200322 [Up-to-date]
    Updating contrib-piohome                 @ 3.1.1          [Up-to-date]
    Updating tool-scons                      @ 2.20501.191222 [Up-to-date]
    Updating contrib-pysite                  @ 2.27.191017    [Up-to-date]
    
    Platform Manager
    ================
    Platform Espressif 8266
    --------
    Updating espressif8266                   @ 2.4.0          [Up-to-date]
    Updating tool-esptoolpy                  @ 1.20800.0      [Up-to-date]
    Updating tool-mkspiffs                   @ 1.200.0        [Up-to-date]
    Updating tool-esptool                    @ 1.413.0        [Up-to-date]
    Updating framework-arduinoespressif8266  @ 3.20603.200130 [Up-to-date]
    Updating toolchain-xtensa                @ 2.40802.191122 [Up-to-date]
    
    
    Library Manager
    ===============
    Library Storage: /Users/viktor/.platformio/lib
    Updating ESPAsyncTCP                     @ 1.2.2          [Up-to-date]

d-a-v, 2020-04-12:
Well now I read it through, the issue is the following:

Starting from 2.6.3, it is preferred to give mDNS which interface  it is to run on.
In 2.6.3, default interface was AP when both interfaces where up (AP+STA).
In current master and future 2.7.0, STA is used.

In both case, giving the interface you want works:

```
if (!MDNS.begin(apName, WiFi.localIP())) // start on STA
```

BbIKTOP, 2020-04-14:
Yes, thank you. I have checked this lwip implementation and found that iit cannot be done automatically. So, second instance.

To be able to create second instance you'd probably need to replace
lib/ESP8266mDNS/src/LEAmDNS_Helpers.cpp

    if (m_pUDPContext->listen(IP4_ADDR_ANY, DNS_MQUERY_PORT))

to something like 

    if (m_pUDPContext->listen(ip_2_ip4(&m_netif->ip_addr), DNS_MQUERY_PORT))

To prevent err -8 ERR_USE

d-a-v, 2020-06-21:
@BbIKTOP This issue is still opened.
@LaborEtArs provided us a new version of mDNS, which is currently in an adaptation process, with @hreintke, to make it interface-agnostic. It is not yet a PR but you can try it [there](https://github.com/Labor-Et-Ars/ESP8266-Arduino/tree/LEAmDNS2x).
[You will need to](https://github.com/Labor-Et-Ars/ESP8266-Arduino/blob/7e739ff5760503aa36a17146241175711fc9a41e/libraries/ESP8266mDNS/src/ESP8266mDNS.h#L45-L51)
```
#define MDNS2_EXPERIMENTAL_V1COMPAT
#include <ESP866mDNS.h>
```

devyte, 2020-07-06:
The updated MDNS code is still showing issues and requires more work. Per internal discussions, pushing back to v3.

d-a-v, 2020-11-09:
mDNS in current master should solve this issue. Please open a new one if that's not the case.

closed 2020-11-09 by d-a-v