esp8266/Arduino #7214: No way to check, MDNSResponder::begin has been called or not

BbIKTOP, 2020-04-14:
There's currently no way to check, is MDNSResponder::begin already called. There shall be a method that returns true if begin has been called and false if not (or close/end has been called). There shall be a bool member field default to false, set to true if begin succeeded and back to false when close/end succedded, and some method like bool isBegan(void) to check it

d-a-v, 2020-04-15:
There are two solutions :
- `MDNSResponder::begin()` returns a `bool`.
  When one needs mDNS, calling it the first time will tell if it has successfully started.
  One can retain the calling result in a personal  `bool` variable to later remember if mDNS start method has already been successfully called.
- Your question seems to suggest you'd like to call it several times and anytime.
  That is possible. First call will return `true` and all subsequent calls will return `false`. One can also ignore this return value.

Now f you read the code, you'll see that there is an internal variable holding that state. You are welcome to *propose a pull request* for a method `bool ::isRunning()`.

closed 2020-04-15 by d-a-v

BbIKTOP, 2020-04-15:
Evaluating begin return value wouldn’t help because it’s not possible to distinguish why was it failed - is there any real fatal error or is it already running. That’s why all kernel calls and standard lib calls either return error code or throw an exception clearly describing the reason.
Sorry, haven’t found this variable, could you give me a hint please?
I can create pull request, do i need to create separate requests for each issue?

d-a-v, 2020-04-15:
[Here is the internal test](https://github.com/esp8266/Arduino/blob/e1789ddf0c73ff00cdca104e420ead5bc8bc016f/libraries/ESP8266mDNS/src/LEAmDNS.cpp#L101). 
It is likely that an additional public `bool` accessor with this test should be sufficient for your needs. It would not need more variables.

Separate pull request are always preferred when they are independent because they need to be commented accurately.

PR #7219 (2020-04-17): Add isRunning getter to check is ::begin already called
Fixes #7214