esp8266/Arduino #7226: Heap problem with WPS - WiFi.beginWPSConfig()

Socket2000, 2020-04-18:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12|]
- Core Version:			[2.6.3]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows 7]

### Settings in IDE

- Module:  			[Generic ESP8266 Module]
- Flash Mode:			[qio]
- Flash Size: 		[4MB/1MB]
- lwip Variant:   [v2 Lower Memory]
- Reset Method:		[nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz|160MHz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200|

### Problem Description

It seems to be an issue with the use of WiFi.beginWPSConfig() when the heap has a free memory lower than a certain amount. After many test i found that in my large code, that is using many ram and left about 10000..14000 bytes free, a call to WiFi.beginWPSConfig() never connect to router. It find the ssid but not the shared password so it fails to connect. After resizing String allocation escluding some, obtaining about 20000..23000 bytes free, the function success.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

#include <ESP8266WiFi.h>

boolean wpsOK;
byte buffer[35000]; // used to fill memory
void setup(void)
{ 
  Serial.begin(115200);
  Serial.println();
  pinMode(5, INPUT);  // it start the wps search
  for (int x=0; x<35000; x++) buffer[x]=120; // fill buffer
}

void loop() {
  delay(500);
  Serial.println(String(ESP.getFreeHeap()));
  if (digitalRead(5)==LOW) {
    Serial.printf("Wi-Fi mode set to WIFI_STA %s\n", WiFi.mode(WIFI_STA) ? "" : "Failed!");
    Serial.print("Begin WPS (press WPS button on your router) ... ");
    wpsOK=WiFi.beginWPSConfig();
    Serial.println(wpsOK ? "Success" : "Failed");

    Serial.print("ssid: ");
    Serial.println(WiFi.SSID().c_str());
    Serial.print("password: ");
    Serial.println(WiFi.psk().c_str());
    
    Serial.println();
    Serial.print("Connected, IP address: ");
    Serial.println(WiFi.localIP());
  } else {
    if (WiFi.status() != WL_CONNECTED) {
      Serial.println("Press button to WPS...");
      delay(500);
    } else {
      Serial.print("Connected, IP address: ");
      Serial.println(WiFi.localIP());
    }
  }
}
### Debug Messages

no message, no crash

devyte, 2020-04-18:
If you build with the OOM debug option enabled, do you get a console message?

Socket2000, 2020-04-19:
reconnect
state: 2 -> 0 (0)
wps discover [SHARP]
scandone
WPS: neg start
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 1
cnt 
process pubkey start
:oom(512)@libtommath.h:2543
state: 5 -> 2 (1c0)
rm 0

devyte, 2020-04-19:
Yeah, the oom means you're out of memory. Pubkey hints at an encrypted process, so it makes sense. 
You need the free heap. 
BTW, our WPS is a paper thin wrapper around Espressif's sdk api, see here:
https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/src/ESP8266WiFiSTA-WPS.cpp
Closing because nothing to do in our core.

closed 2020-04-19 by devyte