esp8266/Arduino #7238: SPIFF file fails to clear using "W" or "W+" using SPIFFS.Open

Buzzardbait, 2020-04-22:
### Basic Infos

- [ ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [ ] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ ] I have filled out all fields below.

#### Platform

- Hardware:					NodeMCU(ESP-12E Module)
- Core Version:			[latest git hash or date]
- Development Env:		Arduino IDE
- Operating System:		Windows

### Settings in IDE

- Module:  			NodeMCU(ESP-12E Module)
- Flash Mode:			[qio|dio|other]
- Flash Size: 		[4MB/1MB]
- lwip Variant:   [v1.4|v2 Lower Memory|Higher Bandwidth]
- Reset Method:		|nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		80Mhz|
- Upload Using:		SERIAL
- Upload Speed:   [115200|(serial upload only)

### Problem Description
Using NodeMCU to acquire time and temperature data and record to a .csv SPIFFS file. The file fills up with previous data and I want the option to clear the file during Setup so I test a boolean and if true I execute the following code to clear the file:

if (ClrTempCSVonStart) {  // Clear the temp.csv file.... This doesn't work. Why?
      File tempLog = SPIFFS.open("/temp.csv", "W+");
      delay(1000);
      tempLog.close();
     /* 
      *  Creates an empty file for writing. 
      *  If a file with the same name already exists, 
      *  its content is erased and the file is considered 
      *  as a new empty file.
     */
     Serial.println("temp.csv has been cleared"); 
    }
The code is executed and everything continues normally but the file has not been cleared.
According to the Arduino Core Documentation the "W" or "W+" mode flags should have caused the file to have been emptied.
Currently, the only way I can clear this file is to do a sketch data upload. I suppose I could do a record by record erase until reaching the EOF but that would not be very elegant. :-)
Why doesn't this work?

devyte, 2020-04-22:
The issue template above is included, but not filled out, and the M**C**VE is far from **C**omplete.
Please edit the original issue and complete the missing info to have this considered, or close and open a new one.

Buzzardbait, 2020-04-22:
Can't even edit most of the questions in the post. Half of the info requested is unknown and/or irrelevant. Thanks for nothing. I'll go elsewhere to ask for help.

devyte, 2020-04-22:
BTW, SPIFFS will be deprecated and eventually removed. You might want to migrate to LittleFS sooner rather than later.

closed 2020-04-22 by devyte

Buzzardbait, 2020-04-22:
Thank you for the tip.

Buzzardbait, 2020-05-10:
BTW, it was my typo that caused the issue. I used upper case W in stead of w. Works fine now. Or it did, now spiffs fail to upload on any NodeMCU while sketches upload just fine. Will the wonders of technology never end?