esp8266/Arduino #7242: ESP8622 ESP-201 esptool.FatalError

rserroyen, 2020-04-23:
#### Platform

- Hardware:					[ESP-201]
- Core Version:			[]
- Development Env:		[Arduino IDE|]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Generic ESP8266 Module]
- Flash Mode:			[qio]
- Flash Size: 		[1MB]
- lwip Variant:   [v2 Lower Memory]
- Reset Method:		[ck|nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[OTA|SERIAL]
- Upload Speed:   [115200]

### Problem Description

Hello

I use an ESP8622 [ESP-201](https://www.14core.com/wiring-and-flashing-the-esp8266-esp201-wifi-transceiver-with-arduino/) but I can't program it!

I use a USB CH340 to TTL and mounting https://www.hackster.io/Hulkco/esp-201-esp8266-temperature-humidity-web-server-9f2fa9

I tested the WiFiWebServer but the upload does not work

I tested different speed values but without success

why

thank you

I had a stroke! I do not write well: oops: !!

### Debug Messages

```
Executable segment sizes:

IROM   : 280860          - code in flash         (default or ICACHE_FLASH_ATTR) 

IRAM   : 27692   / 32768 - code in IRAM          (ICACHE_RAM_ATTR, ISRs...) 

DATA   : 1264  )         - initialized variables (global, static) in RAM/HEAP 

RODATA : 1008  ) / 81920 - constants             (global, static) in RAM/HEAP 

BSS    : 25352 )         - zeroed variables      (global, static) in RAM/HEAP 

Le croquis utilise 310824 octets (71%) de l'espace de stockage de programmes. Le maximum est de 434160 octets.
Les variables globales utilisent 27624 octets (33%) de mémoire dynamique, ce qui laisse 54296 octets pour les variables locales. Le maximum est de 81920 octets.
esptool.py v2.8
Serial port COM6
Connecting....
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: 5c:cf:7f:01:eb:82
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 1MB
Flash params set to 0x0220
Compressed 314976 bytes to 226191...

Writing at 0x00000000... (7 %)
Writing at 0x00004000... (14 %)
Writing at 0x00008000... (21 %)Traceback (most recent call last):
  File "C:\Users\rserr\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3/tools/upload.py", line 65, in <module>
    esptool.main(cmdline)
  File "C:/Users/rserr/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/tools/esptool\esptool.py", line 2938, in main
    operation_func(esp, args)
  File "C:/Users/rserr/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/tools/esptool\esptool.py", line 2374, in write_flash
    esp.flash_defl_block(block, seq, timeout=DEFAULT_TIMEOUT * ratio * 2)
  File "C:/Users/rserr/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/tools/esptool\esptool.py", line 104, in inner
    return func(*args, **kwargs)
  File "C:/Users/rserr/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/tools/esptool\esptool.py", line 672, in flash_defl_block
    self.ESP_FLASH_DEFL_DATA, struct.pack('<IIII', len(data), seq, 0, 0) + data, self.checksum(data), timeout=timeout)
  File "C:/Users/rserr/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/tools/esptool\esptool.py", line 369, in check_command
    val, data = self.command(op, data, chk, timeout=timeout)
  File "C:/Users/rserr/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/tools/esptool\esptool.py", line 347, in command
    p = self.read()
  File "C:/Users/rserr/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/tools/esptool\esptool.py", line 292, in read
    return next(self._slip_reader)
  File "C:/Users/rserr/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/tools/esptool\esptool.py", line 2045, in slip_reader
    raise FatalError("Timed out waiting for packet %s" % waiting_for)
esptool.FatalError: Timed out waiting for packet header
esptool.FatalError: Timed out waiting for packet header

```

rserroyen, 2020-04-24:
Hello
Oddly sometimes he writes everything but same esptool.FatalError: Timed out waiting for packet header

thank

### Debug Messages

```
Configuring flash size...
Auto-detected Flash size: 1MB
Flash params set to 0x0020
Compressed 314000 bytes to 225581...

Writing at 0x00000000... (7 %)
Writing at 0x00004000... (14 %)
Writing at 0x00008000... (21 %)
Writing at 0x0000c000... (28 %)
Writing at 0x00010000... (35 %)
Writing at 0x00014000... (42 %)
Writing at 0x00018000... (50 %)
Writing at 0x0001c000... (57 %)
Writing at 0x00020000... (64 %)
Writing at 0x00024000... (71 %)
Writing at 0x00028000... (78 %)
Writing at 0x0002c000... (85 %)
Writing at 0x00030000... (92 %)
Writing at 0x00034000... (100 %)
Wrote 314000 bytes (225581 compressed) at 0x00000000 in 19.9 seconds (effective 126.0 kbit/s)...
Traceback (most recent call last):
  File "C:\Users\rserr\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.2/tools/upload.py", line 65, in <module>
    esptool.main(cmdline)
  File "C:/Users/rserr/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.6.2/tools/esptool\esptool.py", line 2938, in main
    operation_func(esp, args)
  File "C:/Users/rserr/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.6.2/tools/esptool\esptool.py", line 2398, in write_flash
    res = esp.flash_md5sum(address, uncsize)
  File "C:/Users/rserr/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.6.2/tools/esptool\esptool.py", line 104, in inner
    return func(*args, **kwargs)
  File "C:/Users/rserr/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.6.2/tools/esptool\esptool.py", line 691, in flash_md5sum
    timeout=timeout)
  File "C:/Users/rserr/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.6.2/tools/esptool\esptool.py", line 369, in check_command
    val, data = self.command(op, data, chk, timeout=timeout)
  File "C:/Users/rserr/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.6.2/tools/esptool\esptool.py", line 347, in command
    p = self.read()
  File "C:/Users/rserr/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.6.2/tools/esptool\esptool.py", line 292, in read
    return next(self._slip_reader)
  File "C:/Users/rserr/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.6.2/tools/esptool\esptool.py", line 2045, in slip_reader
    raise FatalError("Timed out waiting for packet %s" % waiting_for)
esptool.FatalError: Timed out waiting for packet header
esptool.FatalError: Timed out waiting for packet head
```

devyte, 2020-04-24:
The previous log shows core 2.6.2, while the log before that shows core 2.6.3. So which core are you using? Are you sure your installation is correct?

rserroyen, 2020-04-25:
> So which core are you using? 

now 2.6.3
whatever version -> same results : esptool.FatalError: Timed out waiting for packet header

> Are you sure your installation is correct?

I used ARDUINO 1.8.12 -> Tools> Board> Boards Manager> ESP8266 by ESP8266 Community to install the Board.
how to know if the installation is good ??

thank

devyte, 2020-04-25:
What about the ch340 driver? I seem to remember several drivers out there, and some had issues while others worked.

rserroyen, 2020-04-26:
> What about the ch340 driver?

ch341s64.sys 3.50 from www.winchiphead.com
final version

I will investigate ch340 driver

Thank

rserroyen, 2020-04-29:
I did not put jumper on VCC and 3.3

Now he's works

Thank you

closed 2020-04-29 by rserroyen