esp8266/Arduino #7243: When the same code works fine on NodeMCU-12E but falls on Wemos D1 Pro

woodlist, 2020-04-23:
#### Platform

- Hardware:		                [Wemos D1 Mini Pro] Or may be a clone
- Core Version:			[ets Jan  8 2013]
- Development Env:		[Arduino IDE Ver. 1.8.9 and 1.8.13beta]
- Operating System:		[Windows 7]

### Settings in IDE

- Module:  			[Wemos D1 mini Pro]
- Flash Mode:			[Only Sketch]
- Flash Size: 		[16MB/1MB]
- lwip Variant:   [v2 Lower Memory]
- Reset Method:		[ck|nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz|160MHz]
- Upload Using:		[SERIAL]
- Upload Speed:   [921600] (serial upload only)

### Problem Description
Modules on trial-2pcs, the problem is identical for both.
The module does not connect to AP, but brodcast strainght own SSID if  " WiFi.mode(WIFI_STA);" is not declared in setup()-aka the antenna performance is ok. The Serial Monitor gives readable info at 
74880 baud only
NodeMCU-12E performs pretty good with same code. AP(Slave) runs on ESP32 and has not any misworking or unstability with 2 Station NodeMCU-12E.

### [MCVE]Sketch
#include <ESP8266WiFi.h>        // Include the Wi-Fi library
#include <ESPAsyncWebServer.h>

const char* ssid     = "BOM";         // The SSID (name) of the Wi-Fi network you want to connect to
const char* password = "123456789";     // The password of the Wi-Fi network

IPAddress staticIP(192, 168, 4, 30);     // Left IP: 192.168.4.20 COM:24 COM:8 Right IP: 192.168.4.30 COM:23 COM:9
IPAddress gateway(192, 168, 4, 1);
IPAddress subnet(255, 255, 255, 0);

int rr;
AsyncWebServer server(80);

String readTemp() {
  rr = analogRead(A0);
  Serial.println(rr);
  return String(rr);
}

void setup() {
 Serial.begin(74880);         // Start the Serial communication to send messages to the computer
  WiFi.config(staticIP, gateway, subnet);
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);             // Connect to the network
  delay(10);
  /*
     // *** DATA OF THE MAC (Access Point and Station) of the ESP *** //
    Serial.print ("MAC Access Point of this ESP:"); Serial.println (WiFi.softAPmacAddress ());
    Serial.print ("Station MAC of this ESP:"); Serial.println (WiFi.macAddress ());

     WiFi.printDiag(Serial);
       while (WiFi.status() != WL_CONNECTED) {
     Serial.print('.');
     delay(500);
    }
    Serial.println('Connected');
  */
  server.on("/readsensor", HTTP_GET, [](AsyncWebServerRequest * request) {
    request->send_P(200, "text/plain", readTemp().c_str());
  });
  // Start server
  server.begin();
}

void loop() {

}

```

### Debug Messages

22:09:32.451 ->  ets Jan  8 2013,rst cause:2, boot mode:(3,7)
22:09:32.451 -> 
22:09:32.485 -> load 0x4010f000, len 1392, room 16 
22:09:32.485 -> tail 0
22:09:32.485 -> chksum 0xd0
22:09:32.485 -> csum 0xd0
22:09:32.485 -> v3d128e5c
22:09:32.485 -> ~ld

devyte, 2020-04-23:
The core version above doesn't make sense, that's just the message coming from the bootup.
Which core version did you build against? 2.6.3, or some specific commit from master branch?

Also, I don't understand the problem description. You said:
> The module does not connect to AP, but brodcast strainght own SSID if " WiFi.mode(WIFI_STA);" is not declared in setup()-aka the antenna performance is ok.

Please explain in more detail.

devyte, 2020-04-23:
Also, did you wipe the entire flash before downloading the sketch?

woodlist, 2020-04-24:
Thanks for your attention to given topic. Frankly said I'm unable to report about core version you requesting for. This is the compilation report's part beneath:
esptool.py v2.8
Serial port COM23
Connecting....
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: 84:f3:eb:04:f7:9b
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Auto-detected Flash size: 16MB
Compressed 296368 bytes to 214037...

I have compiled with all possible "Erase Flash" options, inclusive the "All Flash Contents".
When I comment up the "WiFi.mode(WIFI_STA);" line, the module starts the WiFi on AP+STA mode, brodcasting somewhat firmware SSID, called ESP-e75C one and ESP-e93A the another. In meantime both modules are unable to get connected to the gate, as configured in given sketch.

d-a-v, 2020-04-24:
@woodlist Do you have one working esp8266 (nodemcu) and one not working esp8266 (wemos d1) ?
(you state about esp32 which are running a different core).

@devyte suggested to select the flash menu option to "erase all flash".
I also suggest to enable the debug port option to "Serial". If one of the two boards boots, you'll see the current version of this arduino core you're running.

woodlist, 2020-04-24:
Thanks for advise. I have two working NodeMCU and two non-working Wemos D1 respectively. The slave is the same-ESP32 in both case. As per your advise on compiling option, the serial monitor date is following:
10:20:13.549 -> SDK:2.2.2-dev(38a443e)/Core:2.6.3=20603000/lwIP:STABLE-2_1_2_RELEASE/glue:1.2-16-ge23a07e/BearSSL:89454af
10:20:13.617 -> scandone
10:20:16.448 -> scandone
10:20:17.394 -> state: 0 -> 2 (b0)
10:20:17.428 -> state: 2 -> 3 (0)
10:20:17.428 -> state: 3 -> 5 (10)
10:20:17.428 -> add 0
10:20:17.428 -> aid 1
10:20:17.428 -> cnt 
10:20:21.414 -> state: 5 -> 2 (fc0)
10:20:21.414 -> rm 0
10:20:22.413 -> reconnect
10:20:22.413 -> state: 2 -> 0 (0)
10:20:22.533 -> scandone
10:20:23.528 -> reconnect
10:20:26.376 -> scandone
10:20:27.394 -> reconnect
10:20:30.224 -> scandone
10:20:30.224 -> state: 0 -> 2 (b0)
10:20:30.264 -> state: 2 -> 3 (0)
10:20:30.264 -> state: 3 -> 5 (10)
10:20:30.264 -> add 0
10:20:30.264 -> aid 1
10:20:30.264 -> cnt 
10:20:33.322 -> 
10:20:33.322 -> connected with BOMAG, channel 1
10:20:33.322 -> ip:192.168.4.30,mask:255.255.255.0,gw:192.168.4.1
10:20:33.322 -> ip:192.168.4.30,mask:255.255.255.0,gw:192.168.4.1

woodlist, 2020-04-24:
And for second board:
10:25:38.621 -> SDK:2.2.2-dev(38a443e)/Core:2.6.3=20603000/lwIP:STABLE-2_1_2_RELEASE/glue:1.2-16-ge23a07e/BearSSL:89454af
10:25:38.655 -> scandone
10:25:41.533 -> scandone
10:25:41.533 -> no BOMAG found, reconnect after 1s
10:25:41.601 -> reconnect
10:25:44.464 -> scandone
10:25:44.464 -> no BOMAG found, reconnect after 1s
10:25:44.566 -> reconnect
10:25:47.397 -> scandone
10:25:48.353 -> state: 0 -> 2 (b0)
10:25:48.353 -> state: 2 -> 3 (0)
10:25:48.387 -> state: 3 -> 5 (10)
10:25:48.387 -> add 0
10:25:48.387 -> aid 1
10:25:48.387 -> cnt 
10:25:51.421 -> 
10:25:51.421 -> connected with BOMAG, channel 1
10:25:51.421 -> ip:192.168.4.20,mask:255.255.255.0,gw:192.168.4.1
10:25:51.421 -> ip:192.168.4.20,mask:255.255.255.0,gw:192.168.4.1

woodlist, 2020-04-24:
But both moduls again are not able to interact with Slave board.

woodlist, 2020-04-24:
Please, advise, is it required to be introduced the sketch of Slave module?

woodlist, 2020-04-24:
The serial monitor data (continued):
10:27:44.031 -> bcn_timout,ap_probe_send_start
10:28:06.333 -> bcn_timout,ap_probe_send_start
10:28:31.909 -> bcn_timout,ap_probe_send_start
10:28:49.498 -> bcn_timout,ap_probe_send_start
10:28:52.001 -> ap_probe_send over, rest wifi status to disassoc
10:28:52.001 -> state: 5 -> 0 (1)
10:28:52.001 -> rm 0
10:28:54.825 -> scandone
10:28:54.825 -> no BOMAG found, reconnect after 1s
10:28:54.945 -> reconnect
10:28:57.766 -> scandone
10:28:57.766 -> state: 0 -> 2 (b0)
10:28:57.806 -> state: 2 -> 3 (0)
10:28:57.806 -> state: 3 -> 5 (10)
10:28:57.806 -> add 0
10:28:57.806 -> aid 2
10:28:57.806 -> cnt 
10:29:00.847 -> 
10:29:00.847 -> connected with BOMAG, channel 1
10:29:00.847 -> ip:192.168.4.20,mask:255.255.255.0,gw:192.168.4.1
10:29:00.847 -> ip:192.168.4.20,mask:255.255.255.0,gw:192.168.4.1
10:31:30.304 -> bcn_timout,ap_probe_send_start

woodlist, 2020-04-24:
An predictable behavior has the Slave board. When it unable to get data from Masters (Wemos D1), it restarts over and over again. But with NodeMCU this does not happens.

woodlist, 2020-04-24:
Serial monitor data on Slave, with working two NodeMCU:
10:56:30.962 -> Server Name: http://192.168.4.20/readsensor  HTTP Response code: 200
10:56:31.002 -> Server Name: http://192.168.4.20/readsensor  HTTP Response code: 200
10:56:31.116 -> Server Name: http://192.168.4.30/readsensor  HTTP Response code: 200
10:56:31.150 -> Server Name: http://192.168.4.30/readsensor  HTTP Response code: 200
10:56:31.558 -> Server Name: http://192.168.4.20/readsensor  HTTP Response code: 200
10:56:31.598 -> Server Name: http://192.168.4.20/readsensor  HTTP Response code: 200
10:56:31.704 -> Server Name: http://192.168.4.30/readsensor  HTTP Response code: 200
10:56:31.744 -> Server Name: http://192.168.4.30/readsensor  HTTP Response code: 200

woodlist, 2020-04-24:
And the Serial monitor data, with two Wemos D1:
10:58:53.393 -> Rebooting...
10:58:53.393 -> ets Jun  8 2016 00:22:57
10:58:53.393 -> 
10:58:53.393 -> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
10:58:53.393 -> configsip: 0, SPIWP:0xee
10:58:53.393 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
10:58:53.433 -> mode:DIO, clock div:1
10:58:53.433 -> load:0x3fff0018,len:4
10:58:53.433 -> load:0x3fff001c,len:1044
10:58:53.433 -> load:0x40078000,len:8896
10:58:53.433 -> load:0x40080400,len:5816
10:58:53.433 -> entry 0x400806ac
10:59:00.562 -> E (6978) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
10:59:00.562 -> E (6978) task_wdt:  - async_tcp (CPU 0/1)
10:59:00.562 -> E (6978) task_wdt: Tasks currently running:
10:59:00.562 -> E (6978) task_wdt: CPU 0: IDLE0
10:59:00.562 -> E (6978) task_wdt: CPU 1: loopTask
10:59:00.562 -> E (6978) task_wdt: Aborting.
10:59:00.562 -> abort() was called at PC 0x400e0313 on core 0
10:59:00.562 -> 
10:59:00.562 -> Backtrace: 0x4008c454:0x3ffbe170 0x4008c685:0x3ffbe190 0x400e0313:0x3ffbe1b0 0x40084791:0x3ffbe1d0 0x4016bd67:0x3ffbc100 0x400e173e:0x3ffbc120 0x4008a381:0x3ffbc140 0x40088b9d:0x3ffbc160
10:59:00.602 -> 
10:59:00.602 -> Rebooting...

d-a-v, 2020-04-24:
I can see one board able to connect to wifi, not the other.

1) About the board connecting to WiFi, we can't say why your ESP32 can't connect to it. We don't know your esp8266 code and if it is supposed to open a TCP port (?). Your PC should be also able to connect to it ?

2) About your second board that seems to have WiFi problems (`bcn_timeout`), It was suggested to you the option Flash > Erase All Flash. Did you do it ?

woodlist, 2020-04-24:
By your opinion, if the board has WiFi problem, why that is able to brodcast firmware SSID with excellent value of signal strainght? Yes, I do confirmation that the compilation gone with erase all flash contents. Anyway, I did external antenna socket desoldering and mounting on NodeMCU. Now, wemos boards are useless. I can deliver that to everyone, who would like to test those himself.

closed 2020-04-24 by woodlist

devyte, 2020-04-24:
Let's see if I understand your setup:
- you have an esp32 serving a wifi access point with ssid BOM
- you have a nodemcu in station mode that can connect to it correctly 
- you have a wemos d1 pro in station mode that can't connect to it, even though it's using the same code as the nodemcu

Is the above correct? 

Some questions:
- which country are you in? 
- which channel is the esp32 serving the access point on?
- on the wemos pro, does a wifi scan show the ssid of the esp32?
- when you try to connect the wemos pro to the esp32, are there any other devices already connected to the esp32 ap?
- what happens when you replace your static IP config with dhcp on the wemos pro? 
- how far away are the wemos pro/nodemcu from the esp32? 
- what happens if you move e. g. 10m further away?
- which antenna are you using for the pro?
- what happens if you use a different antenna?
- what happens if you change phy mode to G on the pro? to B? 
- what happens if you lower Tx power down to 80%? (I don't remember the db value for that, you'll have to figure it out) 
- what is brand of the flash chip on the pro? 
- what happens if you select board wemos d1 mini r2 and flash the pro with that? 

Please be aware that support for the wemos pro is still experimental. The board itself is different from other esp12-based, and it's susceptible to manufacturing variations and emf (no shielding).
Also, it is strongly advised to NOT use spiffs with the pro, but use LittleFS instead Which should be done anyways, because spiffs will be deprecated and eventually removed.

woodlist, 2020-04-24:
Thanks for continous efforts on support. I do confirmation that you understood my setup correctly.
Now, let me try to answer to your questions.
1. Am living in Armenia.
2. AP on ESP32 serve on channel 1.
3. Upon "Serial" option activation for compilation I got message that the Wemos has been connected to AP, but since actual data transfer did not happened, the AP gone into restart.
10:29:00.847 -> connected with BOMAG, channel 1
10:29:00.847 -> ip:192.168.4.20,mask:255.255.255.0,gw:192.168.4.1
4. AP has html page, with interative changable refresh rate of page. The data asquisition from two master modules happens as frequently, as is the refresh rate in milliseconds. Most low rate is limited at 300 milliseconds. However, this part of code become active, if at least one client browser has accessed the page. In other words, the browser itself generates this refreshing. If no active session, no refreshing and data asququsition happens. But when this data requests the AP and does not gets, the AP restarts.
5. DHCP mode on Wemos did not worked too.
6. The distance between 3 boards are up to 1 meter, while are on testing. The ESP32 with onboard antenna is accessible even being shielded to ground container, over 15 meters. NodeMCU was accessible being placed into shileded aluminum container, but at no far then 1 meter. Thus I was forced to use external antenna on master module.
7. The question is not clear for me.
8. Power regulation on Tx did not examined.
7. Pro did not worked neither with onboard and external antenna.
8. WINBOND 25Q128JVSQ
9. I have flashed the Wemos with all Wemos available boards options, even with native and successfuly passed NodeMCU-12 board chousing. The same result in all cases.

devyte, 2020-04-24:
I can't tell which answer is for which question. Please fix your previous comment.

woodlist, 2020-04-25:
[Screenrecorder-2020-04-25-10-14-05-196.zip](https://github.com/esp8266/Arduino/files/4532470/Screenrecorder-2020-04-25-10-14-05-196.zip)
In available around WiFi stations You will see 2 stations, whose names are starting with ESP prefix. These are Wemos modules, with firmware SSID. They are bringing up, if bypassed to set the WiFi mode as a station