esp8266/Arduino #7244: Error  with _min and _max macro

warnerthuis, 2020-04-23:
----------------------------- Delete below -----------------------------

If your issue is a general question, starts similar to "How do I..", is related to 3rd party libs, or is related to hardware, please discuss at a community forum like esp8266.com.

INSTRUCTIONS
============
If you do not follow these instructions, your issue may be dismissed.

1. Follow the checklist under Basic Infos and fill in the [ ] spaces with an X.
2. Fill in all the fields under Platform and Settings in IDE marked with [ ] (pick the correct option for you in each case, delete the others).
3. If you haven't already done so, test your issue against current master branch (aka latest git), because it may have been already fixed.
4. Describe your problem.
5. If you have a STACK DUMP decode it:

https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/stack_dump.html

6. Include a Minimal Complete Reproducible Example sketch that shows your issue. Do not include your entire project, or a huge piece of code.
7. Include debug messages:

https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/debugging.html 

8. Use markup (buttons above) and the Preview tab to check what the issue will look like.
9. Delete these instructions from the above to the below marker lines before submitting this issue.

----------------------------- Delete above -----------------------------

### Basic Infos

- [ ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [ ] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ ] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12]
- Core Version:			[2.6.3]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Generic ESP8266 Module]
- Flash Mode:			[qio|dio|other]
- Flash Size: 		[4MB/1MB]
- lwip Variant:   [v1.4|v2 Lower Memory|Higher Bandwidth]
- Reset Method:		[ck|nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz|160MHz]
- Upload Using:		[OTA|SERIAL]
- Upload Speed:   [115200|other] (serial upload only)

### Error  with _min and _max macro

I have code which uses the _max and the _min macro. They are in the Arduino.h library.
That compiles ok with version 2.4.0 of the library where the definition of the maro's is:

#define _min(a,b) ((a)<(b)?(a):(b))
#define _max(a,b) ((a)>(b)?(a):(b))

Now in version 2.6.3 the definition is:

#define _min(a,b) ({ decltype(a) _a = (a); decltype(b) _b = (b); _a < _b? _a : _b; })
#define _max(a,b) ({ decltype(a) _a = (a); decltype(b) _b = (b); _a > _b? _a : _b; })

And the errors are:

C:\Users\warner\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3\cores\esp8266/Arduino.h:254:85: internal compiler error: in cp_parser_abort_tentative_parse, at cp/parser.c:23793

 #define _min(a,b) ({ decltype(a) _a = (a); decltype(b) _b = (b); _a < _b? _a : _b; })

                                                                                          ^

C:\Users\warner\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3\cores\esp8266/Arduino.h:255:53: note: in definition of macro '_max'

 #define _max(a,b) ({ decltype(a) _a = (a); decltype(b) _b = (b); _a > _b? _a : _b; })

                                                           ^

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <Arduino.h>

void setup() {

}

void loop() {

}

```

### Debug Messages

```
Debug messages go here
```

d-a-v, 2020-04-23:
The new code in  2.6.3 is better than before (macro arguments can be reevaluated and this is bad).
You can replace these macro by `std::min` and `std::max` but they may not be usable in other arduino archs.
An internal compiler error is hard to reproduce.

Until we have gcc9.3, I suggest you add this in your code :
```
#ifdef ESP8266
#undef _min
#undef _max
//#define _min(a,b) ((a)<(b)?(a):(b))
//#define _max(a,b) ((a)>(b)?(a):(b))
#define _min(a,b) std::min(a,b)
#define _max(a,b) std::max(a,b)
#endif 
```

Alternatively you can try alpha [version 0.0.2](https://d-a-v.github.io) of this core that already has gcc9.3.

warnerthuis, 2020-04-23:
thanks, I'll try

devyte, 2020-04-23:
For user code, it is strongly advised to use std::max/std::min, and not the macros. 
The original macros you mention have a bug, where if you pass in anything that could have a side effect, you'll have the side effect evaluated twice. 
The new macro improves on that by making a temporary local, which should then get optimized away, but such preprocessor macros are far from ideal. especially when it comes to casting. 
In contrast, the standard std::min/std::max force same-type calculation without multiple side effects, and if you want to cast, you have to do so on purpose (which makes you think about what you're doing).

devyte, 2020-05-16:
@warnerthuis did you try the std min/max?

warnerthuis, 2020-05-18:
sorry for not mentioning that I used successfully the std /min/max aproach. I was too busy with other projects (node-red). It works now in the new version.

closed 2020-05-18 by warnerthuis

arbourp, 2021-11-10:
Wanted to leave a comment to follow-up here.
I ran into a very similar error while using the DMXInputNeoPixel demo from the LXESP8266UARTDMX library.

Oddly enough there is a comment saying that 'min' may fail so they used _min instead. pionting to this issue https://github.com/esp8266/Arduino/issues/263

At this point in time with 3.0.2 version of the esp8266 code I get
/STORAGE/home/megatron/.arduino15/packages/esp8266/hardware/esp8266/3.0.2/cores/esp8266/Arduino.h:243:19: error: statement-expressions are not allowed outside functions nor in template-argument lists

So I modified the code in the DMXInputNeoPixel demo to use 'min' instead of '_min' and it compiles fine.