esp8266/Arduino #7245: esp8266 ESP-12E memory leak under http POST

andyb2000, 2020-04-24:
### Basic Infos

- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [X] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [X] If there is a stack dump, I have decoded it.
- [X] I have filled out all fields below.

#### Platform

- Hardware:				ESP-12E (NodeMCU 1.0)
- Core Version:			1.8.9
- Development Env:		Arduino IDE
- Operating System:		Ubuntu

### Settings in IDE

- Module:  			NodeMCU 1.0 ESP-12E
- Flash Mode:			[qio|dio|other]
- Flash Size: 		4MB
- lwip Variant:   v2 Lower Memory
- Reset Method:		[ck|nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		80Mhz
- Upload Using:		SERIAL
- Upload Speed:   115200

### Problem Description

I'm writing every minute to an https web server and observing that after 1hr (Approx) the unit crashes and restarts due to memory error. I've added debug options and a memory watcher and can see the memory drops after each http.post of data (Testing memory before action and after action). It constantly drops and never returns until the unit gets too low and crashes/restarts.

I've tried with v2 Lower Memory, v2 higher bandwidth, v1.4 Higher Bandwidth.

I've also tried different board release versions:
2.6.3
2.5.2
2.5.1
2.4.2
2.4.0

Problem persists in all combinations.
In my sketch it runs for around 1-2hrs and slowly see memory exhausting. It is always after the http post that my code sees the memory reduction (in my code I've got several lines checking memory to try to find the leak point):
```
after http post (2832335) -> free:  6000 - max:  5624 - frag:   7% <- 
after http post (2892265) -> free:  5712 - max:  5336 - frag:   7% <- 
after http post (2952261) -> free:  5424 - max:  5048 - frag:   7% <- 
after http post (3012266) -> free:  5136 - max:  4760 - frag:   8% <- 
after http post (3072262) -> free:  4848 - max:  4472 - frag:   8% <- 
after http post (3132264) -> free:  4560 - max:  4184 - frag:   9% <- 
after http post (3192289) -> free:  4272 - max:  3896 - frag:   9% <- 
after http post (3252361) -> free:  3984 - max:  3608 - frag:  10% <- 
after http post (3312267) -> free:  3696 - max:  3320 - frag:  11% <- 
after http post (3445539) -> free:  2480 - max:  1536 - frag:  33% <- 
```

Until finally low memory:

```
after http post (3445539) -> free:  2480 - max:  1536 - frag:  33% <- 
[HTTP-Client][end] tcp is closed
Phone home to show I'm awake
[HTTP-Client][begin] url: https://myurl.co.uk:443/update.php
[HTTP-Client][begin] host: myurl.co.uk port: 443 url: /update.php
[HTTP-Client][sendRequest] type: 'POST' redirCount: 0
BSSL:_connectSSL: start connection
BSSL:CERT: 30 82 06 - REDACTED - 06
BSSL:_run_until: Timeout
BSSL:_wait_for_handshake: failed
BSSL:Couldn't connect. Error = 'Unknown error code.'
[... 255 lines cut ...]
0x4022bfff: ip_chksum_pseudo at core/inet_chksum.c line 395
0x401002a2: millis() at /home/browna/.arduino15/packages/esp8266/hardware/esp8266/2.6.3/cores/esp8266/core_esp8266_wiring.cpp line 188
0x40227552: tcp_output at core/tcp_out.c line 1361
0x40223b5d: sys_timeout_abs at core/timeouts.c line 189
0x4020f497: __esp_yield() at /home/browna/.arduino15/packages/esp8266/hardware/esp8266/2.6.3/cores/esp8266/core_esp8266_main.cpp line 107
0x4020fb3d: __delay(unsigned long) at /home/browna/.arduino15/packages/esp8266/hardware/esp8266/2.6.3/cores/esp8266/core_esp8266_wiring.cpp line 57
0x4020fb32: __delay(unsigned long) at /home/browna/.arduino15/packages/esp8266/hardware/esp8266/2.6.3/cores/esp8266/core_esp8266_wiring.cpp line 54
0x40203b18: WiFiClient::connect(IPAddress, unsigned short) at /home/browna/.arduino15/packages/esp8266/hardware/esp8266/2.6.3/libraries/ESP8266WiFi/src/include/ClientContext.h line 133
0x402057ab: BearSSL::WiFiClientSecure::connect(char const*, unsigned short) at /home/browna/.arduino15/packages/esp8266/hardware/esp8266/2.6.3/libraries/ESP8266WiFi/src/WiFiClientSecureBearSSL.cpp line 231
0x4020713c: HTTPClient::connect() at /home/browna/.arduino15/packages/esp8266/hardware/esp8266/2.6.3/libraries/ESP8266HTTPClient/src/ESP8266HTTPClient.cpp line 1195
0x40100fb1: umm_poison_realloc_fl(void*, size_t, char const*, int) at /home/browna/.arduino15/packages/esp8266/hardware/esp8266/2.6.3/cores/esp8266/umm_malloc/umm_local.c line 140
0x40207fd4: HTTPClient::sendRequest(char const*, unsigned char const*, unsigned int) at /home/browna/.arduino15/packages/esp8266/hardware/esp8266/2.6.3/libraries/ESP8266HTTPClient/src/ESP8266HTTPClient.cpp line 668
0x40246fee: umm_info(void*, int) at /home/browna/.arduino15/packages/esp8266/hardware/esp8266/2.6.3/cores/esp8266/umm_malloc/umm_info.c line 164
0x4020e3ab: String::reserve(unsigned int) at /home/browna/.arduino15/packages/esp8266/hardware/esp8266/2.6.3/cores/esp8266/WString.cpp line 146
0x40208194: HTTPClient::POST(unsigned char const*, unsigned int) at /home/browna/.arduino15/packages/esp8266/hardware/esp8266/2.6.3/libraries/ESP8266HTTPClient/src/ESP8266HTTPClient.cpp line 599
0x402081c2: HTTPClient::POST(String const&) at /home/browna/.arduino15/packages/esp8266/hardware/esp8266/2.6.3/libraries/ESP8266HTTPClient/src/ESP8266HTTPClient.cpp line 604
0x40201462: loop() at /media/ABDS2-ssh/Andy_s Documents/arduino/sketchbook/test/test.ino line 161
0x4020d600: fs::FS::open(char const*, char const*) at /home/browna/.arduino15/packages/esp8266/hardware/esp8266/2.6.3/cores/esp8266/FS.cpp line 354
0x401001b4: ets_post(uint8, ETSSignal, ETSParam) at /home/browna/.arduino15/packages/esp8266/hardware/esp8266/2.6.3/cores/esp8266/core_esp8266_main.cpp line 160
0x4020f5ac: loop_wrapper() at /home/browna/.arduino15/packages/esp8266/hardware/esp8266/2.6.3/cores/esp8266/core_esp8266_main.cpp line 180

```

Thanks in advance for your help, I'm trying various things (the above code is a cut down version of my code that I'm doing to test the memory leak and it's present in this cut down version too) and no avail!
Andy.

earlephilhower, 2020-04-24:
@andyb2000 you put an apikey in your post.  If this is anything critical, change it and edit the MCVE above to obscure it ASAP.

What is the beginning free memory reported in your sketch?  Do you start with significantly more free space than the 5.5K shown?

Also, debugging is not terribly optimized, so if you're only running w/debugging enabled you might want to give it a go w/o debugging.  There are quite a few strings added when debug is enabled, and not all of them end up in PROGMEM (so they eat into heap).

Finally, I'd move the BearSSL client from a temporary new to a global instantiation:
`BearSSL::WiFiClientSecure client();`.  There is no MMU and heap fragmentation can be a problem.

andyb2000, 2020-04-24:
Thanks @earlephilhower apikey not relevant so no worries! Thank you.

I'd previously ran without debug and added it as I found more memory leaks so although debug is heavy it's not contributing in this case.
I've been testing various scenarios out all afternoon and have identified it is the SSL+BearSSL components that seem to be leaking memory.

I've changed to this code:
```
        // BearSSL::WiFiClientSecure *client = new BearSSL::WiFiClientSecure();
        // client->setInsecure();
         HTTPClient http;    //Declare object of class HTTPClient
         http.setTimeout(15000);
         int beginResult = http.begin(server);
         http.addHeader("Content-Type", " application/x-www-form-urlencoded; charset=UTF-8");
         http.addHeader("User-Agent", "TestWeb-v1");
         int httpCode = http.POST("&testvar=1&testvar2=2&testvar3=3&testvar4=4&testvar5=5"); 
         payload = http.getString();                  //Get the response payload
```
(note removed the mem check to make it clearer to read) and now I'm getting no further memory loss after initial couple of loops setting up variables, etc.

So the memory loss appears to be relating to the BearSSL part unfortunately.

So I think you're on the right lines to global instantiate the BearSSL, so I'll give that a try now and feed back.

andyb2000, 2020-04-24:
Success.
as @earlephilhower advised. Moving the:
```
BearSSL::WiFiClientSecure *client = new BearSSL::WiFiClientSecure();
```
up out of the loop and into the top definitions has solved the memory leak/loss during code operation. I've had it running now for almost 15 minutes with no loss of memory in my test code.

Thanks @earlephilhower is this worth moving to the bearssl-esp8266 issue tracker or just closing off?

earlephilhower, 2020-04-24:
No, this is a bug in your example code.  I see you `new`ing the client...but I don't see a `delete client` anywhere.  So you've been allocating the SSL contexts without freeing them.

Since the context is a little big, it's a good idea to allocate once (i.e. as a global) and just leave it to avoid memory fragmentation.

closed 2020-04-24 by earlephilhower

Abedi98, 2021-03-07:
> No, this is a bug in your example code. I see you `new`ing the client...but I don't see a `delete client` anywhere. So you've been allocating the SSL contexts without freeing them.
>
> Since the [...]

hi
how to delete client after complete request ??

`BearSSL::WiFiClientSecure *client = new BearSSL::WiFiClientSecure();`

devyte, 2021-03-21:
The answer is in your question: `delete client;`