esp8266/Arduino #7253: Unable to upload sketch for ESP8266 due to some elf2bin.py error

BenniKain, 2020-04-29:
> Hello,
i am trying for hours to get a sensor sending data over wifi to my flask server.
however due to some reason of esp82 error link below i cannot upload the sketch.

it worked 2 times in 4 hours for me seemingly random.

if anyone could help me i d be very thankful

```

Arduino: 1.8.12 (Windows Store 1.8.33.0) (Windows 10), Board: "LOLIN(WEMOS) D1 mini Lite, 80 MHz, Flash, Legacy (new can return nullptr), All SSL ciphers (most compatible), 1MB (FS:64KB OTA:~470KB), v2 Lower Memory, Disabled, None, Only Sketch, 921600"

C:\Users\User\Documents\Arduino\testung_httpclient\testung_httpclient.ino: In function 'void loop()':

C:\Users\User\Documents\Arduino\testung_httpclient\testung_httpclient.ino:56:47: warning: 'bool HTTPClient::begin(String)' is deprecated (declared at C:\Users\User\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.6.3\libraries\ESP8266HTTPClient\src/ESP8266HTTPClient.h:155) [-Wdeprecated-declarations]

    http.begin("http://192.168.8.105:8090/post");      //Specify request destination

                                               ^

Traceback (most recent call last):

  File "C:\Users\User\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.6.3/tools/elf2bin.py", line 169, in <module>

    sys.exit(main())

  File "C:\Users\User\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.6.3/tools/elf2bin.py", line 159, in main

    write_bin(out, args.app, ['.irom0.text', '.text', '.text1', '.data', '.rodata'], 0, args.flash_mode, args.flash_size, args.flash_freq, args.path)

  File "C:\Users\User\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.6.3/tools/elf2bin.py", line 74, in write_bin

    [size, addr] = get_segment_size_addr(elf, segment, path)

  File "C:\Users\User\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.6.3/tools/elf2bin.py", line 55, in get_segment_size_addr

    raise Exception('Unable to find size and start point in file "' + elf + '" for "' + segment + '"')

Exception: Unable to find size and start point in file "C:\Users\User\AppData\Local\Temp\arduino_build_644626/testung_httpclient.ino.elf" for ".text1"

exit status 1

``
`#include <ESP8266WiFi.h>
#include <ESP8266HTTPClient.h>

//#include <Adafruit_Sensor.h>
#include <DHT.h>
#include <DHT_U.h>
//set pin D2 as DHTPin

#define DHTPIN D2
#define DHTTYPE    DHT11     // DHT 11
DHT_Unified dht(DHTPIN, DHTTYPE);

void setup() {
  Serial.begin(115200);                      //Serial connection
  //WiFi.begin("julia handy", "2upFD9up");   //WiFi connection
  WiFi.begin("HUAWEI-40EE", "55275248"); 
  dht.begin();
  //declare sensor
  sensor_t sensor;
  dht.temperature().getSensor(&sensor);

  Serial.print  (F("Sensor Type: ")); Serial.println(sensor.name);
  Serial.print  (F("Driver Ver:  ")); Serial.println(sensor.version);
  Serial.print  (F("Unique ID:   ")); Serial.println(sensor.sensor_id);

  dht.humidity().getSensor(&sensor);
  Serial.print  (F("Sensor Type: ")); Serial.println(sensor.name);
  Serial.print  (F("Driver Ver:  ")); Serial.println(sensor.version);
  Serial.print  (F("Unique ID:   ")); Serial.println(sensor.sensor_id);
  
  while (WiFi.status() != WL_CONNECTED) {  //Wait for the WiFI connection completion
 
    delay(1000);
    Serial.println("Waiting for connection");
 
  }
   Serial.println("Connected");
}

void loop() {
 
 if(WiFi.status()== WL_CONNECTED){   //Check WiFi connection status
   //define variabe event
   sensors_event_t event;
   // set event as dht temperature
   dht.temperature().getEvent(&event);
   //get value of event
   float temperature=  event.temperature;
   dht.humidity().getEvent(&event);
   float humidity=  event.relative_humidity;
   
   String code = "humidity="+ String(humidity) + ",temperature=" +String(temperature);
   HTTPClient http;    //Declare object of class HTTPClient
 
   http.begin("http://192.168.8.105:8090/post");      //Specify request destination
   http.addHeader("Content-Type", "text/plain");  //Specify content-type header

   int httpCode = http.POST(code);   //Send the request
   String payload = http.getString();                  //Get the response payload
 
   Serial.println(httpCode);   //Print HTTP return code
   Serial.println(payload);    //Print request response payload
 
   http.end();  //Close connection
 
 }else{
 
    Serial.println("Error in WiFi connection");   
 
 }
 
  delay(5000);  //Send a request every 30 seconds
 
}`

devyte, 2020-05-01:
I just tried this 5 times in a row, and it builds fine for a wemos D1 r2 mini. 
If your problem is random, and only you seem to be affected, I suggest doing a fresh install from scratch, and make sure you have only one core installed when testing.
Also, my own personal view is that I only trust the IDE downloaded from the Arduino site.
Also, you might want to check whether you have some other Python installed.
Closing due to can't reproduce.

closed 2020-05-01 by devyte

dominikheinz, 2020-09-16:
I am having the same problem. Did you figure out what is causing this?

anilsoni85, 2021-10-30:
I encountered same issue consistently on Windows 10 and I fixed it by adding reattempt logic in get_segment_size_addr function of elf2bin.py. 

[image]

panakos, 2022-12-24:
Citrix Workspace was causing the issue. I updated Citrix and  problem gone!

kah731, 2022-12-27:
I have a recent version of Citrix Workspace - hoping to not have to update/remove.

I reinstalled the latest version of python (python-3.11.1-amd64.exe), and that fixed it.
I also let python install the fix to allow longer paths.

panakos, 2022-12-27:
I tried that as well but didn't work for me. Citrix update to the latest fixed it

kah731, 2022-12-27:
Thanks - somehow it broke again - will try Citrix fix now

bonfix, 2023-07-01:
Updated python, didn't work, had to uninstall citrix.