esp8266/Arduino #7261: WiFiclient::close() no longer working gracefully in 2.6.3

pidloop, 2020-05-02:
Hello!

I have been using 2.4.0 for a few years without any issues but just out of curiosity today I thought I would try 2.7.0. My application is a web server on Adafruit ESP8266 Huzzah.

When using 2.4.0 I could use curl to retrieve web pages without any errors. Now with 2.7.0 curl always reports:

Recv failure: Connection reset by peer

Researching this error message suggests WiFiClient.close() may now be sending TCP RST instead of the more polite FYN-ACK handshake but this is just a guess. All I know is that something has changed for the worse in the intervening releases. I will certainly continue to use 2.4.0 so this is not urgent but someday I suppose I should update.

I apologize for not following the submission guidelines. I am just hoping this might trigger a memory from one of the maintainers to take a look.

Many thanks for a great platform,

Elwood Downey
Tucson AZ

d-a-v, 2020-05-02:
Try to use `while (client.read() != -1);` before closing.

pidloop, 2020-05-02:
Yes, that fixes it, thank you. But why the change?

pidloop, 2020-05-02:
I changed the loop so I could see what more was coming from the client: 

`while (client.available())
        Serial.print ((char)client.read());`

I saw:

`User-Agent: curl/7.64.1`
`Accept: */*`

Now I realize what's happening is my server is going ahead and processing the RESTFUL request as soon as it sees the initial GET command and ignoring the remainder of traffic from the client. Then when close() is called, it is actually correct for it to send RST back to the client because some the client's data has not yet been processed.

So evidently the 2.4.0 close() was not so careful to check whether unread client traffic was pending and 2.7.0 does this correctly. Well done to all.

I can now address this properly.

Many thanks for your help.

Take care.

d-a-v, 2020-05-02:
This stackoverflow answer explains it well:

https://stackoverflow.com/questions/1434451/what-does-connection-reset-by-peer-mean

Several bugs were fixed since 2.4.x and it all ended in a normal and predictable network behavior with esp8266 arduino. I can't say exactly why it was working. 

What I surely can say is that your web protocol is not fully complete (the added line ignores received data part of the http protocol) (but it says "thank you " while receiving them - hence peer isn't hit by a slammed door while talking)

devyte, 2020-05-02:
Glad to see this was cleared up. I'm also glad you're updating, it's not good to stay on an older core. 
Closing due to not an issue.

closed 2020-05-02 by devyte