esp8266/Arduino #7269: Updater does not work anymore on ESP-12F since 2.7.0

vdelmedico, 2020-05-04:
#Basic Infos

- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [X] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [X] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12F]
- Core Version:			[2.7.0]
- Development Env:		[Arduino IDE]
- Operating System:		[Linux Mint]

### Settings in IDE

- Module:  			[Wemos D1 mini r2]
- Flash Mode:			[dio]
- Flash Size: 		[4MB]
- lwip Variant:   [v1.4]
- Reset Method:		[ck]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200] 

### Problem Description

I have moved from Core 2.6.3 to core 2.7.0.

Since then, I've spent a long time trying to make HTTPUpdate working, but I couldn't.
Even though everything seems to work perfectly, after reset the hardware is unable to run the new firmware.

I thought I had an issue somewhere, so I remembered that Arduino OTA was working fine.
So I tried to update the firmware with Arduino OTA, and this time it was not working anymore !
Same behaviour : flash write seems OK, but after the reset the new firmware can't run.

Finally, I swapped back from 2.7.0 to 2.6.3, and everything works fine ( Arduino OTA, HTTPUpdate, etc... )

My guess is that core 2.7.0 added a bug in Updater.cpp on ESP8266.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <Arduino.h>

void setup() {
  Serial.begin(115200);
  Serial.println("Booting");
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  while (WiFi.waitForConnectResult() != WL_CONNECTED) {
    Serial.println("Connection Failed! Rebooting...");
    delay(5000);
    ESP.restart();
  }

  ArduinoOTA.onStart([]() {
    String type;
    if (ArduinoOTA.getCommand() == U_FLASH) {
      type = "sketch";
    } else { // U_FS
      type = "filesystem";
    }

    // NOTE: if updating FS this would be the place to unmount FS using FS.end()
    Serial.println("Start updating " + type);
  });
  ArduinoOTA.onEnd([]() {
    Serial.println("\nEnd");
  });
  ArduinoOTA.onProgress([](unsigned int progress, unsigned int total) {
    Serial.printf("Progress: %u%%\r", (progress / (total / 100)));
  });
  ArduinoOTA.onError([](ota_error_t error) {
    Serial.printf("Error[%u]: ", error);
    if (error == OTA_AUTH_ERROR) {
      Serial.println("Auth Failed");
    } else if (error == OTA_BEGIN_ERROR) {
      Serial.println("Begin Failed");
    } else if (error == OTA_CONNECT_ERROR) {
      Serial.println("Connect Failed");
    } else if (error == OTA_RECEIVE_ERROR) {
      Serial.println("Receive Failed");
    } else if (error == OTA_END_ERROR) {
      Serial.println("End Failed");
    }
  });
  ArduinoOTA.begin();
  Serial.println("Ready");
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());
}

void loop() {
ArduinoOTA.handle();
}

```

devyte, 2020-05-04:
What is the brand of your flash chip?

alf159, 2020-05-04:
I have this same problem on ESP-12F, 4MB EEPROM. I had to disassemble modules from application and flash sketch via serial port and 2.6.3 core.
Brand EEPROM i dont know, i have only prtscr from new ESP.
[image]

vdelmedico, 2020-05-04:
> What is the brand of your flash chip?

How can you read that info ?

alf159, 2020-05-04:
May be write **esptool.py --port COMx flash_id**
or remove heatspreader  :-(

vdelmedico, 2020-05-04:
```
#python esptool.py --port /dev/ttyUSB0 flash_id
esptool.py v2.8
Serial port /dev/ttyUSB0
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: xx:xx:xx:xx:xx:xx
Uploading stub...
Running stub...
Stub running...
Manufacturer: 20
Device: 4016
Detected flash size: 4MB
Hard resetting via RTS pin...
```

Jason2866, 2020-05-04:
It is a xmc flash. Same issue as #7267

devyte, 2020-05-04:
Closing as duplicate of #7267.

closed 2020-05-04 by devyte