esp8266/Arduino #7270: I2C VL53L0X Sensors cause WDT resets

martin072, 2020-05-04:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:				[Wemos D1 Mini]
- Core Version:			[2_6_3]
- Development Env:		[Arduino IDE 1.8.10]
- Operating System:		[MacOS]

### Settings in IDE

- Module:  			[Wemos D1 mini]
- Flash Mode:			[-]
- Flash Size: 		[4MB]
- lwip Variant:   [v2 Lower memory]
- Reset Method:		[-]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[ESP8266HTTPUpdateServer]
- Upload Speed:   [-)

### Problem Description

I have a system that utilises 2 VL53L0X sensors. This works fine for the majority of time, but after around 20 hours till 2 days (varying), the ESP resets due to Software a WDT, I've managed to do a stack trace, and this is the result:  

```
    PC: 0x40100529: Twi::busywait(unsigned char) at /Users/martin/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/cores/esp8266/core_esp8266_si2c.cpp line 271
    EXCVADDR: 0x00000000
    
    Decoding stack results
    0x40218f25: Twi::readFrom(unsigned char, unsigned char*, unsigned int, unsigned char) at /Users/martin/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/cores/esp8266/core_esp8266_si2c.cpp line 405
    0x4010083e: millis() at /Users/martin/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/cores/esp8266/core_esp8266_wiring.cpp line 188
    0x4021913c: twi_readFrom(unsigned char, unsigned char*, unsigned int, unsigned char) at /Users/martin/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/cores/esp8266/core_esp8266_si2c.cpp line 1042
    0x402090b0: TwoWire::requestFrom(unsigned char, unsigned int, bool) at /Users/martin/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/libraries/Wire/Wire.cpp line 129
    0x402090dc: TwoWire::requestFrom(unsigned char, unsigned char) at /Users/martin/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/libraries/Wire/Wire.cpp line 143
    0x402140eb: VL53L0X::readReg(unsigned char) at /Users/martin/Documents/Arduino/libraries/vl53l0x-arduino-1.02/VL53L0X.cpp line 322
    0x40214442: VL53L0X::readRangeContinuousMillimeters() at /Users/martin/Documents/Arduino/libraries/vl53l0x-arduino-1.02/VL53L0X.cpp line 816
    0x40201f00: readVLXSensors() at /Users/martin/Documents/Arduino/TEST_WS2812/TEST_WS2812.ino line 674
    0x402083c7: loop() at /Users/martin/Documents/Arduino/TEST_WS2812/TEST_WS2812.ino line 596
    0x40218404: loop_wrapper() at /Users/martin/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/cores/esp8266/core_esp8266_main.cpp line 172
```

This is the function that reads the sensors (initiated from the main loop):

```
    void readVLXSensors() {
    
      if (readTop) {
        distSensorTop_distance = distSensorTop.readRangeContinuousMillimeters();
        if (distSensorTop.timeoutOccurred()) {
          distSensorTop_distance = 99999;
        }
        readTop = !readTop;
        
      } else {
        distSensorBottom_distance = distSensorBottom.readRangeContinuousMillimeters();
        
        if (distSensorBottom.timeoutOccurred()) {
          distSensorBottom_distance = 99999;
        }
        readTop = !readTop;
      }
      
    }

```
As you can see as last desperate measure, I alternate the reading of the sensors, where as before I just read both sensors after each other. But even this approach still gives the WDT resets. 

Any one an approach on this one?

Tech-TX, 2020-05-04:
I have that VL53L0X sensor and have seen bad initialization code for it in some demo code (while...  without yield) that may cause WDTs.  However,  that's your app or the 3rd party library for the sensor,  and not the Wire library at fault.  Nothing in the I2C library is causing it.

devyte, 2020-05-04:
At a glance, I see ReadRangeContinuousBlah() has a while loop without yield. It's pretty obvious.
Closing due to not a core issue.

closed 2020-05-04 by devyte

martin072, 2020-05-04:
Many thanks for the quick responses. Will dig a bit further

gururajcs, 2024-02-18:
> Many thanks for the quick responses. Will dig a bit further

Martin072, I was trying to setup my ESP12F with OLED1306 and VL53L0X to i2c.
I had similar issue (ie. Soft WDT reset, Exception (4) and exactly same stack strace you have posted above). I added delay/yield in the VL53L0X.cpp file in various places with the hint from stacktraces. Then started getting WDT reset issue at Adafruit_SSD1306::display()  Stack trace: display() -> TwoWire::endTransmission->Adafruit_SSD1306::ssd1306_commandList->Twi::writeTo->Twi::write_byte->Twi::busywait

Atlast I used another Library for [SSD1306](https://github.com/cmmakerclub/ESP_Adafruit_SSD1306/tree/master) as provided by in this example [Create a height measuring device by combining ESP8266, VL53L0X, and MPU6050](https://nefastudio.net/2017/04/23/4680.html)...It started working without any issue.

I know its been long for this reply. But thought it will help if someone search for this issue and lands into this page.