esp8266/Arduino #7273: ESP8266WIFI debug file generator location

wadwin, 2020-05-05:
#### Platform

- Hardware:				[ESP8266]
- Core Version:			[esp8266/2.7.0]
- Development Env:		[Arduino IDE]
- Operating System:		[Ubuntu]

### Settings in IDE

- Module:  			[Wemos D1 R1]
- Upload Speed:   [115200]
- **Debug port : [Serial]**
- **Debug level : [WIFI]**

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp
#include <ESP8266WiFi.h>
#include <WiFiClient.h>

// Se connecter au réseau WiFi APTEST avec le mot de pase 12345678
// Puis aller sur http://192.168.4.1 dans un navigateur web
const char *ssid = "APTEST";
const char *password = "12345678";
WiFiServer server(80);

void setup() {
  delay(1000);
  Serial.begin(115200);
  //Serial.println("Configuring access point...");
  WiFi.mode(WIFI_AP);
  WiFi.softAP(ssid, password);

}

void loop() {
}
```

### Debug Messages

```
wifi evt: 7
wifi evt: 7
add 1
aid 1
station: 00:b3:62:1e:fa:7b join, AID = 1
wifi evt: 5
wifi evt: 9
wifi evt: 7
wifi evt: 7
```

Hi all,
I have enabled DEBUG_WIFI mode with :
Arduino Setting -> Debug port : [Serial]
Arduino Setting -> Debug level : [WIFI]

I wonder where is the file who generate somes debugs lines.
I have found that "wifi evt:7", "wifi evt:5", "wifi evt:9" are generate by : /home/utilisateur/.arduino15/packages/esp8266/hardware/esp8266/2.7.0/libraries/ESP8266WiFi/src/ESP8266WiFiGeneric.cpp
(line : 229) DEBUG_WIFI("wifi evt: %d\n", event->event);

but i don't know for :
"add 1"
"aid 1"
"station: 00:b3:62:1e:fa:70 join, AID = 1"

Can somebody can help me please ?

d-a-v, 2020-05-05:
These messages are generated by the firmware which is out of our control.

closed 2020-05-05 by earlephilhower

wadwin, 2020-05-05:
Can I have access to this source-files ?

d-a-v, 2020-05-05:
[We all would like to.](https://github.com/espressif/ESP8266_NONOS_SDK/tree/release/v2.2.x)

wadwin, 2020-05-05:
Thx for answers,
I have found with the keyword firmware : https://github.com/nodemcu/nodemcu-firmware
I have found in the file nodemcu-firmware-master/app/modules/wifi_eventmon.c
(line 203)       EVENT_DBG("Station: " MACSTR "join, AID = %d",
          MAC2STR(evt->event_info.sta_connected.mac),
          evt->event_info.sta_connected.aid);

If I Make the .bin file and I flash will it work ?

d-a-v, 2020-05-05:
This is not as simple as that. I don't know why nodemcu project duplicated these messages.
What I know is that they are also using the same nonos-sdk binary-only firmware.
There are no sources for it.

```shell
$ cd tools/sdk/lib/NONOSDK22x_191122/
$ grep -rl "join, AID ="
libnet80211.a
```
This file is part of NONOS-SDK firmware.

You will anyway *not be able* to use any `.bin` file produced by nodemcu project in esp8266 arduino.

devyte, 2020-05-05:
To be clear, the evt and such messages that you saw come from closed source libs provided by Espressif. Our core is built on top of them. There are no sources available, the libs are provided by Espressif in precompiled form.
We have no relation with the lua nodemcu project, that is a completely different thing.

wadwin, 2020-05-05:
I'm trying to print each password that users try to use to know if someone try to connect without my permission, maybe you will have an idea ?

wadwin, 2020-05-06:
I have find the file who generate the file who generate text : "station: 00:b3:62:1e:fa:7b join, AID = 1" 
/home/utilisateur/.arduino15/packages/esp8266/hardware/esp8266/2.7.0/tools/sdk/lib/NONOSDK22x_190703/libnet80211.a
I have used : https://hexed.it/ to de-compile and edit text to find that.

Can I have the source-code of this file and build it again with my modifications ?

devyte, 2020-05-06:
@wadwin there is no source code available for those libs. We don't have access. We get the libs in precompiled form.

d-a-v, 2020-05-06:
@wadwin We have told you ten times the same thing, including the filename you have found.
Are we speaking french ;) ?

Joke apart, I can reexplain it to you in french if you reach me by email.

wadwin, 2020-05-06:
Bonjour d-a-v
En vf je veux bien XD