esp8266/Arduino #7280: LittleFS and ArduinoJSON, why don't you close the file in the example?

sblantipodi, 2020-05-08:
### Basic Infos

- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [X] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [X] If there is a stack dump, I have decoded it.
- [X] I have filled out all fields below.

#### Platform

- Hardware:					ESP8266EX on D1 Mini Board
- Core Version:			2.7
- Development Env:		PlatformIO
- Operating System:		Windows

### Settings in IDE

- Module:  			Wemos D1 mini 
- Flash Mode:			default
- Flash Size: 		4MB
- Reset Method:		default
- Flash Frequency:	default
- CPU Frequency:		80MHz
- Upload Using:		OTA 
- Upload Speed:   115200

### Problem Description

1) This example does not work really well if you don't close the file:
https://github.com/esp8266/Arduino/blob/master/libraries/esp8266/examples/ConfigFile/ConfigFile.ino
don't understand why you don't close the file when using arduino json. If I don't close the file I can't correctly read/write the file.

earlephilhower, 2020-05-08:
It seems to be working:

````
Mounting FS...
Config saved
Loaded serverName: api.example.com
Loaded accessToken: 128du9as8du12eoue8da98h123ueh9h98
Config loaded
````

Can you be more specific with the problem?

sblantipodi, 2020-05-09:
if you return the StaticJsonDocument instead of returning void, the returned object is empty.
if you close the file before returning the object, there is no problem.

it is right to leave a file open?

earlephilhower, 2020-05-09:
Files are closed on destruction, which in this example happens when the function exits.  Closing as not an issue.  You can get more help on this kind of question at the forums at http://esp8266.com.

closed 2020-05-09 by earlephilhower

devyte, 2020-05-10:
Also, don't return a heavy object like StaticJsonDocument by value, unless you know exactly what you're doing (i.e. how to exactly meet NRVO).