esp8266/Arduino #7283: Strange behavior on while timed light sleep

bejarane, 2020-05-09:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:					[Node MCU 1.0 with esp8266mod]
- Core Version:			[2.7.0]
- Development Env:		[Arduino IDE 1.8.7]
- Operating System:		[Ubuntu]

### Settings in IDE

- Module:  			[Nodemcu 1.0]
- Flash Mode:			[not selectable]
- Flash Size: 		[4MB]
- lwip Variant:   [v2 Lower Memory]
- Reset Method:		[nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [921600] (serial upload only)

### Problem Description

Unexpected behavior when following example of LowPowerDemo.ino included on library examples. Issue is as follows:

When using test number 5 (Timed light sleep) it is possible to use the wake_up_pin for triggering a wakeup before the end of the sleep time established as 10 seconds. Vanilla example works as expected, but when deactivating the serial flush call in printMillis() the interrupt stops working. so, instead of waking up immediately, it prints the content of wakeupCallback() but does not resumes program and waits 10 seconds again to resume and exit the sleep. Current work around is to have the flush() always inside this callback function.

At the beginning I thought it was just a serial error and code was still running, but actually used a counter and the count of sleeps entered on the "onwhite" time does not increases, which means that this unwanted waiting happens inside the same sleep call an does not allows the program to advance.

I call this an strange behavior because removing that specific flush() line should affect only the serial print buffer and not the timing of the code below it. But it seems that the integrity of the wakeup with interrupt on light sleep relays on that serial.flush() call.

My doubt is why is this happening. Is this a bug or is it intended to work this way?

Original function
```
void printMillis() {
   Serial.print(F("millis() = "));  // show that millis() isn't correct across most Sleep modes
   Serial.println(millis());
   Serial.flush();  // needs a Serial.flush() else it may not print the whole message before sleeping
}
```

Faulty function
```
void printMillis() {
   Serial.print(F("millis() = "));  // show that millis() isn't correct across most Sleep modes
   Serial.println(millis());
   //Serial.flush();  // needs a Serial.flush() else it may not print the whole message before sleeping
}
```

devyte, 2020-05-10:
CC @Tech-TX

Tech-TX, 2020-05-10:
I don't exactly know *what* the SDK does with timed light sleep (and *how*), but it's certainly weird.

I went through nearly 50 different iterations of the code to try and figure out what worked and when, but couldn't come up with much more than I'd documented in the program and README.  The Espressif SDK reference is remarkably skimpy on details.

As far as the Serial.flush(), yes, that's required if you have a print statement pending before you enter/exit light sleep.  It won't correctly go into light sleep if any timers are running, and that includes the baud rate timer for the serial port.  From what I could tell, the callback is required by the SDK as part of the timed light sleep, and it's not fully out of timed light sleep until the callback exits, which causes the following delay() to execute.  If it's properly in timed light sleep, that delay() occurs *during* the sleep interval, and if it's not fully in sleep then it occurs afterward.  In the SDK they call the mechanism that allows it to enter sleep 'idle', which equals 'delay()' in the Arduino world.  I tried decompiling the relevant parts of delay() in the SDK and couldn't make sense of it.  I still don't know what 'idle' does, or how to do it without the SDK.

The code in the SDK reference PDF doesn't work.  It may have worked with a different version of the SDK, but it's not working today.  The only functional example I found (after searching for days) was the one I'd linked in the PR notes, and I still needed to tweak it to work with the current SDK.

In order to make timed light sleep work properly, we need to stop all of the ETS_timers and then restart them after sleep exits.  Right now the code manages to stop them, but the restart doesn't function: most of the timers that had been running before sleep are still unhooked by the SDK.  It works OK in the demo because it hits a Deep Sleep as the next test, and the RESET following Deep Sleep reactivates the ETS_timers.

Does any of that make sense?

bejarane, 2020-05-10:
Great! Yes, it makes sense. Thank you very much for all the explanation and effort. Resolving now.

closed 2020-05-10 by bejarane