esp8266/Arduino #7285: Suddenly SPIFFS fail to upload.

Buzzardbait, 2020-05-10:
### Basic Infos
- [ x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ x] I have tested that the issue is present in current master branch (aka latest git).
- [x ] I have searched the issue tracker for a similar issue.
- [ x] If there is a stack dump, I have decoded it.
- [x ] I have filled out all fields below.

#### Platform

- Hardware:					[NodeMCU 1.0 Esp-12E Module]
- Core Version:			[ESP8266 Core 2.7.0]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows|

### Settings in IDE

- Module:  			[Nodemcu]
- Flash Mode:			[dio]
- Flash Size: 		[4MB/1MB]
- lwip Variant:   [v2 Lower Memory]
- Reset Method:		[nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200]

### Problem Description
SPIFFS upload suddenly stopped working. Simply says: "esptool: error: unrecognized arguments: --end 
SPIFFS Upload failed!"
No changes to the IDE Tools menu settings, no known updates to the NodeMCU library or Arduino IDE except routine automatic library updates. Fails on multiple NodeMCU boards. Sketch uploads on the same serial port work fine. I have no control over the arguments given to the esptool during upload. I did a repair on my Python installation but it didn't help.

devyte, 2020-05-10:
The --end error usually happens when you have an installation issue, i. e. something got messed up. I suggest a clean install from scratch. 
Closing.

closed 2020-05-10 by devyte

Buzzardbait, 2020-05-10:
Upgraded to LittleFS and all is working again.

mhightower83, 2020-05-10:
Not long ago I had a similar symptom. I think the problem is that Arduino IDE plugins have no way to check for version updates. For me when I updated the [SPIFFS uploader plugin](https://arduino-esp8266.readthedocs.io/en/latest/filesystem.html?highlight=SPIFFS#uploading-files-to-file-system), this problem went away.

It might be a good idea to check for updates for other plugins. I found the `ESP Exception Decoder` results greatly improved when I also updated it.

Since these tools are in your Arduino sketchbook folder they easily get missed and are just carried forward when you preserve your old sketches and try to install fresh. Since they are already there, you don't get messages about them being missing. Things just don't work well.

@devyte, maybe this should be in a wiki ;)

Buzzardbait, 2020-05-10:
Yeah, thanks there are a bewildering number of libraries and the Arduino IDE doesn't even report what's there for NodeMCU 1.0 ESP-12E such as the File System Upload Tool and(I think) One wire.

Buzzardbait, 2020-05-10:
Say, on the subject of libraries. What is the significance of some #includes being highlighted in red while others are not highlighted such as <FS.h> and <LittleFS.h> ? There must some implication like meaning it's not a standard Arduino library or something? Seems to me several others like stdio are also not highlighted in red.

devyte, 2020-05-10:
IMHO, the highlights are just eye candy. Personally, I don't trust them or believe what they tell me.
Having said that, a library is usually accompanied by a highlight definition file where the keywords to highlight are defined (awful system). It's possible something isn't quite right there.

Buzzardbait, 2020-05-11:
That's not as bad as xtensa showing you warnings in RED during compilations from .cpp libraries over which you have no control or the RED status of SPIFF ROM values which is perfectly normal action. LoL!