esp8266/Arduino #7290: Connections to WiFiServer are abruptly closed in WIFI_AP_STA mode

robertkearns, 2020-05-10:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:				ESP8266
- Core Version:			05/10/2020
- Development Env:		PlatformIO - Clion
- Operating System:		Ubuntu

### Settings in IDE

- Module:  Wesmos D1
- Flash Mode:	Unsure
- Flash Size: 	4MB
- lwip Variant:   Unsure
- Reset Method: Unsure
- Flash Frequency:	40MHz
- CPU Frequency:	80MHz
- Upload Using:     Serial
- Upload Speed:   115200

### Problem Description
A wifi event is triggering all connections to be closed. This only happens when ```WiFi.begin()``` is given bogus connection credentials. It is not the desired behaviour to have server connections on the devices AP be closed when the STA is unable to connect.

To reproduce:

```c++
#include <Arduino.h>
#include <ESP8266WiFi.h>

WiFiServer server {80};

void setup() {
    Serial.begin(115200);
    WiFi.mode(WIFI_AP_STA);
    WiFi.disconnect();  // Clear any old credentials
    IPAddress local(192, 168, 5, 50);
    IPAddress gateway(192, 168, 4,9);
    IPAddress subnet(255, 255, 255, 0);
    WiFi.softAPConfig(local, gateway, subnet);
    WiFi.softAP("ESP8266-AP", __null, 12, 0, 1);
    WiFi.setAutoReconnect(false);
    server.begin();
};

void loop() {
    WiFiClient client = server.available();
    if (!client) {
        delay(1000);
        return;
    };
    Serial.println("Client connected");
    client.read();
    WiFi.begin("TJHKKHKJH", "JLKLLLLJLLJL");  // Nonsense connection credentials
    delay(20000);  // Fake 20 second delay (we are doing something to get the response)
    client.write("HTTP/1.1 200 OK\r\n");
};

```

If you are on linux, you can also use this Python script (must install nmcli ```sudo apt install network-manager```) to emulate client side behavior.

```python
import subprocess
from urllib import request

def get_device_name():
    output = subprocess.check_output(["nmcli", "-t", "dev", "status"])
    devices = output.decode().split('\n')
    for device in devices:
        if "wifi" in device:
            return device.split(':')[0]

def create_saved_network(device, ssid, name):
    connections = subprocess.check_output(["nmcli", "-t", "-f", "name", "con"])
    connections = [con.strip() for con in connections.decode().split('\n')]
    if name in connections:  # Avoid duplicates
        return
    subprocess.check_output(
        ["nmcli", "con", "add", "type", "wifi", "con-name", name, "ssid", ssid, "ifname", device],
    )

def get_current_connection():
    output = subprocess.check_output(["nmcli", "-t", "-f", "name", "con"])
    return output.decode().split('\n')[0]

def connect(name):
    subprocess.check_output(["nmcli", "c", "up", name])

if __name__ == "__main__":
    wifi_device = get_device_name()
    before_connection = get_current_connection()
    create_saved_network(wifi_device, "ESP8266-AP", "esp8266-test")
    connect("esp8266-test")
    try:
        print("Sending request")
        request.urlopen("http://192.168.5.50")
        print("Request complete")
    finally:
        connect(before_connection)
```

### Debug Messages

The relavant parts of the Serial debug messages are:

```
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
add 1
aid 1
station: 10:be:f5:fd:1f:56 join, AID = 1
wifi evt: 5
wifi evt: 9
WS:ac
:rn 115
:ref 1
WS:av status=4 WCav=115
:ref 2
:ur 2
Client connected
scandone
wifi evt: 1
STA disconnect: 201
:close
:ur 1
:dsrcv 115
WS:dis
:del
station: 10:be:f5:fd:1f:56 leave, AID = 1
rm 1
wifi evt: 6
```

And the error raised from Python is 
```
ConnectionResetError: [Errno 104] Connection reset by peer
```
Proving that the connection is indeed being closed before it get a chance to respond. 

### Possible causes

I have had a look and it appears it is the code in ```ESP8266WiFiGenericClass::_eventCallback``` that is responding to an event triggered by the failed wifi connection. 

This in turn calls ```WiFiClient::stopAll()```, killing the AP connection. The first idea for a fix that comes to mind, is creating a seperate function along the lines of ```WiFiClient::stopAllSTA()```, that filters the clients based on their origin/binding.

I am working on a system that allows bootstrapping of an IOT device, by allowing the user to connect to the AP, send wifi credentials, and then get a response on whether the connection worked.

devyte, 2020-05-10:
The esp has a single radio, but offers two interfaces on it. Under some cases, and I believe an sta failed connection attempt is one, the sta side can trigger a radio scan (this haooens inside the sdk and in closed code that is out of our reach) . That then pulls the ap side along for the ride, which in turn drops all ap side clients. Some wifi clients may recover, some may not. Then, some of the sockets on the esp are left in indeterminate states, so the only way to make sure of correct defined behavior is to close everything.
Offering both sta and ap with a single radio has this quirk, and there are limitations. It is for this reason that such devices don't usually offer both sta and ap at the same time, but offer one or the other.
If your esp app requires both sta and ap sides of the radio, then your phone application or whatever must be able to handle being dropped by the esp, and attempt reconnection of the wifi and then reconnection of the relevant sockets. If it's a webapp, then you'll have to deal with the fact that your phone or whatever may decide to give up and connect to some different wifi where the esp isn't available,in which case you'll get a failed reload.
The previous is a discussion as old as the esp. If you find something concrete that could be improved in our core, I'm all ears (or eyes as it were), but there is no complete software fix for the case. Some users have taken to implementing a dual-esp solution, where two esps connect over e. g. spi and one serves the ap while other uses the sta. 
Closing due to out of scope.

closed 2020-05-10 by devyte

robertkearns, 2020-05-10:
Thanks for the response, I was not aware this was a known limitation with the esp. 

I am curious what sort of intermediate states can the sockets typically be left in in this case, and what is the effect? I have experimented with my above suggested fix, and it seems to work well (only closing connections bound to the stations IP). I would love to look into this a bit more and see if there is anything that can be done to make this reliable.

devyte, 2020-05-11:
It's not just the esp, it's all devices that have a single radio for a particular frequency range. I believe that's the original reason why cell phones only offer either wifi or hotspot, but not both. Same with laptops. Today most devices have both a 2.4 and a 5GHz radio. In theory they could serve an ap with one and connect with the other, but I haven't seen that implemented anywhere. 
I don't remember the details of the socket thing anymore, like I said it's a discussion that is ancient. I do seem to remember some mem leak given a set of conditions, but who knows if by now that still applies. I also seem to remember an idea of maintaining different client queues, one per interface, but again I don't remember if that was pursued and discarded, pursued and left unfinished, or never pursued at all. 
You're welcome to investigate it. If you decide to do that, I suggest opening a new issue, follow the template instructions, then state your investigation intent, your criteria for ending the investigation, and then update the issue regularly with your findings, whether they are leads or dead ends or just pings to let us know you're still working on it (we regularly check issues for time since last update and close stale ones). 
If you reach something useful, then please make a PR for testing the idea and general discussions. It doesn't have to be a finalized proposal, in which case please mark it as draft or add WIP in the title. If it doesn't pan out, it can just be closed. If it does, it will be reviewed.

d-a-v, 2020-05-11:
What @devyte refers to seems to be a behavior since the early days of this core.
As stability has improved a bit since then, maybe @robertkearns you are right thinking that something can be done.

> This in turn calls WiFiClient::stopAll(), killing the AP connection. The first idea for a fix that comes to mind, is creating a seperate function along the lines of WiFiClient::stopAllSTA(), that filters the clients based on their origin/binding.

I agree with and I like this idea. I had already though about it some time ago.
There is no reason to close the other connections when they can be left opened. TCP is robust enough to go on with them even if the other (AP) interface should disappear for a short time while making STA permanently stop.
The idea would be to cycle through all opened connections and only close those which have an STA-area IP address, leaving the others alone (or better: if a connection is linked to an up-interface, or to the default interface if it is up).
Such feature should not be limited to STA but to any interface (thinking of the incoming ethernet PR).
I will add it to #6680 is no other PR comes before then.

> I would love to look into this a bit more and see if there is anything that can be done to make this reliable.

The idea is to, for every opened connection, determine if the network address is linked to the closing interface (there are lwIP's function to help comparing addresses and netmask) ***or*** when there is no match, if the default interface is closing. 
Maybe better, the lwIP's function [`tcp_route()`](http://git.savannah.nongnu.org/cgit/lwip.git/tree/src/core/tcp_out.c?h=STABLE-2_1_2_RELEASE#n130) can be used to do this search for us.

This issue might be reopened and addressed for release 3.0 don't you think @devyte ?

devyte, 2020-05-11:
@d-a-v I think it's a bit more complicated than what you're thinking, let's discuss internally.
Given the pov of this report, and that there is investigation involved for the idea, I think it's cleaner to open a new issue and explain the intent straight up: mitigate the closing of all connections by filtering based on interface.

robertkearns, 2020-05-11:
Okay I will create a new issue for this, detailing the proposed changes. I also have a basic function written up that does exactly what I talked about earlier (only killing the stations connections). Building on what @d-a-v has mentioned, it may make sense to have a parameter ```WiFiClient::stopAllExcept(IPAddress localAddress)```, wherein we only remove connections where ```localAddress``` does not match ```getPCB()->local_ip```. That could then be refactored into the right places where it makes sense. Am I thinking right here?

devyte, 2020-05-11:
@robertkearns I think it makes more sense to have a function that stops all connections involved with a particular interface. There are more interfaces coming in addition to sta and ap.