esp8266/Arduino #7297: Delayed or no connection to WiFi

dmelgar, 2020-05-11:
----------------------------- Delete below -----------------------------

If your issue is a general question, starts similar to "How do I..", is related to 3rd party libs, or is related to hardware, please discuss at a community forum like esp8266.com.

INSTRUCTIONS
============
If you do not follow these instructions, your issue may be dismissed.

1. Follow the checklist under Basic Infos and fill in the [ ] spaces with an X.
2. Fill in all the fields under Platform and Settings in IDE marked with [ ] (pick the correct option for you in each case, delete the others).
3. If you haven't already done so, test your issue against current master branch (aka latest git), because it may have been already fixed.
4. Describe your problem.
5. If you have a STACK DUMP decode it:

https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/stack_dump.html

6. Include a Minimal Complete Reproducible Example sketch that shows your issue. Do not include your entire project, or a huge piece of code.
7. Include debug messages:

https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/debugging.html 

8. Use markup (buttons above) and the Preview tab to check what the issue will look like.
9. Delete these instructions from the above to the below marker lines before submitting this issue.

----------------------------- Delete above -----------------------------

### Basic Infos

- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [X] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [X] I have filled out all fields below.

#### Platform

- Hardware:				[ESP-8266 Nodemcu]
- Core Version:			[release 2.7.1]
- Development Env:		[Arduino IDE|Platformio same behavior on both]
- Operating System:		[MacOS]

### Settings in IDE

- Module:  			[Nodemcu]
- Flash Mode:			[qio|dio|other]
- Flash Size: 		[4MB]
- lwip Variant:   [v1.4|v2 Lower Memory|Higher Bandwidth]
- Reset Method:		[ck|nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[OTA|SERIAL - same behavior]
- Upload Speed:   [500000] (serial upload only)

### Problem Description

Detailed problem description goes here.
Attempting to connect to ASUS RT-AC86U router running latest firmware.

Does not connect to WiFi, or connects after extended period of time. Often appears to time out at 4 minutes, uses self assigned IP address. 

This problem does not occur with version 2.5.1 of this same library. Only occurred after updating library. If I go back to 2.5.1 it connects within a couple of seconds. 

Router using WAP2-personal. 
- channel bandwidth 20/40mhz
- auto control channel, currently 8
All other settings set to default.

Returns connection status: 6

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <Arduino.h>

void setup() {
 Serial.begin(500000);
  Serial.setTimeout(2000);
  // Wait for serial to initialize.
  while (!Serial) { }
  Serial.println("Serial debug should be running");
  // Connect to WiFi network
  WiFi.mode(WIFI_STA);
  wifiConnectHandler = WiFi.onStationModeGotIP(onWifiConnect);
  wifiDisconnectHandler = WiFi.onStationModeDisconnected(onWifiDisconnect);

  WiFi.disconnect();
  delay(1);
  WiFi.begin(ssid.c_str(), pw.c_str());

  //--------temp
  while (WiFi.waitForConnectResult() != WL_CONNECTED)
  {
    Serial.printf("Connection status: %d\n", WiFi.status());
    WiFi.printDiag(Serial);
    delay(1000);
    Serial.print(".");
  }
}

void loop() {

}

```

### Debug Messages

```
Debug messages go here
```

Tech-TX, 2020-05-11:
Works just fine for me once I fix your MCVE.  I don't have your Asus router.  
It connects within a few seconds.

```cpp
//#include <Arduino.h>  not needed for this MCVE
#include <ESP8266WiFi.h> // I'm just guessing, but you couldn't possibly have compiled without this

const char* ssid     = "your SSID";
const char* password = "your password";

void setup() {
 Serial.begin(250000);  // my D1 Mini won't run at half a million baud
  Serial.setTimeout(2000);
  // Wait for serial to initialize.
//  while (!Serial) { }  not needed, this isn't a Leonardo
  Serial.println("Serial debug should be running");
  // Connect to WiFi network
  WiFi.mode(WIFI_STA);
//  wifiConnectHandler = WiFi.onStationModeGotIP(onWifiConnect);  not defined
//  wifiDisconnectHandler = WiFi.onStationModeDisconnected(onWifiDisconnect);   not defined

//  WiFi.disconnect();   not needed
  delay(1);
  WiFi.begin(ssid, password);

  //--------temp
  while (WiFi.waitForConnectResult() != WL_CONNECTED)
  {
    Serial.printf("Connection status: %d\n", WiFi.status());
    WiFi.printDiag(Serial);
    delay(1000);
    Serial.print(".");
  }
}

void loop() {

}
```

dmelgar, 2020-05-12:
I didn't know if it contained something unusual.
Even the basic examples from this project fail in the same way. Reverting to v2.2.0 of the ESP8266 overall project of 2.5.1 of the ESP8266 wifi appears to correct the issue. It consistently connects quickly with old version. Current version makes it repeatedly try.

Is there diagnostic information I can collect to help diagnose?

devyte, 2020-05-12:
You said:
> I have searched the issue tracker for a similar issue.
 
and
> All other settings set to default.

which tells me you missed some things.
Did you erase the entire flash when changing core version?
Did you try a different phy on the esp, e. g. 802.11g instead of n?
Did you try lowering the Tx power on the esp? try <= 80% 
Did you try changing parameters on your wifi router? wmm (disable) , channel width (20 only), etc. There are several tweaks reported in various cases. 
Even then, Espressif's sdk has changed compatibility details several times over the years, and there are some router firmwares that cause trouble. In general, there isn't much we can do about compatibility here in our core.
Closing for now.

closed 2020-05-12 by devyte

d-a-v, 2020-05-12:
In addition to what @devyte said,
* that one is important: Did you erase the entire flash when changing core version?
* did you try other FW version by selecting "generic esp8266 board" ?

dmelgar, 2020-05-14:
@d-a-v I will try erasing entire flash.
I changed FW version by cloning repo from git, checking out the appropriate tag, retrieving new tool set and restarting Arduino SDK.
I also have recently moved to PlatformIO and explicitly specifying the version of the platform.
Same behavior with Arduino SDK & PlatformIO.

I had hoped that after a year of development, the platform wouldn't regress to require tweaks on the router so I have not tried those. I have had no trouble with the older version of the platform.

d-a-v, 2020-05-14:
> I will try erasing entire flash.

This option "erase all flash" is here because we have had experienced the same as you do when upgrading from an old closed-firmware to a new closed-one and this one simple action can solve it all.

> I had hoped that after a year of development, the platform wouldn't regress to require tweaks on the router so I have not tried those. I have had no trouble with the older version of the platform.

What can I say. Please read #2330, #2795, #3095, #7160

dmelgar, 2020-05-17:
I used "Erase flash: All Flash Contents" in the Arduino SDK. No change in behavior with new release.
I also attempted the equivalent function in PlatformIO, "Erase flash" task.

With current release (2.7.1), it most commonly will not connect to WiFi. Around 4 minutes in, something times out and it gives itself a self assigned IP address. I am able to connect to the ESP8266 using the self assigned IP.
Sometimes, with the current version, it will connect within a few seconds.

Is there diagnostic information I can collect?

With the older version of this library, 2.5.1, it connects almost immediately and consistently. 

Presumably unrelated, but since erasing flash, I have been unable to get PlatformIO to function properly with platform version 2.2.0 (2.5.1 of this repo). The ESP8266 crashes at startup. With latest version it doesn't crash, but behaves as above regarding WiFi connectivity.

d-a-v, 2020-05-17:
In core version 2.5.1, default FW is `2.2.1(Legacy)`.
This FW is available (but not default) in release 2.7.1 or current master.
Please try this:
- Run Arduino IDE (I cant help with PIO, please use Arduino IDE at least for this try)
- Install esp8266 Arduino core release 2.7.1 or current git master
- Select "generic esp8266 module", keep default values except these for your nodeMCU:
  - Flash mode: QIO
  - Erase Flash: All Flash Content
  - Espressif FW: 2.2.1-Legacy
- Run & Report

dmelgar, 2020-05-17:
@d-a-v That worked!!
FYI: I initially tried just using "generic esp8266 module". That still failed the same way. Most of the time, it tries to connect to WiFi for 4 minutes then reports a self assigned IP address.

After trying the other settings as well, it works successfully! I've done repeated tests. Connected in 6 seconds, 6 seconds, immediately, 3 seconds, immediately.

I'll keep tweaking to see if there's a particular setting that makes the difference. I'm also trying to replicate in PlatformIO.

Thanks!!
If these changes are required they should be added to documentation.

FYI: I've tried three different physical boards with the same behavior. NodeMCU from Amica, NodeMCU from Lolin, and the D1 mini.

LaurentiuGrama, 2020-11-20:
> @d-a-v That worked!!
> FYI: I initially tried just using "generic esp8266 module". That still failed the same way. Most of the time, it tries to connect to WiFi for 4 minutes then reports a self [...]

Hi,
Do you find the solution for the PlatformIO?

bwjohns4, 2021-03-24:
In PlatformIO you can change SDK to this version by passing build flag -D PIO_FRAMEWORK_ARDUINO_ESPRESSIF_SDK221 as described here:
https://docs.platformio.org/en/latest/platforms/espressif8266.html#sdk-version

@d-a-v I'm having similar issues that were solved by rolling back the SDK to this version. Would I be better off to change the SDK to the newest version available to ensure that I'm getting the all the latest bug fixes included also? Or would there be risk of other instabilities?

d-a-v, 2021-03-24:
> change the SDK to the newest version available

Please have a [look to the two pinned issues](https://github.com/esp8266/Arduino/issues)

SoulSlayerPT, 2021-03-26:
I am having a lot of issues with my tasmota devices that have the new ESP core version (2.7.4.9) and my asus mesh network. It loses the connection a lot of times per day. Even with only 2.4GHz Wifi on legacy mode. Can i create with arduino IDE a .bin file with Espressif FW: 2.2.1-Legacy?

Thank you!

d-a-v, 2021-03-26:
> Can i create with arduino IDE a .bin file with Espressif FW: 2.2.1-Legacy?

Yes, in the arduino IDE menu, or PIO option `-DPIO_FRAMEWORK_ARDUINO_ESPRESSIF_SDK221`.