esp8266/Arduino #7300: Problem with certs-from-mozilla.py

gjt211, 2020-05-12:
#### Platform

- Hardware:			ESP8266
- Core Version:			2.7.1
- Development Env:		[Arduino IDE 1.8.12]
- Operating System:		[Windows 7]

### Settings in IDE

- Module:  			[Generic ESP8266 Module]
- Flash Mode:			[DIO]
- Flash Size: 		[4MB/1MB]
- lwip Variant:   [v2 Lower Memory]
- Reset Method:		[nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[160MHz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200]

### Problem Description
I am trying to test the example BearSSL_CertStore1.ini and the python script fails when trying to run it on my Windows7 system.
I need to run the python script to generate the .AR file but it keeps failing.

I have installed the latest python 3.8.2.
The following is the error from the executing `py certs-from-mozilla.py` and note that the `data` directory gets created but there are no files in there.

It says file not found, but doesn't say what file (maybe it is the ca_000.der file but I am not sure). Is there something else that I need to install?
I have tried running cmd.exe as administrator and it makes no difference.
Can anyone suggest how to proceed please?

```
C:\py>py certs-from-mozilla.py
AC Camerfirma, S.A.:AC Camerfirma SA CIF A82743287:http://www.chambersign.org -> data/ca_000.der
Traceback (most recent call last):
  File "certs-from-mozilla.py", line 55, in <module>
    ssl = Popen(['openssl','x509','-inform','PEM','-outform','DER','-out', certName], shell = False, stdin = PIPE)
  File "C:\Users\username\AppData\Local\Programs\Python\Python38-32\lib\subprocess.py", line 854, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "C:\Users\username\AppData\Local\Programs\Python\Python38-32\lib\subprocess.py", line 1307, in _execute_child
    hp, ht, pid, tid = _winapi.CreateProcess(executable, args,
FileNotFoundError: [WinError 2] The system cannot find the file specified

C:\py>
```

JiriBilek, 2020-05-12:
Do you have openssl.exe on your path? It won't work without it.

earlephilhower, 2020-05-12:
As @JiriBilek said, you need to have OpenSSL in the PATH variable.

closed 2020-05-12 by earlephilhower

gjt211, 2020-05-12:
Hi @JiriBilek and @earlephilhower, thanks for your responses.
I have installed and setup openssl and added it to my path and the process now gets further than before but still fails with an error message.

It gets up to `ca_149.der` then fails. There are now 150 der certificates in the data folder, but there is no .ar file. I have included the last two dozen or so lines from the python script output. Is there some other software I need to install to make this work?

Can you again provide some guidance on this please?

```
SwissSign AG:SwissSign AG: -> data/ca_136.der
SwissSign AG:SwissSign AG: -> data/ca_137.der
SwissSign AG:SwissSign AG: -> data/ca_138.der
T-Systems International GmbH (Deutsche Telekom):T-Systems Enterprise Services GmbH:T-Systems Trust Center -> data/ca_139.der
T-Systems International GmbH (Deutsche Telekom):T-Systems Enterprise Services GmbH:T-Systems Trust Center -> data/ca_140.der
Taiwan-CA Inc. (TWCA):TAIWAN-CA:Root CA -> data/ca_141.der
Taiwan-CA Inc. (TWCA):TAIWAN-CA:Root CA -> data/ca_142.der
Telia Company (formerly TeliaSonera):Sonera: -> data/ca_143.der
Telia Company (formerly TeliaSonera):TeliaSonera: -> data/ca_144.der
TrustCor Systems:TrustCor Systems S. de R.L.:TrustCor Certificate Authority -> data/ca_145.der
TrustCor Systems:TrustCor Systems S. de R.L.:TrustCor Certificate Authority -> data/ca_146.der
TrustCor Systems:TrustCor Systems S. de R.L.:TrustCor Certificate Authority -> data/ca_147.der
Trustis:Trustis Limited:Trustis FPS Root CA -> data/ca_148.der
Web.com:Network Solutions L.L.C.: -> data/ca_149.der
Traceback (most recent call last):
  File "C:\py\certs-from-mozilla.py", line 68, in <module>
    call( arCmd )
  File "C:\Users\GlendonT\AppData\Local\Programs\Python\Python38-32\lib\subprocess.py", line 340, in call
    with Popen(*popenargs, **kwargs) as p:
  File "C:\Users\GlendonT\AppData\Local\Programs\Python\Python38-32\lib\subprocess.py", line 854, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "C:\Users\GlendonT\AppData\Local\Programs\Python\Python38-32\lib\subprocess.py", line 1307, in _execute_child
    hp, ht, pid, tid = _winapi.CreateProcess(executable, args,
FileNotFoundError: [WinError 2] The system cannot find the file specified

C:\py>

```

JiriBilek, 2020-05-13:
You need an ar.exe as well. It is located in tools/xtensa-lx106-elf/xtensa-lx106-elf/bin folder. You can append this folder to the PATH or simply copy the ar.exe to the folder with the script.
I don't intend to be rude but why are you not reading the error messages? ;-)

gjt211, 2020-05-13:
Hi @JiriBilek, thanks. There is no mention of ar.exe in the error message. I used over 2 hours today trying to search for ar, python ar, python arCmd and more in Google, just about all results had to do with 'Augmented reality' which is not what I am looking for (even with negative keywords I was unable to find helpful results). I also tried searching for the entire line from the python script and the results only led me to the python file itself or a change that was recently committed for osx compatibility. How am I to know that an executable file is needed?

I don't think you are being rude, I appreciate your help. What I don't understand is why is this obviously basic information not provided at least as a comment/requirements at the start of the file? Surely I can't be the only person with this problem. I am at best, a very poor c programmer, and I know even less about python. I can only use examples provided and when they don't work I try to learn from that searching google and forums. Most of the time I end up working it out but this one had me stumped.

What I do appreciate greatly is the wealth of knowledge, experience, and effort that so many people (such as yourself) put in to assist people such as myself. For that I am eternally grateful, and if you were local I would offer to chat over a coffee/beer.

Thanks again

gnorbsl, 2020-09-02:
> You need an ar.exe as well. It is located in tools/xtensa-lx106-elf/xtensa-lx106-elf/bin folder. You can append this folder to the PATH or simply copy the ar.exe to the folder with the script.
> I don't intend to be rude but why are you not reading the error messages? ;-)

I don't want to be rude but why don't you document that important bit of information properly?

devyte, 2020-09-02:
@gnorbsl ar.exe is the toolchain archiver, its presence and usage is as standard as using gcc.exe or g++.exe. Documenting how to use the compiler toolchain, as well as how to use the PATH env var, is far outside the scope of the core hosted in this repo. That is homework for whoever needs it. There are more than enough tutorials available online, just use Google.

earlephilhower, 2020-09-02:
I think many folks are using Linux or a Mac with devtools so already have `ar` in their paths, so it doesn't come up except when run on Windows.

A pull request with a documentation note would be a good idea, since this seems important to you.

gnorbsl, 2020-09-02:
i just tried it on mac and linux before I landed here and I got the message in both cases.

devyte, 2020-09-02:
@gnorbsl sure thing. Please write up your findings and propose the document or comments in a PR.