esp8266/Arduino #7301: ESP8266 resetting while generating ec_keygen function

ps108, 2020-05-12:
Hi,
I am executing the following code to generate EC keys from esp itself. I am using Arduino 1.8.12.
The flash is proper and the Arduino settings are also mentioned below
CPU Freq: 160Mhz
Vtables: Flash

I am executing the exact same code from the test_crypto.c file as mentioned in the project src code.
https://github.com/earlephilhower/bearssl-esp8266/blob/master/test/test_crypto.c
ESP is resetting and unable to generate any private key.

#include <bearssl/bearssl.h>

String name = "EC_c25519_m15";

static void test_EC_prime_i15(void)
{
  test_EC_keygen("EC_prime_i15", &br_ec_prime_i15,
    (uint32_t)1 << BR_EC_secp256r1
    | (uint32_t)1 << BR_EC_secp384r1
    | (uint32_t)1 << BR_EC_secp521r1);
}

static void test_EC_keygen(const char *name, const br_ec_impl *impl, uint32_t curves)
{
  int curve;
  br_hmac_drbg_context rng;

  printf("Test %s keygen: ", name);
  fflush(stdout);

  br_hmac_drbg_init(&rng, &br_sha256_vtable, "seed for EC keygen", 18);
  br_hmac_drbg_update(&rng, name, strlen(name));

  for (curve = -1; curve <= 35; curve ++) {
    br_ec_private_key sk;
    br_ec_public_key pk;
    unsigned char kbuf_priv[BR_EC_KBUF_PRIV_MAX_SIZE];
    unsigned char kbuf_pub[BR_EC_KBUF_PUB_MAX_SIZE];

    if (curve < 0 || curve >= 32 || ((curves >> curve) & 1) == 0) {
      if (br_ec_keygen(&rng.vtable, impl,
        &sk, kbuf_priv, curve) != 0)
      {
        fprintf(stderr, "br_ec_keygen() did not"
          " reject unsupported curve %d\n",
          curve);
        exit(EXIT_FAILURE);
      }
      sk.curve = curve;
      if (br_ec_compute_pub(impl, NULL, NULL, &sk) != 0) {
        fprintf(stderr, "br_ec_keygen() did not"
          " reject unsupported curve %d\n",
          curve);
        exit(EXIT_FAILURE);
      }
    } else {
      size_t len, u;
      unsigned char tmp_priv[sizeof kbuf_priv];
      unsigned char tmp_pub[sizeof kbuf_pub];
      unsigned z;

      len = br_ec_keygen(&rng.vtable, impl,
        NULL, NULL, curve);
      if (len == 0) {
        fprintf(stderr, "br_ec_keygen() rejects"
          " supported curve %d\n", curve);
        exit(EXIT_FAILURE);
      }
      if (len > sizeof kbuf_priv) {
        fprintf(stderr, "oversized kbuf_priv\n");
        exit(EXIT_FAILURE);
      }
      memset(kbuf_priv, 0, sizeof kbuf_priv);
      if (br_ec_keygen(&rng.vtable, impl,
        NULL, kbuf_priv, curve) != len)
      {
        fprintf(stderr, "kbuf_priv length mismatch\n");
        exit(EXIT_FAILURE);
      }
      z = 0;
      for (u = 0; u < len; u ++) {
        z |= kbuf_priv[u];
      }
      if (z == 0) {
        fprintf(stderr, "kbuf_priv not initialized\n");
        exit(EXIT_FAILURE);
      }
      for (u = len; u < sizeof kbuf_priv; u ++) {
        if (kbuf_priv[u] != 0) {
          fprintf(stderr, "kbuf_priv overflow\n");
          exit(EXIT_FAILURE);
        }
      }
      if (br_ec_keygen(&rng.vtable, impl,
        NULL, tmp_priv, curve) != len)
      {
        fprintf(stderr, "tmp_priv length mismatch\n");
        exit(EXIT_FAILURE);
      }
      if (memcmp(kbuf_priv, tmp_priv, len) == 0) {
        fprintf(stderr, "keygen stutter\n");
        exit(EXIT_FAILURE);
      }
      memset(&sk, 0, sizeof sk);
      if (br_ec_keygen(&rng.vtable, impl,
        &sk, kbuf_priv, curve) != len)
      {
        fprintf(stderr,
          "kbuf_priv length mismatch (2)\n");
        exit(EXIT_FAILURE);
      }
      if (sk.curve != curve || sk.x != kbuf_priv
        || sk.xlen != len)
      {
        fprintf(stderr, "sk not initialized\n");
        exit(EXIT_FAILURE);
      }

      len = br_ec_compute_pub(impl, NULL, NULL, &sk);
      if (len > sizeof kbuf_pub) {
        fprintf(stderr, "oversized kbuf_pub\n");
        exit(EXIT_FAILURE);
      }
      memset(kbuf_pub, 0, sizeof kbuf_pub);
      if (br_ec_compute_pub(impl, NULL,
        kbuf_pub, &sk) != len)
      {
        fprintf(stderr, "kbuf_pub length mismatch\n");
        exit(EXIT_FAILURE);
      }
      for (u = len; u < sizeof kbuf_pub; u ++) {
        if (kbuf_pub[u] != 0) {
          fprintf(stderr, "kbuf_pub overflow\n");
          exit(EXIT_FAILURE);
        }
      }
      memset(&pk, 0, sizeof pk);
      if (br_ec_compute_pub(impl, &pk,
        tmp_pub, &sk) != len)
      {
        fprintf(stderr, "tmp_pub length mismatch\n");
        exit(EXIT_FAILURE);
      }
      if (memcmp(kbuf_pub, tmp_pub, len) != 0) {
        fprintf(stderr, "pubkey mismatch\n");
        exit(EXIT_FAILURE);
      }
      if (pk.curve != curve || pk.q != tmp_pub
        || pk.qlen != len)
      {
        fprintf(stderr, "pk not initialized\n");
        exit(EXIT_FAILURE);
      }

      if (impl->mulgen(kbuf_pub,
        sk.x, sk.xlen, curve) != len
        || memcmp(pk.q, kbuf_pub, len) != 0)
      {
        fprintf(stderr, "wrong pubkey\n");
        exit(EXIT_FAILURE);
      }
    }
    printf(".");
    fflush(stdout);
  }

  printf(" done.\n");
  fflush(stdout);
}

void setup() {
  Serial.begin(115200);
  Serial.setDebugOutput(true);
  
  printf("ok here 1 with: %d\n\n", ESP.getFreeHeap());

  test_EC_prime_i15();
}

void loop() {
  // put your main code here, to run repeatedly:
  delay(1000);
}

Following is the stack trace I am getting

ok here 1 with: 52232

Test EC_prime_i15 keygen: ........................Fatal exception 3(LoadStoreErrorCause):
epc1=0x40216e9e, epc2=0x00000000, epc3=0x00000000, excvaddr=0x4023d1a8, depc=0x00000000

Exception (3):
epc1=0x40216e9e epc2=0x00000000 epc3=0x00000000 excvaddr=0x4023d1a8 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffb20 end: 3fffffc0 offset: 01a0
3ffffcc0:  00000020 4023d350 3ffffe24 3ffffe24  
3ffffcd0:  4023ce80 00000000 3ffef470 40204b3d  
3ffffce0:  4010038e 00000017 3ffe82fc 3ffffe24  
3ffffcf0:  4023ce80 00000017 3ffe82fc 40201128  
3ffffd00:  40003b53 00000030 00000010 ffffffff  
3ffffd10:  40002bf0 00000020 00018000 3ffef5ac  
3ffffd20:  00ff0000 5ffffe00 5ffffe00 00000000  
3ffffd30:  00018000 4023d50d 00000005 3ffef5af  
3ffffd40:  00000016 3ffef5c2 3ffef5ac 00000030  
3ffffd50:  feefeffe feefeffe feefeffe feefeffe  
3ffffd60:  feefeffe feefeffe feefeffe feefeffe  
3ffffd70:  feefeffe feefeffe feefeffe feefeffe  
3ffffd80:  feefeffe feefeffe feefeffe feefeffe  
3ffffd90:  feefeffe feefeffe feefeffe feefeffe  
3ffffda0:  feefeffe feefeffe feefeffe feefeffe  
3ffffdb0:  00000005 00000000 0000000a 00000000  
3ffffdc0:  00000005 00000000 0000000a 00000000  
3ffffdd0:  4010367b 00040000 00000000 00040000  
3ffffde0:  00000000 a0000000 00000000 0000001c  
3ffffdf0:  00005000 401025df 00005000 0000a000  
3ffffe00:  40100338 00000000 00005000 00000001  
3ffffe10:  00000001 3ffe8300 3ffef470 40206af8  
3ffffe20:  40102430 4023d48c eef86122 f6e36752  
3ffffe30:  8248b4ae c3c33b05 27424425 0e15ce3f  
3ffffe40:  136cc09e 3d9c516f 3ffffe70 3ffe8300  
3ffffe50:  0000000a 3ffe8300 3ffef470 40204b3d  
3ffffe60:  3ffef470 3ffe8300 557a5fde ff041e88  
3ffffe70:  b00c148a 554cd450 15a555fd 804ed434  
3ffffe80:  3a433164 c7ecf48a 0000000a 40206011  
3ffffe90:  3ffe8820 00000005 00000000 3ffe880c  
3ffffea0:  3ffe8300 3ffe8300 4023d350 3ffe881e  
3ffffeb0:  3ffe8820 3ffe8300 3ffef470 40206198  
3ffffec0:  00000000 ffffffff ffffffff 00000000  
3ffffed0:  00000005 00000017 fe302064 3ffe8300  
3ffffee0:  00000004 40204c45 00000020 40100a10  
3ffffef0:  feefeffe feefeffe feefeffe 32350100  
3fffff00:  00323332 3ffef2dc 000001ac 402026de  
3fffff10:  40202bad 00000138 00000000 40204c45  
3fffff20:  3fffff80 3fffff70 00000008 00000005  
3fffff30:  00000002 00000000 00000000 3ffee690  
3fffff40:  00000001 00000000 00000016 3ffee690  
3fffff50:  3fffdad0 3ffe880c 3ffe8300 402069a2  
3fffff60:  3fffff80 3fffff70 00000004 40201608  
3fffff70:  00000000 0000cc08 00000001 5ffffe00  
3fffff80:  3fffde60 00000000 3ffee574 3ffee690  
3fffff90:  3fffdad0 00000000 3ffee574 40201426  
3fffffa0:  feefeffe feefeffe 3ffee650 40201da0  
3fffffb0:  feefeffe feefeffe 3ffe84f4 40100c9d  
<<<stack<<<

ps108, 2020-05-12:
Below is Decoded Stack Trace for refefence
Exception 3: LoadStoreError: Processor internal physical address or data error during load or store
PC: 0x40216e9e: br_ec_keygen at src/ec/ec_keygen.c line 44
EXCVADDR: 0x4023d1a8

Decoding stack results
0x40204b3d: _fflush_r at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/fflush.c line 282
0x4010038e: putchar(int) at /Users/admin/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/cores/esp8266/libc_replacements.cpp line 123
0x40201128: test_EC_keygen(uint32_t, br_ec_impl const*, char const*) at /Volumes/Untitled 2/Prashant/studies/arduino/ecc/beasssltest/bearssltest/bearssltest.ino line 54
0x40100338: _write_r(_reent*, int, char*, int) at /Users/admin/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/cores/esp8266/libc_replacements.cpp line 90
0x40206af8: __swrite at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/stdio.c line 105
0x40204b3d: _fflush_r at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/fflush.c line 282
0x40206011: __sfputs_r at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/nano-vfprintf.c line 433
0x40206198: _vfprintf_r at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/nano-vfprintf.c line 531
0x40204c45: __sfmoreglue at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/findfp.c line 100
0x40100a10: malloc(size_t) at /Users/admin/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/cores/esp8266/umm_malloc/umm_malloc.cpp line 511
0x402026de: _malloc_r(_reent*, size_t) at /Users/admin/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/cores/esp8266/heap.cpp line 123
0x40202bad: uart_flush(uart_t*) at /Users/admin/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/cores/esp8266/uart.cpp line 544
0x40204c45: __sfmoreglue at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/findfp.c line 100
0x402069a2: printf at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/printf.c line 61
0x40201608: HardwareSerial::begin(unsigned long, SerialConfig, SerialMode, unsigned char, bool) at /Users/admin/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/cores/esp8266/HardwareSerial.cpp line 51
0x40201426: setup() at /Volumes/Untitled 2/Prashant/studies/arduino/ecc/beasssltest/bearssltest/bearssltest.ino line 172
0x40201da0: loop_wrapper() at /Users/admin/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/cores/esp8266/core_esp8266_main.cpp line 177

devyte, 2020-05-12:
That code is for host side execution, i. e. with the backing of a real OS, for unit testing of the code. The exit() and fprintf to stderr should make that obvious. It is not meant to run on the esp.
That doesn't necessarily mean ot can't run on the esp, just that it needs to be adapted. 
Closing due to not an issue.

closed 2020-05-12 by devyte

ps108, 2020-05-13:
I have tried with simple execution on ESP too with directly calling the 
br_ec_keygen(&rng.vtable, impl, &sk, kbuf_priv, curve) function from my main code. It is still crashing at same point. From what I could gather is that line 44 of ec_keygen.c

https://github.com/arduino-libraries/ArduinoBearSSL/blob/master/src/bearssl/ec_keygen.c
It seems like some not found pointer location issue. I have tried all possible migrations to run this code in esp. It is also a part of main sdk.

devyte, 2020-05-13:
Bearssl is NOT part of the sdk.
I suggest checking for oom (debug options, then check the serial output for oom msg).

earlephilhower, 2020-05-13:
@ps108, while what you're doing can probably be made to work it's not part of the BSSL usage on the 8266 so you're on your own getting it working.

My guess is that you're overrunning the main stack.  Look at the BSSL stack thunk code on how to do this, and be aware that the currently allocated BSSL stack may not be enough to make that call work.

But, please, move this kind of question to https://esp8266.com or https://gitter.im/esp8266/Arduino as this is not the appropriate spot.

ps108, 2020-05-14:
Thanks for your inputs.
Will work another method for this.