esp8266/Arduino #7304: DHCP IP address reservation support

tloki, 2020-05-15:
### Basic Infos

- [X ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [X ] I have searched the issue tracker for a similar issue.

### Problem Description

I think it would be usefull to have option of DHCP IP address reservations.

### [MCVE] - how it might look like

```cpp

#include <Arduino.h>
#include <ESP8266WiFi.h>

const char* ssid = "wifi_ssid";
const char* password = "wifi_pass";

IPAddress staticIP(10,0,3,1);
IPAddress gateway(10,0,3,1);
IPAddress subnet(255,255,255,0);

// set up of our reserved client
uint8_t importantClientMAC[6] = [0x5e, 0xcf, 0x7f, 0x8b, 0x10, 0x13];
IPAddress importantClientIP(10,0,3,14);

void setup() {
  Serial.begin(115200);
  Serial.println();
  

  WiFi.softAPConfig(staticIP, gateway, subnet);
  
  // an example of how it might work
  WiFi.addReservedIP(importantClientMAC , importantClientIP);

  WiFi.softAP(ssid, pass);
}

void loop() {
}

```

This has many use cases - for instance when your IoT device does not support static IP address setup etc.

d-a-v, 2020-05-15:
This is already provided in #5594.
Further work on DHCP server can happen when [#6680](https://github.com/esp8266/Arduino/pull/6680/files#diff-4b4e9efdf71444364dcd453931632f3f) is merged.

tloki, 2020-05-15:
As far as I can tell, you cannot specify address (easily) - but yeah, this does indeed help use case I described. Closing.

closed 2020-05-15 by tloki