esp8266/Arduino #7305: Using SPIFFS for certificates is causing delay stop working and WDT reset (MQTT TLS connection)

Zico56, 2020-05-15:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:				[ESP-12]
- Core Version:			[2.7.1]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Nodemcu]
- Flash Mode:			[qio|dio|other]
- Flash Size: 		[4MB]
- lwip Variant:   [v2 Lower Memory]
- Reset Method:		[nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200)

### Problem Description
Hello,
I have stored certificates files (pem format) on ESP8266 flash.
I managed to load them using SPIFFS and initialize WifiClientSecure (cf. code below).
But when trying to connect, TCP connection failed and **delay() instruction stops working**...causing "**WDT reset**" (probably because of the connection loop is going endless because connection is not ok).

I tried to add yield() instruction. This cause "User exception (panic/abort/assert) / Panic core_esp8266_main.cpp:133 __yield".

I tried limiting number connection attemps in loop with counter. Result is delay() is still not working. ESP is going to deepSleep(0) (last sketch instruction) **but wakup by itselft with WDT reset reason**. 

When storing certificates data as PROGMEM constants, all is working pretty well.

I logged the free heap size (cf. logs below), but seems that loading certificates from files doesn't affect it so much.

Is it bug from BearSLL, which may not support data not stored as PROGMEM ?
Is it due buffer size when loading files ? But why free heap size would be ok then ?

I tried lot of things but do not manage to find where the issue is exactly coming from.
Thanks in advance for your help.

**For context:**
I don't want to use PROGMEM hardcoded certificates but need to find a way to store them certificates as files (pem format). I use AWS IoT service with ["fleet provisionning with device that dont have device certificates"](https://docs.aws.amazon.com/iot/latest/developerguide/provision-wo-cert.html)
Goal is having 1st set of certificates for provisionning, onboarded when making device. Then AWS generates and returns specific certificate for device (not shared with other IoT) on JSON format.
I want to save this return as certificates files on ESP8266 flash.
Moreover, having externalized files instead of hardcoded constant is better device industrialization process.

### Sketch

```cpp

void setup() {  
  Serial.begin(115200);

/*---- doing wifi initialization--------*/

  Serial.println(F("Setting X509 time...")); 
  WiFiUDP ntpUDP;
  NTPClient timeClient(ntpUDP, "pool.ntp.org");
  timeClient.begin();
  while(!timeClient.update()){
    timeClient.forceUpdate();
  }
[... 1806 lines cut ...]
09:54:53.261 -> 3fff3180:  6e6f6320 7463656e 206e6f69 6f727265  
09:54:53.261 -> 3fff3190:  30203a72 00000000 00000000 a5a5a5a5  
09:54:53.261 -> 3fff31a0:  063e064d 000007fa a5f4d400 a5a5a5a5  
09:54:53.261 -> 3fff31b0:  063e064d 05a807d8 b7c4b482 20abae0f  
09:54:53.261 -> 3fff31c0:  fe21cf3d bc4afb95 344c0973 7f8ab4a1  
09:54:53.261 -> 3fff31d0:  b9758ee6 76ba7c2b 1fb7e0c5 00000000  
09:54:53.295 -> 3fff31e0:  a5a5a5a5 39774463 063e06b8 00000351  
09:54:53.295 -> 3fff31f0:  a5a5a5a5 41038230 29028230 010203a0  
09:54:53.295 -> 3fff3200:  06130202 99cf9f6c 390a8cbf 8a78f0e2  
09:54:53.295 -> 3fff3210:  3696e643 0d30ca5b 862a0906 0df78648  
09:54:53.295 -> 3fff3220:  050b0101 31393000 0609300b 06045503  
09:54:53.295 -> 3fff3230:  53550213 0d300f31 04550306 4106130a  
09:54:53.295 -> 3fff3240:  6f7a616d 3019316e 55030617 10130304  
09:54:53.295 -> 3fff⸮
09:54:53.295 ->  ets Jan  8 2013,rst cause:4, boot mode:(3,6)
09:54:53.329 -> 
09:54:53.329 -> wdt reset
09:54:53.329 -> load 0x4010f000, len 3456, room 16 
09:54:53.329 -> tail 0
09:54:53.329 -> chksum 0x84
09:54:53.329 -> csum 0x84
09:54:53.329 -> va5432625
09:54:53.329 -> ~ld
09:54:53.396 -> 
```

d-a-v, 2020-05-15:
Dump without decoding are useless.
Try to `#include <LittleFS.h>` and replace all occurrences of `SPIFFS` by `LittleFS` (your filesystem will need to be repopulated).

Zico56, 2020-05-15:
Both modifications (LittleFS & frequency) have been done.
Issue is still there.

I've updated description with dump decoding.

d-a-v, 2020-05-15:
I would
- show `ESP.getMaxFreeBlockSize()`,
- enable debug options,
- remove big objects from stack (like the private key),
- check how SSL examples are written (`new` is used when key is a local var)

Zico56, 2020-05-15:
- Made a new test with "ESP.getMaxFreeBlockSize()".

It displays before loop and crash (rest of logs is same as previous test)
```
14:58:31.761 -> system_get_free_heap_size():36824 / ESP.getMaxFreeBlockSize():36112
14:58:31.966 -> system_get_free_heap_size():33440 / ESP.getMaxFreeBlockSize():33104
14:58:31.966 -> system_get_free_heap_size():33440 / ESP.getMaxFreeBlockSize():33104
14:58:31.966 -> MQTT initialization...
14:58:31.966 -> system_get_free_heap_size():33376 / ESP.getMaxFreeBlockSize():33104
```

- Debug options were already activated on previous provided logs (SSL + TLS_MEM).
Which other logs level would like me to add ?

- Removing big objects was already tested (and with smaller values too). It run normally in that case (except that connection normally fails because of wrong key). But goal is to read those values from files. So, cannot redure size of these objects. Heap size seems to show that it is not full....however, size of the arrays seems to be source of the problem. But where is the link ? 

- Sorry, i did not understand your latest point ?

By the way, just for information, I made equivalent test on a ESP32, it works.

d-a-v, 2020-05-15:
My last point is your private keys stored into the stack.
Can you make it a pointer w/`new` or move it to global ?
If your key is too large it might not fit into the stack (like in the BearSSL examples).

devyte, 2020-05-15:
@d-a-v is correct, this:
```
File crtFile = SPIFFS.open(AWS_CERTIFICATE, "r");
char crtBuffer[crtFile.size()];
```
is likely blowing up the stack. 
Closing xue to user error.

closed 2020-05-15 by devyte