esp8266/Arduino #7308: LittleFs fails to read files/list dirs created by spiffs

WhiteLionATX, 2020-05-18:
I wanted to switch from spiffs to littleFS ESP8266 / 12E.
While spiffs works fine (tested list root dir and read a files) littleFs failed on this.

Here is my code (Its nearly the same as here on github***) :

`
void setup() 
{ 
  Serial.begin(9600);
  Serial.println("Woke up !");

  // works fine: -> lists files
  spiffsListAllFiles();#
  // does not work: -> zero files are listed
  littleFsListDir("/");

  // works fine: -> returns the file as string
  littleFsReadFile("/settings.json");
  // does not work: -> "Failed to open file for reading"
  littleFsReadFile("/settings.json");
}

void littleFsListDir(const char * dirname) 
{
  Serial.printf("Listing directory: %s\n", dirname);

  Dir root = LittleFS.openDir(dirname);

  while (root.next()) 
  {
    File file = root.openFile("r");
    Serial.print("  FILE: ");
    Serial.print(root.fileName());
    Serial.print("  SIZE: ");
    Serial.print(file.size());
    time_t cr = file.getCreationTime();
    time_t lw = file.getLastWrite();
    file.close();
    struct tm * tmstruct = localtime(&cr);
    Serial.printf("    CREATION: %d-%02d-%02d %02d:%02d:%02d\n", (tmstruct->tm_year) + 1900, (tmstruct->tm_mon) + 1, tmstruct->tm_mday, tmstruct->tm_hour, tmstruct->tm_min, tmstruct->tm_sec);
    tmstruct = localtime(&lw);
    Serial.printf("  LAST WRITE: %d-%02d-%02d %02d:%02d:%02d\n", (tmstruct->tm_year) + 1900, (tmstruct->tm_mon) + 1, tmstruct->tm_mday, tmstruct->tm_hour, tmstruct->tm_min, tmstruct->tm_sec);
  }
v

String littleFsReadFile(const char * path) 
{
  Serial.printf("Reading file: %s\n", path);

  String res = "";
  File file = LittleFS.open(path, "r");
  if (!file) 
  {
    Serial.println("Failed to open file for reading");
    return res;
  }

  Serial.print("Read from file: ");
  res = file.readString();
  file.close();
  Serial.print(res);
  Serial.print("------------------");  
  return res;
}
`

***[(https://github.com/esp8266/Arduino/blob/master/libraries/LittleFS/examples/LittleFS_Timestamp/LittleFS_Timestamp.ino)]

earlephilhower, 2020-05-18:
You have both SPIFFS and LittleFS running at the same time in your incomplete snippet.  They use the exact same flash, so it's not going to work, period.  One FS will corrupt the other and vice-versa.  Only one FS at a time is allowed, of course.

closed 2020-05-18 by earlephilhower

WhiteLionATX, 2020-05-18:
Thx for your fast answer. The snippet was not poted correctly. Of course SPIFFS lines where commented out (to avoid the use of SPIFFS and LittleFS at the same time).  As I mention: the example didnt worked. Now I found the reason: I accidentally had "build_flags = -Wl,-Teagle.flash.4m3m.ld" as you can see in the first post. The solution is to change it to "build_flags = -Teagle.flash.4m3m.ld".