esp8266/Arduino #7315: Bootloader appears to have run at either 74880 or 115200

pfeerick, 2020-05-19:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:				Wemos D1 Mini
- Core Version:			2.7.1
- Development Env:		PlatformIO
- Operating System:		Manjaro Linux

### Settings in IDE

- Module:  			Wemos D1 mini 
- Flash Mode:			default (DIO)
- Flash Size: 			4MB
- lwip Variant:   		
- Reset Method:		
- Flash Frequency:		40Mhz
- CPU Frequency:		80Mhz or 160MHz
- Upload Using:		SERIAL
- Upload Speed:   		460800

### Problem Description

It appears the bootloader (eboot?) has two different serial baud rates depending upon whether the system self-resets via `ESP.reset();/ESP.restart();` or via a hard reset. On hard reset, it runs at 74880, but on a self reset it runs at 115200. Making it so you can't always just run at 74880 if you want to capture boot messages. I also ran at 9600, and the bootloader messages were garbled both for hard and self-reset, thus it's not following the last set baud. 

Intended / unavoidable behaviour, or a bug?

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <Arduino.h>

void setup() {
//  Serial.begin(9600);
  Serial.begin(74880);
//  Serial.begin(115200);
  Serial.println();
  Serial.println("Init!");
}

void loop() {
  while(millis() < 5000)
  {
    delay(1);
  }

  ESP.reset();
//  ESP.restart();
}
```

### Debug Messages

```
Debug messages go here
```

earlephilhower, 2020-05-20:
That's the expected behavior.  The "load...tail...cksum..." dump you get on power-up is from the ROM, not our eboot, so there's nothing we can do.  Eboot only dumps the ~1-char status before jumping to the app.

closed 2020-05-20 by earlephilhower

pfeerick, 2020-05-20:
So 74880 baud messaging is from the ROM, and 115200 baud messaging is from eboot? Gotcha. Thanks! :)