esp8266/Arduino #7316: Strange WiFiClient.remoteIP() == 0

kirillandy, 2020-05-19:
### Basic Infos

- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [X] I have filled out all fields below.

#### Platform

- Hardware:				[ESP8266]
- Core Version:			[2.6.3 (I think...)]
- Development Env:		[Visual Micro (Visual Studio)]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Generic ESP8266 Module]
- Flash Mode:			[dout]
- Flash Size: 		[1MB]
- lwip Variant:   [v2 Lower Memory]
- Reset Method:		[nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200]

### Problem Description

I stumbled upon this strange occurrence which I wouldn't call a bug just yet. The main problem is that I have not been able to reproduce it. My solution that I'm working on consists of a couple of projects and a good few dozen files so there is no way I will be able to somehow post all that here. When I tried to force the same type of behavior in a simple sketch, nothing gave the result I wanted, so I apologize for not providing an MCVE.
I was hoping somebody might have a general idea of why this is happening. 
Basically: I've got a WiFiClient sending data (10-20 bytes/session) to a WiFiServer (so I have 2 identical boards). The client simply connects, sends and disconnects by itself. The server-side never calls .stop(), so when I do server.available(), there is a chance that the WiFiClient object it returns will give a CLOSED status. However, this is not guaranteed, I could retrieve it at any moment.
Now, I heavily rely on being able to retrieve the sender's IP address which I do using the .remoteIP() function of the WiFiClient I get from server.available(). My experiments in my simple sketch have showed me that it remains intact even AFTER the client has sent everything and disconnected and even when there are a bunch of Clients in the server queue. However, in my large project, I (randomly?) get an IP of 0.0.0.0 when calling remoteIP(). Nothing happens in between. It's literally:
globalWiFiClient = globalServer.available();
if(!globalWiFiClient) {[wait for one to arrive]}
unsigned int ipAddr = globalWiFiClient.remoteIP();
Digging around in the source code (WiFiClient.cpp, WiFiServer.cpp, ClientContext.h) didn't give me any useful hints as to why this might be the case. The most I could gather was I would get a zero if ClientContext::_pcb is NULL or if &_pcb->remote_ip is 0 (if that is even possible). A level lower is the tcp_pcb structure which I simply couldn't understand (and I'm not sure if I was looking at the correct version). Has anyone had a similar occurrence or is there any straightforward reason why remoteIP would occasionally be 0 and what could be resetting it?

kirillandy, 2020-05-20:
Here are the results from my first experiment: boards A and B communicating.
Transmitting a request 50 times with a delay of X == 30 gave 1 reset IP address during session 39

Transmitting a request 100 times more gave 2 zero IPs (session 50 and 53). What's remarkable is the fact that the VERY FIRST transmission was a hiccup! How this happens I have no idea. I spent a good few seconds changing the client parameters in between the first 50 messages and the next 100 so there was enough time for things to 'settle down'.

[image]
[image]

Using 2 different boards (C and D) with a delay of 30 only gave 1 (!!!) wrong IP over ~2000 sessions (session 336, I only captured a part of them in the screenshot but I kept at it for >2000 transmissions with no further errors)
[image]

devyte, 2020-05-21:
@kirillandy your client is connect-send-closing without expecting any response. If the connection is closed before you handle it on the server, the server-side client won't have a remote side (it's already done with being notified as far as TCP is concerned), so you get a remoteIP of 0.0.0.0.
You can still read the data out of the server-side client because it's contained in the internal buffers, but at that point you can't get info on the remote client because it's already gone. 
What you see is working as intended. 
Why that happens only sometimes has to do with how you structured your code. Essentially, you have a race condition between your client and your server. 

At a glance, I see several choices for you:
- change the notify to expect a reply, change the server to send it. The client would know that the server handled the message, so can freely close the connection. 
- package the remoteIP as part of the message, so that it's available aftwr reading feom tje server-side client. 
- ignore the lack of remoteIP (maybe you don't need it?)

Closing due to not an issue.

closed 2020-05-21 by devyte

kirillandy, 2020-05-21:
Ah, I see. And to think I spent hours poking around trying to get it to give a 0 🤦 
I've modified the test a bit and made the server print out the received messages after the Client is done sending all of them and I got zeroes all across the board.
I wonder: is this also something done in other TCP APIs like WinSock? I should keep that in mind and check once I get there...

As for your possible solutions:
1) That's what I had before and it was even nastier. I have a "wifi_exchange" function that does exactly that and makes both sides send 1 message. The only problem was that it caused a really painful deadlock in the 'proper' circumstances (I won't go into the details), so I'd much rather deal with a race condition 😆
2) Yeah, that crossed my mind as soon as I got my first 0 IPs, but it felt like a patch for improper TCP behavior so I decided to determine whether this was in fact a bug before changing my message formats. It is a bit unusual to be able to get a message but to have lost the ID of the client that sent it. Seems like I might opt to do this now.
3) No, not an option. I've got a network of devices communicating with one another and they trigger physical events depending on who sent the message so it's imperative that I know who it was. Otherwise it might cause a bit of chaos 😋

Thank you for looking into this @devyte, much appreciated 👍