esp8266/Arduino #7318: NodeMCU Serial Port error message: ets Jan 8 2013, rst cause:4, boot mode:(3,6)

EdCaneda, 2020-05-20:
### Basic Infos

- [/ ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [ ] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ ] I have filled out all fields below.

#### Platform

- Hardware:                      NodeMCU 1.0, ESP-12E module
- Core Version:			I have no idea
- Development Env:		Arduino IDE 1.8.9
- Operating System:		Windows

### Settings in IDE

- Module:  			NodeMCU 1.0, ESP-12E module
- Flash Mode:			N/A
- Flash Size: 		        4MB
- lwip Variant:                   v2 Lower Memory
- Reset Method:		N/A
- Flash Frequency:	        N/A
- CPU Frequency:		80Mhz
- Upload Using:		OTA
- Upload Speed:                115200

### Problem Description
Here's the screenshot of the serial log: 
[image]

###Sketch
### Debug Messages

None
Debug messages go here
No debug done yet

pfeerick, 2020-05-20:
Examining the list of reset causes and boot modes indicates that your problem is being caused by a watchdog reset. This is confirmed by the fourth line of output stating `wdt reset`. It is likely something in your code is locking the ESP8266 up for more than 6 seconds, hence the reset. Without you providing the sketch / MVCE, it's impossible to diagnose. 

```
    reset causes:
        0:
        1: normal boot
        2: reset pin
        3: software reset
        4: watchdog reset

    boot device:
        0:
        1: ram
        3: flash
```

earlephilhower, 2020-05-20:
No MCVE present, template not followed.  Closing.

Please use https://esp8266.com or https://gitter.im/esp8266/Arduino for general questions like this.

closed 2020-05-20 by earlephilhower

EdCaneda, 2020-05-20:
> Examining the list of reset causes and boot modes indicates that your problem is being caused by a watchdog reset. This is confirmed by the fourth line of output stating `wdt reset`. It is likely something in your code is locking the ESP8266/NodeMCU up for more than 6 seconds, hence the reset. Without you providing the sketch / MCVE, it's impossible to diagnose.
> 
> ```
>     reset causes:
>         0:
>         1: normal boot
>         2: reset pin
>         3: software reset
>         4: watchdog reset
> 
>     boot device:
>         0:
>         1: ram
>         3: flash
> ```

**Hi @pfeerick , I got this code from online. It was wrong a couple of time . Sudden after few days, it returns the error.

Here's the code below:**

/*********
  Rui Santos
  Complete project details at http://randomnerdtutorials.com  
*********/

// Load Wi-Fi library
#include <ESP8266WiFi.h>

// Replace with your network credentials
const char* ssid     = "CCTN-WIFI";
const char* password = "francis123";

// Set web server port number to 80
WiFiServer server(80);

// Variable to store the HTTP request
String header;

// Auxiliar variables to store the current output state
String output5State = "off";
String output4State = "off";

// Assign output variables to GPIO pins
const int output5 = 5;
const int output4 = 4;

// Current time
unsigned long currentTime = millis();
// Previous time
unsigned long previousTime = 0; 
// Define timeout time in milliseconds (example: 2000ms = 2s)
const long timeoutTime = 2000;

void setup() {
  Serial.begin(115200);
  // Initialize the output variables as outputs
  pinMode(output5, OUTPUT);
  pinMode(output4, OUTPUT);
  // Set outputs to LOW
  digitalWrite(output5, LOW);
  digitalWrite(output4, LOW);

  // Connect to Wi-Fi network with SSID and password
  Serial.print("Connecting to ");
  Serial.println(ssid);
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
[... 80 lines cut ...]
            } else {
              client.println("<p><a href=\"/4/off\"><button class=\"button button2\">OFF</button></a></p>");
            }
            client.println("</body></html>");
            
            // The HTTP response ends with another blank line
            client.println();
            // Break out of the while loop
            break;
          } else { // if you got a newline, then clear currentLine
            currentLine = "";
          }
        } else if (c != '\r') {  // if you got anything else but a carriage return character,
          currentLine += c;      // add it to the end of the currentLine
        }
      }
    }
    // Clear the header variable
    header = "";
    // Close the connection
    client.stop();
    Serial.println("Client disconnected.");
    Serial.println("");
  }
}