esp8266/Arduino #7319: configTime(TZ,DS,server) gives wrong offset

Juppit, 2020-05-20:
- [*] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [*] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [*] I have tested that the issue is present in current master branch (aka latest git).
- [ ] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [*] I have filled out all fields below.

#### Platform

- Hardware:				[ESP-12]
- Core Version:			[2.7.0 2.7.1]
- Development Env:		[Arduino IDE]
- Operating System:		[WindowsS]

### Settings in IDE

- Module:  		[Wemos D1 mini r2]
- lwip Variant:   [v2.1]
- Upload Using:		[OTA or SERIAL]
- Upload Speed:   [19200)
- all other setings are untached/default, not visible

### Problem Description

The old configTime api with timezone and daylight in seconds produces wrong timestamp.
To generate the right timestamps one can use negative timezone an daylight setting.

This appears after core version 2.6.3.
Git version 2.7.0-dev-nightly+20200218 was still correct.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```
#include "pass.h"  // MY_SSID, MY_PWD
#include <ESP8266WiFi.h>

static int tz = -1; // start with GMT-1
static int ds = 0;  // no daylight time

static time_t now = 0;

void setup() {
  Serial.begin(19200);
  Serial.printf("\nCORE version is %s", ESP.getCoreVersion().c_str());

  WiFi.mode(WIFI_STA);
  WiFi.begin(MY_SSID, MY_PWD);
  while (WiFi.status() != WL_CONNECTED)
    delay(500);

  while (! now )
  {
    now = time(nullptr);
    delay(500);
  }
}

void printTime() {
  struct tm* t;
  now = time(nullptr);
  t = localtime((const time_t*)&now);
  Serial.printf("\nTZ: GMT%+03d DS: %i - %02d:%02d", tz, ds, t->tm_hour, t->tm_min);
}

void loop() {
  configTime ( tz * 3600, ds * 3600, (char*)"pool.ntp.org") ;
  delay(2000);
  printTime();
  if (++tz > 1)
  {
    tz = -1;
    ds = !ds;
    Serial.println();
    delay(5000);
  }
}
```

### Debug Messages

```
CORE version is 2_6_3

TZ: GMT-01 DS: 0 - 20:20
TZ: GMT+00 DS: 0 - 21:20
TZ: GMT+01 DS: 0 - 22:20

TZ: GMT-01 DS: 1 - 21:20
TZ: GMT+00 DS: 1 - 22:20
TZ: GMT+01 DS: 1 - 23:20

CORE version is 2_7_0

TZ: GMT-01 DS: 0 - 22:25
TZ: GMT+00 DS: 0 - 21:25
TZ: GMT+01 DS: 0 - 20:25

TZ: GMT-01 DS: 1 - 21:25
TZ: GMT+00 DS: 1 - 20:25
TZ: GMT+01 DS: 1 - 19:25

CORE version is 2_7_1
TZ: GMT-01 DS: 0 - 22:28
TZ: GMT+00 DS: 0 - 21:28
TZ: GMT+01 DS: 0 - 20:28

TZ: GMT-01 DS: 1 - 21:28
TZ: GMT+00 DS: 1 - 20:28
TZ: GMT+01 DS: 1 - 19:28

d-a-v, 2020-05-20:
Thanks for reporting. Sorry for the mistake.
Fix will come with for 2.7.2 second bugfix release.

Juppit, 2020-05-20:
@d-a-v there is a little type in my comment:

> To generate the right timestamps one can use negative timezone an daylight setting.

should be:

> use negative timezone and daylight setting

It means, the dayligth setting is also wrong.

d-a-v, 2020-05-20:
timezone and daylight are now right with #7320 .
Would you confirm that ? Testing it should be [pretty straightforward](https://github.com/esp8266/Arduino/pull/7320/files#diff-1eb89666c4f155202926695147d6a1a0).

Juppit, 2020-05-20:
Yes I confirm, it's now right with #7320 

> CORE version is 3_0_0_DEV_NIGHTLY
TZ: GMT-01 DS: 0 - 17:53
TZ: GMT+00 DS: 0 - 18:53
TZ: GMT+01 DS: 0 - 19:53

>TZ: GMT-01 DS: 1 - 18:53 #73
TZ: GMT+00 DS: 1 - 19:53
TZ: GMT+01 DS: 1 - 20:53

I overlooked the fact that both parts were solved with one patch.
This issue can be closed after merging #7320

PR #7320 (2020-05-20): Fix configTime(tz,dst,)
fixes #7319

Timezone sign was incorrectly set
closed 2020-05-20 by d-a-v