esp8266/Arduino #7326: Trim BearSSL::WiFiClientSecure memory consumption to address WDT reset/OOM

marcelstoer, 2020-05-25:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:				WeMos D1 Mini Pro
- Core Version:			2.7.1
- Development Env:		Arduino IDE
- Operating System:		MacOS

### Settings in IDE

- Module:  			Wemos D1 mini Pro
- Flash Mode:			[qio|dio|other]
- Flash Size: 		16MB
- lwip Variant:   v2 Lower Memory
- Reset Method:		[ck|nodemcu]
- Flash Frequency:	-
- CPU Frequency:		80Mhz
- Upload Using:		SERIAL
- Upload Speed:   921600

### Problem Description

My app seems to have too little heap available to make successful TLS connections with BearSSL. So, (how) can I trim/reduce the memory consumption? I have currently even enabled `client.setInsecure()`.
When the available heap is around 21.5KB `connect()` (silently) fails but with debugging enabled I see `BSSL:_connectSSL: OOM error`.
When the available heap is around 26KB `connect()` triggers a WDT reset around `WiFiClientSecure::_run_until` as per the below stacktrace.

```
0x4020bed2: BearSSL::WiFiClientSecure::_run_until(unsigned int, bool) at /Users/marcelstoer/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.1/cores/esp8266/PolledTimeout.h line 243
0x4023d146: br_ssl_engine_hs_reset at src/ssl/ssl_engine.c line 1320
0x4020c1f4: BearSSL::WiFiClientSecure::_wait_for_handshake() at /Users/marcelstoer/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.1/libraries/ESP8266WiFi/src/WiFiClientSecureBearSSL.cpp line 564
0x4020c45b: BearSSL::WiFiClientSecure::_connectSSL(char const*) at /Users/marcelstoer/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.1/libraries/ESP8266WiFi/src/WiFiClientSecureBearSSL.cpp line 1122
0x40100b61: check_poison_neighbors(unsigned short) at /Users/marcelstoer/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.1/cores/esp8266/umm_malloc/umm_local.c line 71
0x40100b61: check_poison_neighbors(unsigned short) at /Users/marcelstoer/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.1/cores/esp8266/umm_malloc/umm_local.c line 71
0x4020a9d7: WiFiClient::connect(IPAddress, unsigned short) at /Users/marcelstoer/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.1/libraries/ESP8266WiFi/src/WiFiClient.cpp line 177
0x4020a9b0: WiFiClient::connect(IPAddress, unsigned short) at /Users/marcelstoer/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.1/libraries/ESP8266WiFi/src/include/ClientContext.h line 154
0x40100a4c: get_poisoned(void*, size_t) at /Users/marcelstoer/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.1/cores/esp8266/umm_malloc/umm_poison.c line 112
0x402372f2: ip4_output_if at core/ipv4/ip4.c line 1550
0x40237e9b: ip_chksum_pseudo at core/inet_chksum.c line 395
0x402333e6: tcp_output at core/tcp_out.c line 1361
0x402326d1: tcp_create_segment at core/tcp_out.c line 190
0x402147dc: __esp_yield() at /Users/marcelstoer/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.1/cores/esp8266/core_esp8266_main.cpp line 119
0x40215006: __delay(unsigned long) at /Users/marcelstoer/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.1/cores/esp8266/core_esp8266_wiring.cpp line 54
0x4020a998: WiFiClient::connect(IPAddress, unsigned short) at /Users/marcelstoer/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.1/libraries/ESP8266WiFi/src/include/ClientContext.h line 144
0x4020c5c3: BearSSL::WiFiClientSecure::connect(char const*, unsigned short) at /Users/marcelstoer/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.1/libraries/ESP8266WiFi/src/WiFiClientSecureBearSSL.cpp line 231
```

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp
  const char host[] = "adsbexchange.com";
  String path = "/api/aircraft/json/lat/37.16611/lon/-119.44944/dist/50/";
  int port = 443;

//  // Cloudflare CA cert
//  static const char cloudflare[] PROGMEM = R"EOF(
//-----BEGIN CERTIFICATE-----
//MIIDozCCAougAwIBAgIQD/PmFjmqPRoSZfQfizTltjANBgkqhkiG9w0BAQsFADBa
//MQswCQYDVQQGEwJJRTESMBAGA1UEChMJQmFsdGltb3JlMRMwEQYDVQQLEwpDeWJl
//clRydXN0MSIwIAYDVQQDExlCYWx0aW1vcmUgQ3liZXJUcnVzdCBSb290MB4XDTE1
//MTAxNDEyMDAwMFoXDTIwMTAwOTEyMDAwMFowbzELMAkGA1UEBhMCVVMxCzAJBgNV
//BAgTAkNBMRYwFAYDVQQHEw1TYW4gRnJhbmNpc2NvMRkwFwYDVQQKExBDbG91ZEZs
//YXJlLCBJbmMuMSAwHgYDVQQDExdDbG91ZEZsYXJlIEluYyBFQ0MgQ0EtMjBZMBMG
//ByqGSM49AgEGCCqGSM49AwEHA0IABNFW9Jy25DGg9aRSz+Oaeob/8oayXsy1WcwR
//x07dZP1VnGDjoEvZeFT/SFC6ouGhWHWPx2A3RBZNVZns7tQzeiOjggEZMIIBFTAS
//BgNVHRMBAf8ECDAGAQH/AgEAMA4GA1UdDwEB/wQEAwIBhjA0BggrBgEFBQcBAQQo
//MCYwJAYIKwYBBQUHMAGGGGh0dHA6Ly9vY3NwLmRpZ2ljZXJ0LmNvbTA6BgNVHR8E
//MzAxMC+gLaArhilodHRwOi8vY3JsMy5kaWdpY2VydC5jb20vT21uaXJvb3QyMDI1
//LmNybDA9BgNVHSAENjA0MDIGBFUdIAAwKjAoBggrBgEFBQcCARYcaHR0cHM6Ly93
//d3cuZGlnaWNlcnQuY29tL0NQUzAdBgNVHQ4EFgQUPnQtH89FdQR+P8Cihz5MQ4NR
//E8YwHwYDVR0jBBgwFoAU5Z1ZMIJHWMys+ghUNoZ7OrUETfAwDQYJKoZIhvcNAQEL
//BQADggEBADhfp//8hfJzMuTVo4mZlmCvMsEDs2Xfvh4DyqXthbKPr0uMc48qjKkA
//DgEkF/fsUoV2yOUcecrDF4dQtgQzNp4qnhgXljISr0PMVxje28fYiCWD5coGJTH9
//vV1IO1EB3SwUx8FgUemVAdiyM1YOR2aNbM2v+YXZ6xxHR4g06PD6wqtPaU4JWdRX
//xszByOPmGcFYOFLi4oOF3iI03D+m968kvOBvwKtoLVLHawVXLEIbLUiHAwyQq0hI
//qSi+NIr7uu30YJkdFXgRqtltU39pKLy3ayB2f6BVA3F59WensKAKF1eyAKmtz/9n
//jD4m5ackvMJvEOiJxnCl0h+A7Q0/JxM=
//-----END CERTIFICATE-----
//)EOF";
//  
  WiFiClientSecure client;
//  X509List cert(cloudflare);
//  client.setTrustAnchors(&cert);
  client.setInsecure();

  Serial.println("Free Heap: " + String(ESP.getFreeHeap()));
  Serial.printf("Requesting resource at %s:%u%s\n", host, port, path.c_str());
  if(client.connect(host, port)) {
    Serial.println("[HTTP] connected, now GETting data");
    client.print("GET " + path + " HTTP/1.1\r\n"
                 "Host: " + host + "\r\n"
                 "Api-Auth: some-api-key\r\n"
                 "Accept: application/json\r\n"
                 "Connection: close\r\n\r\n");

    while (client.connected() || client.available()) {
      if (client.available()) {
        ...
      }
      // give WiFi and TCP/IP libraries a chance to handle pending events
      yield();
    }
    client.stop();
  } else {
    Serial.println("[HTTP] failed to connect to host");
  }
}

```

earlephilhower, 2020-05-25:
This is a "How do I..." question without an MCVE (only a snippet).  The proper place to take this up would be https://esp8266.com or https://gitter.im/esp8266/Arduino .

That said, BearSSL itself doesn't allocate memory once the connection object is made, but the rest of the TCP stack and core most certainly do.  If you're *even close* OOM, you need to reduce the heap usage elsewhere in your app.  It is possible to set the SSL RX buffer size smaller than the standard, but you risk breaking SSL connections that try and send larger amounts of data than the allocated size.  See the examples for more info.

In general, allocate a single WiFiClientSecure object at the start of the app (global is a good way to do this) as that will allocate all needed memory in one go and not have to fight w/a fragmented heap.  Use PROGMEM wherever possible, avoid `String`s, etc.

closed 2020-05-25 by earlephilhower

devyte, 2020-05-25:
Hi @marcelstoer, as additional comments, the bssl connection requires a 16KB buffer plus a ~6KB stack, so by itself that's already ~22KB. You need some leeway in addition for the actual functioning of the code. 
I would ask here: what happens if you try with more free heap available? I'm thinking about the wdt. Strictly speaking, you should either see an oom, or get an error condition, but not a wdt.

earlephilhower, 2020-05-26:
https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/examples/BearSSL_MaxFragmentLength/BearSSL_MaxFragmentLength.ino  shows the MFLN usage and the buffer shrinking.

marcelstoer, 2020-05-26:
> without an MCVE (only a snippet)

Sorry about this. I don't normally do this but the API (ADS-B Exchange) is non-public anyway.

> In general, allocate a single WiFiClientSecure object at the start of the app (global is a good way to do this) as that will allocate all needed memory in one go and not have to fight w/a fragmented heap. Use PROGMEM wherever possible, avoid `String`s, etc.

That's how I got from 21.5KB free heap to 26KB with changing only a few lines. I'm working with an old code basis I'm not (yet) familiar with that was programmed in a rather wasteful way. I have quite a bit more refactoring to do it seems.

> I would ask here: what happens if you try with more free heap available?

I started with a blank sketch initially that, in the end, contained only connecting to WiFi, NTP-sync, and TLS connection verified against a CA cert. I needed this to establish a base line. It leaves plenty of heap (~43KB) before the TLS connection is established.

> I'm thinking about the wdt. Strictly speaking, you should either see an oom, or get an error condition, but not a wdt.

My thinking exactly. I wasn't surprised to get the OOM but very surprised about the latter.

earlephilhower, 2020-05-26:
If the ADSB-E server supports MFLN, look at the sample https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/examples/BearSSL_MaxFragmentLength/BearSSL_MaxFragmentLength.ino because it can shrink the 16K SSL receive buffer to .5K , legally.  With that, ~8KB total for SSL connections is possible.

If it doesn't, but the protocol is guaranteed to be simple enough to only reply with <= (some number of bytes) you can arbitrarily set smaller buffers , even w/o MFLN, with the understanding if the proto changes or the server tries to reply with > buffsize bytes it will fail.

WDT could be memory corruption (someone not checking malloc return in the binary blob, LWIP, the core, or the app).  That's a way bigger heap of trouble than getting a single app running...

marcelstoer, 2020-05-26:
> WDT could be memory corruption [...]. That's a way bigger heap of trouble than getting a single app running...

Indeed, and the fact that WDT is always around `BearSSL::WiFiClientSecure::_run_until` doesn't necessarily mean there's actually anything wrong with _that_ function.

mizuapo, 2023-02-14:
> If the ADSB-E server supports MFLN, look at the sample https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/examples/BearSSL_MaxFragmentLength/BearSSL_MaxFragmentLength.ino because [...]

Solved it for me, thanks!

`client.setBufferSizes(512, 512);`

marcelstoer, 2023-02-14:
@mizuapo Thanks for the feedback Gabor.

> Solved it for me

Does this refer to the ADS-B Exchange API specifically or was it a general remark?