esp8266/Arduino #7327: wire.setClock() no longer works with values above 334000

bulldog4, 2020-05-26:
### Basic Infos

- [x ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x ] I have tested that the issue is present in current master branch (aka latest git).
- [x ] I have searched the issue tracker for a similar issue.
- [x ] If there is a stack dump, I have decoded it.
- [x ] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12]
- Core Version:			[2.7.1]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Nodemcu]
- Flash Mode:			[qio]
- Flash Size: 		[4MB]
- lwip Variant:   [v2 Lower Memory]
- Reset Method:		[nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200](serial upload only)

### Problem Description
Previously working program using core 2.6.3 no longer works with 2.7.0 or 2.7.1.  This program uses a PCF8574 with I2C interface and communicates with a matrix keypad. My program has always contained the statement:

      wire.setClock(400000);

Using 2.7.1, the program, containing the above statement, no longer detects any key presses. I noticed that changes were made in v2.7.0 to allow I2C clocks below 100k. I believe these changes caused the new problem. Now if I change the above line to:

      wire.setClock(334000);

the program works as it did before. If I change the frequency to 335000 or higher, the PCF8574 no longer detects any key presses.

I reverted back to v2.6.3 and the problem does not exist. I also found that if I comment out the setClock() line, the program functions correctly with v2.7.1, though I'm not sure what clock frequency is being used.

devyte, 2020-05-26:
CC @Tech-TX

earlephilhower, 2020-05-26:
I'm seeing pretty good clocks with master.  I used the `master_writer.ino` example and added a line in `setup()` to set it to 400khz:
````
void setup() {
  Wire.begin(SDA_PIN, SCL_PIN, I2C_MASTER); // join i2c bus (address optional for master)
  Wire.setClock(400000);
}
````

It's showing 350-360 khz clocking which is as close as you're going to get w/SW-driven I2C:
[image]

My guess, @bulldog4, is that the older version was running slower than requested (i.e. you asked for 400khz but got closer to 300khz)), whereas now the current master is running a closer to spec and going beyond the board/chip's capabilities.  @Tech-TX spent some time working on the formula and measuring clock speeds for his changes, and I believe they generally were closer to requested clock (i.e. faster) than the older code.

Can you put on a scope or logic analyzer and catch the clocks you're generating?

earlephilhower, 2020-05-26:
I checked out 2.6.2 and re-ran the exact same test.  Guess what the clocks came out as when I requested 400khz:
[image]

Yup, 320khz.  So older version was (320-400)/400 = 20% slower than requested, but master is now closer at (355 - 400)/400 = 11% off.

So, no code problem here.  It's the circuit or IC which is having trouble at the higher speed (355 vs 320)

Tech-TX, 2020-05-26:
PCF8574 and 8574A are 100KHz bus.  If you got it to run faster before,  you were lucky.  As Earle mentioned,  yes,  the bus clock is a bit more accurate now,  so you should drop that wire.setClock() to a more reasonable speed. Violating the datasheet spec is never a good idea.

https://www.nxp.com/products/interfaces/ic-spi-serial-interface-devices/ic-general-purpose-i-o/remote-8-bit-i-o-expander-for-icbus-with-interrupt:PCF8574_74A

closed 2020-05-26 by earlephilhower

bulldog4, 2020-05-26:
Thanks for the quick response. I forgot to mention that the code to set the clock frequency to 400k came with some code I'm using to drive an SSD1306 OLED display. I guess I was just lucky that it worked with the PCF8574 in the older code. Thanks for the work on this.

Tech-TX, 2020-05-27:
The default I2C bus clock is 100KHz unless you specify otherwise.  >99% of parts will run with that, so it's a good starting point.

I have half of a tube of TI PCF8574AN leftovers, so I immediately knew where you'd gone astray. ;-)