esp8266/Arduino #7330: ESP8266 BearSSL Server how to reduce memory consumption

esseredigitale, 2020-05-27:
### Basic Infos

- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [X] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [X] If there is a stack dump, I have decoded it.
- [X] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12F]
- Core Version:			[2.3.3]
- Development Env:		[Platformio]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[other]
- Flash Mode:			[dio]
- Flash Size: 		[4MB]
- lwip Variant:   [Higher Bandwidth]
- Reset Method:		[ck|nodemcu] I don't know
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[160MHz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200)

### Problem Description

I implemented my https server using the example at the link https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WebServer/examples/HelloServerBearSSL/HelloServerBearSSL.ino .
At server startup the available memory is 30000 but after the first request it becomes 4000 and after a couple of other requests, the system restarts.
How can i apply MLFN or other gimmicks to reduce or free memory?
I read the article at https://arduino-esp8266.readthedocs.io/en/latest/esp8266wifi/bearssl-client-secure-class.html, but I don't know how to apply client use in my code because I implement the server like the example.
Can you help me?

earlephilhower, 2020-05-27:
Closing because template ignored was ignored, no MCVE, and the "How do I ..." question.  Please have such discussions on https://esp8266.com or https://gitter.im/esp8266/Arduino

Short answer is BSSL supports MFLN out of the box, with no add'l work, but it's a client-side request (and not used by any major web browser) so the server can't save any recv. buffer with it, anyway, because the buffer needs to be there before connections are made.

closed 2020-05-27 by earlephilhower

esseredigitale, 2020-05-27:
Sorry but, MCVE was https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WebServer/examples/HelloServerBearSSL/HelloServerBearSSL.ino and "How do I ..." was referred to that example.
Thanks anyway