esp8266/Arduino #7331: ESP crashes on ESPhttpUpdate.update while using BearSSL

wahibmichael, 2020-05-27:
- [x ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x ] I have tested that the issue is present in current master branch (aka latest git).
- [x ] I have searched the issue tracker for a similar issue.
- [x ] If there is a stack dump, I have decoded it.
- [x ] I have filled out all fields below.

#### Platform

- Hardware:				[ESP-12]
- Core Version:			2.7.1 
- Development Env:		[Platformio]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Nodemcur]
- Flash Mode:			[dio]
- Flash Size: 		[4MB]
- lwip Variant:   [v1.4|v2 Lower Memory]
- Reset Method:		[nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200|] (serial upload only)

### Problem Description

-I'm pushing OTA update from HTTS website using ESPhttpUpdate.update
-I'm passing the URL and certificate fingerprint in a JSON sting using MQTT
-once the execution reaches ESPhttpUpdate.update ESP crashes
-if I'm using HTTP update it works fine with no crashes
- If I copy/ paste my update code in an empty sketch it works fine
- free heap memory before execution is 27704

```cpp

if (topicstr == Otaupdate)
  {
      ESPhttpUpdate.rebootOnUpdate(false);
      ESPhttpUpdate.closeConnectionsOnUpdate(false);
      ESPhttpUpdate.setLedPin(LED_BUILTIN, LOW);
     t_httpUpdate_return ret;
      if (port == 443)
      {
        BearSSL::WiFiClientSecure wifisecure;
        wifisecure.probeMaxFragmentLength("raw.githubusercontent.com", 443, 1024);
        wifisecure.setBufferSizes(1024, 1024);
        wifisecure.setFingerprint("70 94 DE DD E6 C4 69 48 3A 92 70 A1 48 56 78 2D 18 64 E0 B7");
        ret = ESPhttpUpdate.update(wifisecure, "https://raw.githubusercontent.com/wahibmichael/OTA/master/firmware.bin");

### Debug Messages

Decoding stack results
0x40246aea: br_ssl_engine_hs_reset at src/ssl/ssl_engine.c line 1320
0x40237369: glue2esp_linkoutput at glue-esp/lwip-esp.c line 301
0x40237a14: ethernet_output at netif/ethernet.c line 312
0x402375f6: new_linkoutput at glue-lwip/lwip-git.c line 260
0x40237a14: ethernet_output at netif/ethernet.c line 312
0x4023f1cc: etharp_output_to_arp_index at core/ipv4/etharp.c line 770
0x4023f420: etharp_output_LWIP2 at core/ipv4/etharp.c line 885
0x40240c14: ip4_output_if_opt_src at core/ipv4/ip4.c line 1764
0x40241998: new_input at glue-lwip/lwip-git.c line 284
0x4023f358: etharp_output_LWIP2 at core/ipv4/etharp.c line 792
0x402375b8: new_linkoutput at glue-lwip/lwip-git.c line 220
0x40237500: netif_sta_status_callback at glue-lwip/lwip-git.c line 301
0x40241998: new_input at glue-lwip/lwip-git.c line 284
0x4023f358: etharp_output_LWIP2 at core/ipv4/etharp.c line 792
0x402375b8: new_linkoutput at glue-lwip/lwip-git.c line 220
0x40238fd0: dns_recv at core/dns.c line 1171
0x40238b74: lwip_cyclic_timer at core/timeouts.c line 233
0x40238b74: lwip_cyclic_timer at core/timeouts.c line 233
0x40238b74: lwip_cyclic_timer at core/timeouts.c line 233
0x40238b74: lwip_cyclic_timer at core/timeouts.c line 233
0x40238b74: lwip_cyclic_timer at core/timeouts.c line 233
0x40238b74: lwip_cyclic_timer at core/timeouts.c line 233
0x40238b74: lwip_cyclic_timer at core/timeouts.c line 233
0x40271220: ethernet_input at glue-esp/lwip-esp.c line 332
0x40280338: etharp_output at glue-esp/lwip-esp.c line 329
0x402413e8: sntp_try_next_server at apps/sntp/sntp.c line 401
0x40238c14: tcpip_tcp_timer at core/timeouts.c line 145
0x4023e2b0: dhcp_recv at core/ipv4/dhcp.c line 1762
0x402414a8: sntp_recv at apps/sntp/sntp.c line 437
0x402464e0: jump_handshake at src/ssl/ssl_engine.c line 1081
0x40246811: br_ssl_engine_sendrec_ack at src/ssl/ssl_engine.c line 1172

earlephilhower, 2020-05-27:
Please post an MCVE.  A little snippet is not enough to reproduce your crash.

Also, please include the faulting PC, too.  You've got the stack dump, but not the current function that's running, which isn't terribly useful.  Cut-n-paste the entire serial output and the decoder will parse the register dump, too, to get the faulting PC.

So, cut and paste from the "Exception" line...
````
Exception (29):
epc1=0x40201d16 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

>>>stack>>>
ctx: cont
sp: 3ffffdb0 end: 3fffffc0 offset: 0190
3fffff40:  40210630 3ffee630 3ffee89c 40201d14  
3fffff50:  00000000 00000000 00000000 00000000  
...
3fffffb0:  feefeffe feefeffe 3ffe8540 40100df9  
<<<stack<<<
````

The decoded output will include the "Exception" # decoded, like:
````
Exception 29: StoreProhibited: A store referenced a page mapped with an attribute that does not permit stores
PC: 0x40201d16: setup() at C:\Users\earle\AppData\Local\Temp\arduino_modified_sketch_549402/HelloServerBearSSL.ino line 143
EXCVADDR: 0x00000000

Decoding stack results
0x40210630: std::_Function_handler ::_M_invoke(std::_Any_data const&) at c:\users\earle\documents\arduino\hardware\esp8266com\esp8266\tools\xtensa-lx106-elf\xtensa-lx106-elf\include\c++\4.8.2/functional line 2069
...
````
My guess is you're OOM and some portion of the blob, LWIP, or code is blowing up.  Especially since you're talking about JSON/etc., I imagine lots of String usage and lots of memory fragmentation.

But, w/o a full dump and MCVE there's nothing we can look at, yet.

wahibmichael, 2020-05-27:
Thanks for your feedback

unfortunately, the faulting PC is not appearing in the stack dump, shown below the start of the dump. I also did a Serial Debug and included the results below, I'll be working on MCVE

Soft WDT reset

>>>stack>>>

ctx: sys
sp: 3fff0750 end: 3fffffb0 offset: 01a0
3fff08f0:  00000000 00000000 3fff5324 40209db1  
3fff0900:  00416021 00000000 3fff5324 4022a65e  
3fff0910:  00000000 00000000 3fff5cdc 00000008  
3fff0920:  00003a98 000538c8 3fff0e88 4020b223  
3fff0930:  000000e6 00000005 794407ce 00000000  
3fff0940:  00000001 00002d0c 3fff5cdc 40246b46  
3fff0950:  3fff61dc 3fff5444 3fff5cdc 00000001  
3fff0960:  00000001 00000001 3fff0e88 4020b514  
3fff0970:  3fff5444 00000000 3fff0e88 4020b6f7  
3fff0980:  00000035 3fff1228 00000001 4022a5f4  
3fff0990:  00000000 00000000 00000000 4022abde  
3fff09a0:  00000000 3fff0e88 3fff5324 4020a17c  
3fff09b0:  3ffeff90 00000d50 00000000 00000000  
3fff09c0:  000001bb 3fff0e88 3fff5444 00000000  
3fff09d0:  000001bb 3fff0e88 3fff5444 4020b815  
3fff09e0:  4022f370 85f06597 4022f370 85f06597  

[HTTP-Client][begin] url: https://raw.githubusercontent.com/wahibmichael/OTA/master/firmware.bin
[HTTP-Client][begin] host: raw.githubusercontent.com port: 443 url: /wahibmichael/OTA/master/firmware.bin
[HTTP-Client][sendRequest] type: 'GET' redirCount: 0
[hostByName] request IP for: raw.githubusercontent.com
[hostByName] Host: raw.githubusercontent.com IP: 151.101.240.133
:ref 1
BSSL:_connectSSL: start connection

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 3656, room 16 
tail 8
chksum 0x0c
csum 0x0c
v9c56ed1f
~ld

SDK:2.2.2-dev(38a443e)/Core:2.7.0=20700000/lwIP:STABLE-2_1_2_RELEASE/glue:1.2-30-g92add50/BearSSL:5c771be

earlephilhower, 2020-05-27:
That crash dump is coming from SYStem context:

>ctx: sys

If it was dying in running BearSSL code, it would say `ctx: bearssl`.  In user code (app/core) it would be `ctx: cont`.

So my best guess is that you're OOM and something in the binary blob or the ROM is going nuts.  The binary stuff we can't touch, and the blob can malloc() memory at any time (for WiFi buffers, etc.) so there needs to be some free RAM margin available always.

Another possibility is that you are blowing the user stack.  Don't allocate large objects on the stack (the WiFiClientSecure, the certs, etc.).  Make them global or allocate right in `setup()` while saving a global pointer to them.  There is no MMU, so if you overallocate from the 4K (total!) user stack, it'll overwrite the SYS stack and destroy things.

wahibmichael, 2020-05-27:
Hello probably it's a memory issue while creating the MCVE I removed some functions from MQTT receive topics and it didn't crash
but the free heap memory was 29200 and my sketch crashes when it was 27704 so it's not enough?
Yes I'm not declaring large object at setup and I'm declaring them just before usage thinking that global allocation takes more memory reservation
excuse me I'm a little bit noob in this area if it's possible to point me to an article that describes the memory map and the way to detect that I'm out of memory I'm using ESP.getFreeHeap() and it was thinking it's fine as I have memory left

earlephilhower, 2020-05-27:
A SSL connection needs about 6KB stack, 16KB recv, and 1K xmit buffers, so ~23KB total.  Additional ones won't have the 6KB stack (only 1 needed no matter how many connections), but still will need ~17KB per.

Allocating 16K contiguous and 6K contiguous blocks can be tough if you have fragmentation.  There is no MMU, so `getFreeHeap()` can truthfully say "you have 30KB free" and `malloc(1024)` could fail.  That's why you'd want to make the large WiFiClientSecure object as a global or allocated in setup, to make it easy to find space for them.  And, once those large chunks are allocated, it will make it that much harder for the SYS/etc. to find space for their own large allocations.

Check the https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/examples/BearSSL_MaxFragmentLength/BearSSL_MaxFragmentLength.ino example for a way to reduce the memory needs of SSL, assuming the server you're talking to supports a special extension (not all do).

Check with http://esp8266.com or https://gitter.im/esp8266/Arduino for more assistance, there's a lot of knowledge there.

closed 2020-05-27 by earlephilhower

wahibmichael, 2020-05-27:
Thanks a lot for hinting, 
note for future readers the idea described clearly here:
https://www.esp8266.com/viewtopic.php?p=69937
and here:
https://cpp4arduino.com/2018/11/06/what-is-heap-fragmentation.html