esp8266/Arduino #7333: Wemos d1 mini PRO 16MB SPIFFS Data upload failed

rserroyen, 2020-05-28:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:					Wemos d1 mini PRO 16MB 
- Core Version:			[latest git hash or date]
- Development Env:		Arduino IDE
- Operating System:		Windows

### Settings in IDE

- Module:  			Wemos D1 R1
- Flash Mode:			dio
- Flash Size: 		4MB
- lwip Variant:   Higher Bandwidth
- Reset Method:		ck
- Flash Frequency:	40Mhz
- CPU Frequency:		80Mhz
- Upload Using:		SERIAL
- Upload Speed:   115200

### Problem Description

Hello
When I run the ESP8266 LittleFS program (or ESP8266 Sketch) or I am fine. But when you run to look at dir / data there is nothing !!! Why

Thank you

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```#include <LittleFS.h>   //Include File System Headers

#include <LittleFS.h>   //Include File System Headers

FSInfo fs_info;
void setup() {
  //------------------------------------------------Serial
  Serial.begin(115200);
  while (!Serial) {} // phase développement
  Serial.println("Booting");

  //----------------------------------------------LittleFS
  // Get all information of your LittleFS

 FSInfo fs_info;
  LittleFS.info(fs_info);

  Serial.println("File system info.");

  Serial.print("Total space:      ");
  Serial.print(fs_info.totalBytes);
  Serial.println("byte");

  Serial.print("Total space used: ");
  Serial.print(fs_info.usedBytes);
  Serial.println("byte");

  Serial.print("Block size:       ");
  Serial.print(fs_info.blockSize);
  Serial.println("byte");

  Serial.print("Page size:        ");
  Serial.print(fs_info.totalBytes);
  Serial.println("byte");

  Serial.print("Max open files:   ");
  Serial.println(fs_info.maxOpenFiles);

  Serial.print("Max path lenght:  ");
  Serial.println(fs_info.maxPathLength);

  Serial.println();

  Serial.print("Chip ID: ");
  Serial.println(ESP.getFlashChipId());

  Serial.print("Chip Real Size: ");
  Serial.println(ESP.getFlashChipRealSize());

  Serial.print("Chip Size: ");
  Serial.println(ESP.getFlashChipSize());

  Serial.print("Chip Speed: ");
  Serial.println(ESP.getFlashChipSpeed());

  Serial.print("Chip Mode: ");
  Serial.println(ESP.getFlashChipMode());

  FlashMode_t ideMode = ESP.getFlashChipMode();
  Serial.printf("Flash ide mode:  %s\n", (ideMode == FM_QIO ? "QIO" : ideMode == FM_QOUT ? "QOUT" : ideMode == FM_DIO ? "DIO" : ideMode == FM_DOUT ? "DOUT" : "UNKNOWN"));

  
  uint32_t realSize = ESP.getFlashChipRealSize();
  uint32_t ideSize = ESP.getFlashChipSize();
  
  Serial.printf("Flash ide speed: %u Hz\n", ESP.getFlashChipSpeed());

  if (ideSize != realSize) {
    Serial.println("Flash Chip configuration wrong!\n");
  } else {
    Serial.println("Flash Chip configuration ok.\n");
  }

  File root = LittleFS.open("/data/", "r"); // pour esp8266
  Serial.println("LittleFS.open passed ");

Dir dir = LittleFS.openDir("/data");

Serial.println("LittleFS.dir passed ");

while (dir.next()) {
    Serial.print(dir.fileName());
    if(dir.fileSize()) {
        File f = dir.openFile("r");
        Serial.println(f.size());
    }
}
  
  
  LittleFS.end();
  Serial.println("LittleFS.close passed ");
}

void loop() {
  // put your main code here, to run repeatedly:

}
```

### Debug Messages

``
[LittleFS] data    : D:\Data Jeanne\Arduino\spiff\4_read_files\data
[LittleFS] size    : 2024
[LittleFS] page    : 256
[LittleFS] block   : 8192
/debug.log

/index.html

/script.js

/testfile.txt.txt

/w3.css

[LittleFS] upload  : C:\Users\rserr\AppData\Local\Temp\arduino_build_353114/4_read_files.mklittlefs.bin
[LittleFS] address : 0x200000
[LittleFS] reset   : --before default_reset --after hard_reset
[LittleFS] port    : COM8
[LittleFS] speed   : 115200
[LittleFS] python   : C:\Users\rserr\AppData\Local\Arduino15\packages\esp8266\tools\python3\3.7.2-post1\python3.exe
[LittleFS] uploader : C:\Users\rserr\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.1\tools\upload.py

esptool.py v2.8

Serial port COM8

Connecting....

Chip is ESP8266EX

Features: WiFi

Crystal is 26MHz

MAC: 84:f3:eb:8e:50:99

Uploading stub...

Running stub...

Stub running...

Configuring flash size...

Auto-detected Flash size: 16MB

Compressed 2072576 bytes to 4611...

Writing at 0x00200000... (100 %)
Wrote 2072576 bytes (4611 compressed) at 0x00200000 in 0.4 seconds (effective 40515.1 kbit/s)...

Hash of data verified.

Leaving...

Hard resetting via RTS pin...
```
```
15:15:30.889 -> File system info.
15:15:30.889 -> Total space:      4277137406byte
15:15:30.889 -> Total space used: 4277137406byte
15:15:30.889 -> Block size:       4277137406byte
15:15:30.889 -> Page size:        4277137406byte
15:15:30.889 -> Max open files:   4277137406
15:15:30.889 -> Max path lenght:  4277137406
15:15:30.889 -> 
15:15:30.889 -> Chip ID: 1589448
15:15:30.889 -> Chip Real Size: 16777216
15:15:30.889 -> Chip Size: 16777216
15:15:30.889 -> Chip Speed: 40000000
15:15:30.889 -> Chip Mode: 2
15:15:30.889 -> Flash ide mode:  DIO
15:15:30.889 -> Flash ide speed: 40000000 Hz
15:15:30.889 -> Flash Chip configuration ok.
15:15:30.889 -> 
15:15:30.889 -> LittleFS.open passed 
15:15:30.889 -> LittleFS.dir passed 
15:15:30.889 -> root.openNextFile passed  
15:15:30.889 -> LittleFS.close passed 

```

earlephilhower, 2020-05-28:
You never started LittleFS.  Try adding `LittleFS.begin()`

rserroyen, 2020-05-29:
Added
```
.............  
if (ideSize != realSize) {
    Serial.println("Flash Chip configuration wrong!\n");
  } else {
    Serial.println("Flash Chip configuration ok.\n");
  }
LittleFS.begin();

  File root = LittleFS.open("/data/", "r"); // pour esp8266
  Serial.println("LittleFS.open passed ");

Dir dir = LittleFS.openDir("/data");

Serial.println("LittleFS.dir passed ");

while (dir.next()) {
    Serial.print(dir.fileName());
    if(dir.fileSize()) {
        File f = dir.openFile("r");
        Serial.println(f.size());
    }
```
Debug Messages no file fond !!

```17:46:28.676 -> File system info.
17:46:28.676 -> Total space:      4277137406byte
17:46:28.676 -> Total space used: 4277137406byte
17:46:28.676 -> Block size:       4277137406byte
17:46:28.676 -> Page size:        4277137406byte
17:46:28.676 -> Max open files:   4277137406
17:46:28.676 -> Max path lenght:  4277137406
17:46:28.676 -> 
17:46:28.676 -> Chip ID: 1589448
17:46:28.676 -> Chip Real Size: 16777216
17:46:28.676 -> Chip Size: 16777216
17:46:28.676 -> Chip Speed: 40000000
17:46:28.676 -> Chip Mode: 2
17:46:28.676 -> Flash ide mode:  DIO
17:46:28.676 -> Flash ide speed: 40000000 Hz
17:46:28.676 -> Flash Chip configuration ok.
17:46:28.676 -> 
17:46:28.676 -> LittleFS.open passed 
17:46:28.676 -> LittleFS.dir passed 
17:46:28.676 -> LittleFS.close passed 
```

closed 2020-05-29 by earlephilhower